mpromonet/v4l2rtspserver #81: Half of video stream corrupted unless “-c” option is used

roger-, 2017-12-19:
Using the “-s” read interface and Logitech C920 camera on x64 Ubuntu.

mpromonet, 2017-12-19:
Hi Roger,
This is quite confusing '-s' is for using hte live mainloop for capture, to enable the read interface you should use '-r'.
Nevertheless if the '-c' has an impact tht means problably the split of frames to extract (and repeat) SPS/PPS have some problems.
Maybe using -vvv to enable more log will helps to localize the problems. 
In the v4l2tools project I add recently a tool v4l2dump that dump the H264 frames using h264bitstream, it could also helps.
Best Regards,
Michel.

roger-, 2017-12-20:
Hi Michel,

Yes if you have a look at issue #79 I mentioned that “-s” is the only option that doesn’t crash on OpenWrt/LEDE (I’ve noticed the same on x64 I believe).

I will try collecting more logs on this issue too.

Thanks.

Regards,
Roger

mpromonet, 2017-12-20:
Hi Roger,
If it crash without -s with openwrt, it could be some inconsistencies with the thread option (missing -pthread compiling lib4vl2cpp)
Best Regards,
Michel.

roger-, 2017-12-20:
FYI `-vvv -s` gives me this:

```
        Create V4L2 Source.../dev/video0
[NOTICE] src/V4l2Device.cpp:117
        driver:uvcvideo 84200001
[NOTICE] src/V4l2Device.cpp:121
        /dev/video0 support streaming
[ERROR] src/V4l2Device.cpp:126
        Mandatory capability not available for device:/dev/video0
[ERROR] src/V4l2Device.cpp:67
        Cannot init device:/dev/video0
```

I'll have to look into the thread issues too.

mpromonet, 2017-12-20:
Hi Roger,
Did you specified also '-r' because as your device support only MMAP, this capabilities is missing to run with '-r'.
I add the mandatory mask in the log in commit https://github.com/mpromonet/v4l2rtspserver/commit/d7e8dd4d35a802219222642a2e3b28f2dc5069c1
Best Regards,
Michel.

roger-, 2017-12-21:
Sorry, my last message should have said “-vvv -r”. 

I will try logging with and without “-c” if that helps.

closed 2018-12-16 by mpromonet

roger-, 2022-10-17:
FYI taking a cue from [here](https://npm.io/package/homebridge-camera-multistream
) I'm having success with this command:

```
v4l2rtspserver -I 0.0.0.0  -Q 512 -F 0 -c -s /dev/video0
```