mpromonet/v4l2rtspserver #82: RTP over TCP?

roger-, 2017-12-19:
By default the SDP output seems to have `m=video 0 RTP/AVP 96`, which I believe means it uses UDP+TCP. Is there a way to enable TCP only (interleaved) output?

mpromonet, 2017-12-20:
Hi Roger,
Reading https://tools.ietf.org/html/rfc2326, I guess `RTP/AVP` means UDP, to describe an RTP over TCP session it may be `RTP/AVP/TCP`.
The default behaviour of live555 is to send RTP/AVP in DECRIBE answer and next the SETUP could ask for RTPoverTCP (or multicast session).
This may be overidable but probably not easily, the message is formatted in [OnDemandServerMediaSubsession::setSDPLinesFromRTPSink](https://github.com/rgaufman/live555/blob/master/liveMedia/OnDemandServerMediaSubsession.cpp)
Maybe a simple way you could filter UDP with iptables 
Best Regards,
Michel.

roger-, 2017-12-21:
You’re right, I remember now that the client has to initiate the TCP switch so everything probably fine.

closed 2017-12-21 by roger-