mpromonet/v4l2rtspserver #83: mmap: no such device when running as a service or under screen

scaryfish, 2017-12-21:
Compiled from source, running on a Raspberry Pi zero W under Raspbian Stretch.  The RTSP server runs fine with the command:
`/usr/local/bin/v4l2rtspserver -F 10 -f -P 8555 /dev/video0`
when run over SSH.

When trying to set it up to run as a systemd service, or trying to run under screen, I get the following error:
```
2017-12-22 11:57:53,417 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:563
        Create V4L2 Source.../dev/video0
2017-12-22 11:57:53,690 [NOTICE] - src/V4l2Device.cpp:117
        driver:raspicam capabilities:5000001 mandatory:4000001
2017-12-22 11:57:53,691 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video0 support capture
2017-12-22 11:57:53,691 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video0 support read/write
2017-12-22 11:57:53,691 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video0 support streaming
2017-12-22 11:57:53,987 [NOTICE] - src/V4l2Device.cpp:182
        /dev/video0:H264 size:1920x1080 bufferSize:262144
2017-12-22 11:57:53,989 [NOTICE] - src/V4l2Device.cpp:203
        fps:1/10
2017-12-22 11:57:53,989 [NOTICE] - src/V4l2Device.cpp:204
        nbBuffer:4
2017-12-22 11:57:54,101 [NOTICE] - src/V4l2MmapDevice.cpp:71
        Device /dev/video0 nb buffer:10
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
2017-12-22 11:57:54,123 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:581
        Create Source .../dev/video0
2017-12-22 11:57:54,127 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:161
        Play this stream using the URL "rtsp://192.168.0.82:8555/unicast"
2017-12-22 11:57:54,130 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
Segmentation fault
```

mpromonet, 2017-12-22:
Hi scaryfish,
The mmap call fails reporting device doesnot support mmap, but your device say it cupport streaming, so it should...
Are you using the same user ?
You could try with the '-r' option in order to use read api instead of mmap api.
Best Regards,
Michel.

scaryfish, 2017-12-23:
Hi,
That is the unusual thing.  mmap works fine if I just run the command myself, but if I run it in a screen it doesn't.  The user is the same (pi) for both.  It also fails with the same error if I try to run it through a service, whether running as pi or root.

I can run it with the -r option and it doesn't crash, but I don't get any video when connecting to the stream with VLC.

Thanks for your help.

mpromonet, 2017-12-23:
Hi,
I don't reproduce this with a raspberry pi2 or a nanopi, starting from rc.local works.
But I still use jessie on raspberry and dietpi on nanopi, maybe some new permissions restriction.
In order to debug you coud try to run under strace to try to localize what is the system call that fails.
Best Regards,
Michel.

scaryfish, 2017-12-25:
I've run strace and attached the files.  One is running it "normally" when it works fine (and killing it via ctrl-C when it's started up) and the other is running it under screen when it crashes.

Thanks.
[normal_strace.txt](https://github.com/mpromonet/v4l2rtspserver/files/1586238/normal_strace.txt)
[crash_strace.txt](https://github.com/mpromonet/v4l2rtspserver/files/1586239/crash_strace.txt)

scaryfish, 2017-12-25:
One difference I can spot is the normal strace has the following:
`ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=9, memory=V4L2_MEMORY_MMAP, m.offset=0, length=0, bytesused=0, flags=0, ...}) = 0`
whereas the crashing one has:
`ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=9, memory=0 /* V4L2_MEMORY_??? */, length=262144, bytesused=0, flags=0, ...}) = 0`
which looks like it has a code comment in it?

mpromonet, 2017-12-26:
Hi,
Looking to your logs, i guess the différence is some environment variable like LD_PRELOAD, we see in the normal log that libuv4lext.so is loaded.
Best regards
Michel

scaryfish, 2017-12-26:
Ah great!  That did the trick, it was that environmental variable.  Thanks for your help.

closed 2017-12-26 by scaryfish

IT-Experte, 2019-02-18:
Thanks. It also solve my problem too. Just simply setup the environment variable.

```
LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so
/usr/local/bin/v4l2rtspserver -F 10 -f -P 8555 /dev/video0
```