mpromonet/v4l2rtspserver #85: Pi Zero DietPi - make not working

MrGlasspoole, 2017-12-24:
I have DietPi on a Zero.
If i try:
`cd live
./genMakefiles linux
make CPPFLAGS=-DALLOW_RTSP_SERVER_PORT_REUSE=1 install`

i get:
`root@DietPi:~/live# make CPPFLAGS=-DALLOW_RTSP_SERVER_PORT_REUSE=1 install
cd liveMedia ; make install
make[1]: Entering directory '/root/live/liveMedia'
c++ -c -Iinclude -I../UsageEnvironment/include -I../groupsock/include -I. -O2 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -Wall -DBSD=1 -DALLOW_RTSP_SERVER_PORT_REUSE=1  Media.cpp
make[1]: c++: Command not found
Makefile:32: recipe for target 'Media.o' failed
make[1]: *** [Media.o] Error 127
make[1]: Leaving directory '/root/live/liveMedia'
Makefile:44: recipe for target 'install' failed
make: *** [install] Error 2`

mpromonet, 2017-12-24:
Hi,
Did you install g++ ?
Regards
Michel

MrGlasspoole, 2017-12-26:
Thanks, installing g++ did it.

closed 2018-01-01 by mpromonet

tentner, 2018-01-01:
Our office is closed from 23rd December 2017 to 7th January 2018. We will come back to you starting from 8th January.