mpromonet/v4l2rtspserver #87: Compile error:

phgachoud, 2018-01-08:
` cmake .
CMake Error: The source directory "/root/live" does not appear to contain CMakeLists.txt.`

After downloading with `  wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz -O - | tar xvzf -` as mentioned into readme

mpromonet, 2018-01-08:
Hi,
live555 doesnot build with cmake.
Did you run `./genMakefiles linux && make` ?
Best Regards,
Michel.

closed 2018-02-01 by mpromonet

TopperBG, 2018-02-13:
Here too same error, latest live555.tar.gz
NanoPi Air Ubuntu 16.04.3 LTS 3.4.39-h3

mpromonet, 2018-02-13:
Hi TopperBG,
Please use the issue associated to the project in this case. Posting issue about project X on project Y is confusing. https://github.com/mpromonet/libv4l2cpp/issues
Anyway if you like to install in /usr prefix you should be root, otherwise you can specify the PREFIX variable to indicate where to install.
Regards,
Michel.

TopperBG, 2018-02-13:
Manage to start cmake but new errors:

> -- CMAKE_TOOLCHAIN_FILE=
> -- CMAKE_C_COMPILER=/usr/bin/cc
> -- CMAKE_CXX_COMPILER=/usr/bin/c++
> -- CMAKE_FIND_ROOT_PATH=/build/v4l2rtspserver/armv7l
> -- CMAKE_STAGING_PREFIX=/build/v4l2rtspserver/armv7l
> -- Log4Cpp available
> -- SystemD available
> -- SystemD directory '/lib/systemd/system'
> CMake Error at CMakeLists.txt:118 (install):
>   install FILES given unknown argument "EXCLUDE_FROM_ALL".
> 
> 
> -- Configuring incomplete, errors occurred!
> 

It's consider new service file.
How to skip it?

TopperBG, 2018-02-13:
Resolve by removing EXCLUDE_FROM_ALL in CMakeLists.txt

TopperBG, 2018-02-13:
According error in the topic - please change some kind guide for install because many users try to exec cmake. IN live555 build dir, not in v4l2rtspserver where mean to be!!!

mpromonet, 2018-02-13:
@TopperBG : Please use the button Create issues ans do not reuse issue that is not related to your problem. The problem with cmake come from the version of cmake.
I will delete your comments that are out of the subject of the issue.
Regards,
Michel.

TopperBG, 2018-02-13:
That's resolve the topics and it's problem of step by step guide.

> According error in the topic - please change some kind guide for install because many users try to exec cmake. IN live555 build dir, not in v4l2rtspserver where mean to be!!!

Thank you for the efforts !!