mpromonet/v4l2rtspserver #89: updated raspbian lite kills everything

TechMasterJoe, 2018-01-26:
> root@CM3LB001:/home/pi# v4l2rtspserver -F30 -W1280 -H720 -P 8554 /dev/video0
> 2018-01-26 07:51:45,896 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:436
>         Create V4L2 Source.../dev/video0
> 2018-01-26 07:51:45,899 [ERROR ] - src/V4l2Device.cpp:82
>         Cannot open device:/dev/video0 Operation not permitted
> 2018-01-26 07:51:45,899 [ERROR ] - src/V4l2Device.cpp:65
>         Cannot init device:/dev/video0
> VIDIOC_STREAMOFF: Bad file descriptor
> VIDIOC_REQBUFS: Bad file descriptor
> root@CM3LB001:/home/pi#
> 

on cam 1

> > root@CM3LB001:/home/pi# v4l2rtspserver -F30 -W1280 -H720 -P 8554 /dev/video1
> 2018-01-26 07:56:08,882 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:436
>         Create V4L2 Source.../dev/video1
> VIDIOC_REQBUFS: Inappropriate ioctl for device
> 2018-01-26 07:56:08,882 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:455
>         Create Source .../dev/video1
> 2018-01-26 07:56:08,883 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:93
>         begin thread
> 2018-01-26 07:56:08,884 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:148
>         Play this stream using the URL "rtsp://192.168.3.100:8554/unicast"
> 2018-01-26 07:56:08,884 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:206
>         V4L2DeviceSource::getNextFrame no data errno:0 Success
> 2018-01-26 07:56:08,884 [ERROR ] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:107
>         error:Success
> 2018-01-26 07:56:08,884 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2DeviceSource.cpp:118
>         end thread

so that is all i get after having it working and doing 
apt-get update
apt-get upgrade
rpi-update (to make some hardware work I need 4.9.75+ that has the patch I need)
I'm using a CM3L in a CMIOV3 board with 2x SONY IMX219PQ aka V2 cameras

TechMasterJoe, 2018-01-26:
don't know if it will help
but i get the same errors using the current raspbian lite img 
burn img to card add ssh file using EX2FS for windows i edit dhcpcd.conf with static ip.

boot sd card
run apt-get update
sudo apt-get install device-tree-compiler
sudo dtc -I dts -O dtb -o /boot/dt-blob.bin dt-blob.dts  
/*
enables both cameras on CM3 and set default pin states for some things.
at this point Raspivid works fine with both cameras 
*/
install.sh is stupid simple file
```
sudo apt-get -y install git
sudo apt-get -y install cmake
cd /home/pi
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver
cmake . && make
cpack .
```
once it's in a deb i just install it that way makes it easy to remove

this is what i see when running that install.sh
and installing the package

```
root@raspberrypi:/home/pi# sudo ./install.sh
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  git-man liberror-perl
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-arch git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man liberror-perl
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,839 kB of archives.
After this operation, 26.3 MB of additional disk space will be used.
Get:1 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf liberror-perl all 0.17024-1 [26.9 kB]
Get:2 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf git-man all 1:2.11.0-3+deb9u2 [1,432 kB]
Get:3 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf git armhf 1:2.11.0-3+deb9u2 [3,380 kB]
Fetched 4,839 kB in 3s (1,581 kB/s)
Selecting previously unselected package liberror-perl.
(Reading database ... 34443 files and directories currently installed.)
Preparing to unpack .../liberror-perl_0.17024-1_all.deb ...
Unpacking liberror-perl (0.17024-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../git-man_1%3a2.11.0-3+deb9u2_all.deb ...
Unpacking git-man (1:2.11.0-3+deb9u2) ...
Selecting previously unselected package git.
Preparing to unpack .../git_1%3a2.11.0-3+deb9u2_armhf.deb ...
Unpacking git (1:2.11.0-3+deb9u2) ...
Setting up git-man (1:2.11.0-3+deb9u2) ...
Setting up liberror-perl (0.17024-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up git (1:2.11.0-3+deb9u2) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  cmake-data libarchive13 libjsoncpp1 liblzo2-2 libuv1
Suggested packages:
  codeblocks eclipse ninja-build lrzip
The following NEW packages will be installed:
  cmake cmake-data libarchive13 libjsoncpp1 liblzo2-2 libuv1
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,059 kB of archives.
[... 466 lines cut ...]
  int audioFreq = 44100;
      ^~~~~~~~~
/home/pi/v4l2rtspserver/src/main.cpp:399:6: warning: unused variable ‘audioNbChannels’ [-Wunused-variable]
  int audioNbChannels = 2;
      ^~~~~~~~~~~~~~~
[100%] Linking CXX executable v4l2rtspserver
[100%] Built target v4l2rtspserver
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: v4l2rtspserver
CPack: - Install project: v4l2rtspserver
CPack: Create package
CPack: - package: /home/pi/v4l2rtspserver/v4l2rtspserver-0.0.6-9-gbe42d61-Linux-armv7l.deb generated.
root@raspberrypi:/home/pi# dpkg -i /home/pi/v4l2rtspserver/v4l2rtspserver-0.0.6-9-gbe42d61-Linux-armv7l.deb
Selecting previously unselected package v4l2rtspserver.
(Reading database ... 37537 files and directories currently installed.)
Preparing to unpack .../v4l2rtspserver-0.0.6-9-gbe42d61-Linux-armv7l.deb ...
Unpacking v4l2rtspserver (0.0.6-9-gbe42d61) ...
Setting up v4l2rtspserver (0.0.6-9-gbe42d61) ...
root@raspberrypi:/home/pi# ls
install.sh  v4l2rtspserver
root@raspberrypi:/home/pi# reboot
```
that's it after the reboot Raspivid works fine
but i get the first post errors with rtspserver

mpromonet, 2018-01-26:
HI TechMasterJoe,
It seems you cannot access to v4l2 device.
Did you tried something like : 

     v4l2-ctl -d /dev/video0 --all

Best Regards,
Michel.

MrGlasspoole, 2018-02-26:
I also have this errors in DietPi on a RPi 3 with the V2.1 camera :

`VIDIOC_STREAMON: Operation not permitted`
`VIDIOC_DQBUF: Invalid argument`

tentner, 2018-02-26:
I am at our Booth 4-628 at the Embedded World in Nuremberg.

I will respond to your e-mail when I am back in the office on 1st March 2018.

mpromonet, 2018-02-26:
Hi MrGlasspoole,

It seems you cannot access to v4l2 device.
Did you tried something like :

 v4l2-ctl -d <your device> --all

Best Regards,
Michel.

walterav1984, 2018-03-09:
I was in the same situation using the current Raspbian stretch lite image, however I do have a working older raspbian stretch lite image that does work :-D ! Both use a raspberry pi zero. I noticed that after a reboot on the newer raspbian release that v4l2-ctl finds a v4l2 device and v4l2server does work! 

 I can show some differences to help bisect/discover regression?

Working v4l2rtspserver on older raspbian stretch:
```
uname -a
Linux raspberrypi 4.9.46+ #1032 Wed Aug 30 11:20:57 BST 2017 armv6l GNU/Linux

lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch

cd v4l2rtspserver/
git show
commit 70c82207cd2a501c5d2e81544adc84850900ae54
Merge: e16d29d 9a45066
Author: Michel Promonet <michel.promonet@free.fr>
Date:   Thu Aug 31 20:52:54 2017 +0200

    Merge pull request #76 from mpromonet/add-code-of-conduct-1
    
    Create CODE_OF_CONDUCT.md

sudo modprobe bcm2835-v4l2
dmesg
[  154.242310] media: Linux media interface: v0.10
[  154.282352] Linux video capture interface: v2.00
[  154.345868] bcm2835-v4l2: scene mode selected 0, was 0
[  154.346448] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720
[  154.351574] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.

./v4l2rtspserver -W 1280 -H 720 -F 25 -P 8555
2018-03-09 11:27:48,556 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:446
	Create V4L2 Source.../dev/video0
2018-03-09 11:27:48,562 [NOTICE] - src/V4l2Device.cpp:115
	driver:bm2835 mmal 85200005
2018-03-09 11:27:48,564 [NOTICE] - src/V4l2Device.cpp:117
	/dev/video0 support read/write
2018-03-09 11:27:48,566 [NOTICE] - src/V4l2Device.cpp:119
	/dev/video0 support streaming
2018-03-09 11:27:48,570 [NOTICE] - src/V4l2Device.cpp:177
	/dev/video0:H264 size:1280x720 bufferSize:921600
2018-03-09 11:27:48,572 [NOTICE] - src/V4l2Device.cpp:198
	fps:1/25
2018-03-09 11:27:48,573 [NOTICE] - src/V4l2Device.cpp:199
	nbBuffer:1
2018-03-09 11:27:48,584 [NOTICE] - src/V4l2MmapDevice.cpp:70
	Device /dev/video0 nb buffer:10
2018-03-09 11:27:49,137 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:464
	Create Source .../dev/video0
2018-03-09 11:27:49,142 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:158
	Play this stream using the URL "rtsp://192.168.1.11:8555/unicast"
2018-03-09 11:27:49,145 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
2018-03-09 11:27:49,149 [NOTICE] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:63
	profile-level-id=640028;sprop-parameter-sets=J2QAKKwrQCgC3QDxImo=,KO4CXLA=

v4l2-ctl -d /dev/video0 --all
Driver Info (not using libv4l2):
	Driver name   : bm2835 mmal
	Card type     : mmal service 16.1
	Bus info      : platform:bcm2835-v4l2
	Driver version: 4.9.46
	Capabilities  : 0x85200005
		Video Capture
		Video Overlay
[... 242 lines cut ...]
                   h264_profile (menu)   : min=0 max=4 default=4 value=4

Camera Controls

                  auto_exposure (menu)   : min=0 max=3 default=0 value=0
         exposure_time_absolute (int)    : min=1 max=10000 step=1 default=1000 value=1000
     exposure_dynamic_framerate (bool)   : default=0 value=0
             auto_exposure_bias (intmenu): min=0 max=24 default=12 value=12
      white_balance_auto_preset (menu)   : min=0 max=9 default=1 value=1
            image_stabilization (bool)   : default=0 value=0
                iso_sensitivity (intmenu): min=0 max=4 default=0 value=0
           iso_sensitivity_auto (menu)   : min=0 max=1 default=1 value=1
         exposure_metering_mode (menu)   : min=0 max=2 default=0 value=0
                     scene_mode (menu)   : min=0 max=13 default=0 value=0

JPEG Compression Controls

            compression_quality (int)    : min=1 max=100 step=1 default=30 value=30

cat /etc/apt/sources.list
cat /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ stretch main contrib non-free rpi
```

mpromonet, 2018-03-09:
Hi walterav1984 
Your problems seems different, I don't see any logs like descibe in the issue

> Cannot open device:/dev/video0 Operation not permitted

You can look to the captured frame using -vvv, probably the SPS/PPS are not read/detected.

Best Regards,
Michel.

walterav1984, 2018-03-10:
I meant to say that it does work now but it needed a restart? I was having this error:
```
VIDIOC_STREAMON: Operation not permitted
VIDIOC_DQBUF: Invalid argument
```
But now I realize this was @MrGlasspoole his error and not exactly from the OP @TechMasterJoe. I do use the newer sony noir camera's.

Gramler, 2018-04-19:
Had a similar situation

I forced uv4l to "video1"
sudo uv4l --driver raspicam -n video1 --framerate 30 --extension-presence=0

and then
cvlc v4l2c:///dev/video1:width=640:height=480:chroma=H264 --sout '#rtp{sdp=rtsp://:8554/}' --demux h264

That made it work

mpromonet, 2018-04-30:
I close the issue, that becomes far from the orgiinal subject.

closed 2018-04-30 by mpromonet