mpromonet/v4l2rtspserver #90: Error " install FILES given unknown argument "EXCLUDE_FROM_ALL" at cmake

TopperBG, 2018-02-13:
> -- CMAKE_TOOLCHAIN_FILE=
> -- CMAKE_C_COMPILER=/usr/bin/cc
> -- CMAKE_CXX_COMPILER=/usr/bin/c++
> -- CMAKE_FIND_ROOT_PATH=/build/v4l2rtspserver/armv7l
> -- CMAKE_STAGING_PREFIX=/build/v4l2rtspserver/armv7l
> -- Log4Cpp available
> -- SystemD available
> -- SystemD directory '/lib/systemd/system'
> CMake Error at CMakeLists.txt:118 (install):
>   install FILES given unknown argument "EXCLUDE_FROM_ALL".
> 
> 
> -- Configuring incomplete, errors occurred!
> 

Resolve by removing EXCLUDE_FROM_ALL in CMakeLists.txt regarding service file.

mpromonet, 2018-02-17:
Hi TopperBG,
I guess this is link to the cmake version, this seems supported since 3.0.
What version of cmake are you using ?
I add this option in order to include systemd script on DEB package but not on TGZ.
Best Regards,
Michel.

spaul13, 2018-02-19:
@mpromonet & @TopperBG  I am getting the same error and compilation stopped. 

How can I resolve this? I am using cmake version 3.0.2.

TopperBG, 2018-02-20:
@mpromonet 
```
root@NanoPi-NEO-Air:~# cmake --version
cmake version 3.5.1
```
@spaul13 

> Resolve by removing EXCLUDE_FROM_ALL in CMakeLists.txt regarding service file.

spaul13, 2018-02-20:
After removing that I am still getting this error. Can u ( @mpromonet and @TopperBG) please help me telling why am I getting this error? and how to resolve this?

running git submodule init && git submodule update && make -C v4l2wrapper CC=/usr/bin/cc PREFIX=/home/sibendu/Downloads/v4l2rtspserver-master/x86_64 && cp v4l2wrapper/libv4l2wrapper.a . && make -C v4l2wrapper clean 2>&1
fatal: Not a git repository (or any of the parent directories): .git
-- v4l2wrapper built in V4L2WRAPPER_LIBRARY=V4L2WRAPPER_LIBRARY-NOTFOUND
-- SystemD available
-- SystemD directory '/lib/systemd/system'
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
V4L2WRAPPER_LIBRARY
    linked by target "v4l2rtspserver-master" in directory /home/sibendu/Downloads/v4l2rtspserver-master

-- Configuring incomplete, errors occurred!

spaul13, 2018-02-20:
Right, I have updated my CMake and the CMake version is 3.11.0-rc1.
I have also deleted EXCLUDE_FROM_ALL . Still, I am getting the same error I just posted above.

@mpromonet and @TopperBG can anyone please tell me why I am getting this error?

TopperBG, 2018-02-20:
You can't download from git repo v4l2wrapper for any reason.
Try to git clone http://github.com/address-of-v4l2wrapper.git into v4l2rtspserver dir.

mpromonet, 2018-03-25:
Hi,
I removed the EXCLUDE_FROM_ALL and the tgz package... maybe I will try to find later another solution to build DEB & TGZ.
Then I close the issue.
Best Regards,
Michel.

closed 2018-03-25 by mpromonet