mpromonet/v4l2rtspserver #91: Running RTSP server on terminal after making it

spaul13, 2018-02-19:
I just want to run the RTSP server on my desktop. I have compiled the codes successfully. 

**Now, How to run the RTSP server code ??
it's giving me an error while I just run ./v4l2rtspserver
(error: ./v4l2rtspserver: No such file or directory)**

I am using the rtspclient from a mediaplayer  there I have to specify
rtsp://serverip/filepath

Do I need to give the port number also?

mpromonet, 2018-02-19:
Hi spaul13,

In order to run it you should run the exexecutable you build.
As it is named with the folder name, if you clone the repository in `foo` folder, the executable will be named `foo` and you should be able to run it using `./foo`.

By default the rtsp port is 554, when you are using rtsp://serverip/filepath, it means rtsp://serverip:554/filepath.
You can specify the RTSP server port using -P option, by default it use 8554 (see -h description)

Best Regards,
Michel.

spaul13, 2018-02-19:
I am getting some error during cmake (compilation) and compilation stopped. I think that's why executable is not built yet.

How can I resolve this? I am using cmake version 3.0.2.

**CMake Error at CMakeLists.txt:118 (install):
install FILES given unknown argument "EXCLUDE_FROM_ALL".
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
V4L2WRAPPER_LIBRARY
linked by target "v4l2rtspserver-master" in directory /home/sibendu/Downloads/v4l2rtspserver-master
-- Configuring incomplete, errors occurred**

spaul13, 2018-02-19:
@mpromonet  I am also attaching the CMake error log and output log. Can u please tell me why I am possibly getting the error?

[CMakeError.log](https://github.com/mpromonet/v4l2rtspserver/files/1738306/CMakeError.log)
[CMakeOutput.log](https://github.com/mpromonet/v4l2rtspserver/files/1738307/CMakeOutput.log)

mpromonet, 2018-03-25:
Hi spaul13,
From your log : 
```
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
..
/usr/bin/ld: cannot find -lpthreads
```
As it is, the program need to create threads... then either you remove the thread using only the live555 mainloop, or you need to install pthreads on your platform.
Best Regards,
Michel.

closed 2018-03-25 by mpromonet