mpromonet/v4l2rtspserver #97: Autostv4l2rtspserver as service

MarkusEakar, 2018-04-28:


mpromonet, 2018-04-30:
Hi MarkusEakar,
If you installed from the deb package, it should have installed v4l2rtspserver.service in systemd folder.
Then in order to enable the service at next boot, you should try: 

    sudo systemctl enable v4l2rtspserver
And to start it

    sudo systemctl start v4l2rtspserver

Best Regards,
Michel.

MarkusEakar, 2018-05-01:
Thank you for these very useful hints-I‘ll try them out soon! :-)
Perhaps this Information could be added to the project description as well?

Another question:
How can the enabled service be configured now?

MarkusEakar, 2018-05-03:
I tried again with v0.0.7 armv7 .deb package.

Installed it on Raspberry Pi 3 with up to date Raspian and NanoPi NEO with Armbian.

Unfortunately with both of them didn‘t have success enabling v4l2rtspserver.service - „Not found“. Should it have been lockten in /etc/systemd/system? Didn‘t find any file with that name after installation of the .deb.
Installation was tried both with „sudo gdebi install...“ and „sudo dpkg-i...“

Did I make some mistakes or could there be some mistake in installation routine of v0.0.7 .deb package?

mpromonet, 2018-05-04:
Hi MarkusEakar,
The systemd file is not included there is still problem detecting systemd and integrate the systemd file it in package.
You can create the systemd file manually in `/lib/systemd/system/v4l2rtspserver.service` with the content : 
```
[Unit]
Description=V4L2 RTSP server

[Service]
Type=simple
ExecStart=/usr/local/bin/v4l2rtspserver
```
Best Regards,
Michel.

MarkusEakar, 2018-05-14:
Got this working now on RPi3/Raspbian and NanoPi NEO/Armbian with your hints.

However had to add some additional lines for the service unit file:
`[Unit]
Description=V4L2 RTSP server
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=1
User=user
ExecStart=/home/user/startrtspserver.sh

[Install]
WantedBy=multi-user.target`

closed 2018-05-14 by MarkusEakar

IT-Experte, 2019-02-18:
I found the solution into this [issues](https://github.com/mpromonet/v4l2rtspserver/issues/83).

My Systemd service file v4l2rtspserver.service look like this:
```
[Unit]
Description=V4L2 RTSP server
After=network.target
 
[Service]
Type=simple
Environment=LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so
ExecStart=/usr/local/bin/v4l2rtspserver -F 20 -W 1280 -H 720 -P 8554 /dev/video0
User=pi
Group=pi

[Install]
WantedBy=multi-user.target
```