mpromonet/v4l2rtspserver #98: Write log with each connection

RogerHardiman, 2018-05-05:
Hi
I use the v4l2rtspserver in the Pi RPOS project (an ONVIF Server for the Pi)

I wondered if it is possible to modify the code to write a log message (or write to stdout) with each RTSP connection (either on the SETUP or on the PLAY).
What I'm interested in is seeing if a RTSP client connects in UDP, TCP or Multicast based on the ONVIF messages I return.

Thanks
Roger

RogerHardiman, 2018-05-06:
I had a look at live555. I had hoped there was something that could be overloaded via right now it looks like I need to look at some fprintf statements wrapped by a DEBUG #define in RTSPServer.cpp
I don't think there is an easy way to do it without making changes in the live555 library itself.

mpromonet, 2018-05-09:
HI Roger,

The live555 RTSPserver is overided in https://github.com/mpromonet/v4l2rtspserver/blob/master/inc/HTTPServer.h, in order to improve the HLS support and add MPEG DASH.
Maybe adding some log  in  : 
		
                RTSPServer::RTSPClientConnection* createNewClientConnection(int clientSocket, struct sockaddr_in clientAddr) 
		{
			return new HTTPClientConnection(*this, clientSocket, clientAddr);
		}
Otherwise you can overide the RTSPServer::RTSPClientConnection method called by DESCRIBE or SETUP.

Best Regards,
Michel.

RogerHardiman, 2018-05-09:
Thanks, I will take a look.
Roger

closed 2020-09-19 by mpromonet