mpromonet/v4l2rtspserver #100: Please, I can't make sound work. !!

sukhum, 2018-05-26:
I'm on RPI 3 with official camera module.

$  cat /proc/asound/cards
```
 0 [ALSA           ]: bcm2835_alsa - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB PnP Sound Device
                      C-Media Electronics Inc. USB PnP Sound Device at usb-3f980000.usb-1.2, full spe

```

$ cat /proc/asound/card1/stream0
```
C-Media Electronics Inc. USB PnP Sound Device at usb-3f980000.usb-1.2, full spe : USB Audio

Capture:
  Status: Stop
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 1
    Endpoint: 2 IN (ADAPTIVE)
    Rates: 48000, 44100
```

$ cat ~/.asoundrc
```
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
}
```

$ arecord -l
```
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
My microphone works with arecord to .wav file(s) , so I'm start with

$ ./v4l2rtspserver -h
```
./v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]
                  [-I interface] [-P RTSP port] [-p RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout] [-T] [-S[duration]]
                  [-r] [-w] [-s] [-f[format] [-W width] [-H height] [-F fps] [device] [device]
         -v               : verbose
         -vv              : very verbose
         -Q <length>      : Number of frame queue  (default 10)
         -O <output>      : Copy captured frame to a file or a V4L2 device
         RTSP/RTP options
         -I <addr>        : RTSP interface (default autodetect)
         -P <port>        : RTSP port (default 8554)
         -p <port>        : RTSP over HTTP port (default 0)
         -U <user>:<pass> : RTSP user and password
         -R <realm>       : use md5 password 'md5(<username>:<realm>:<password>')
         -u <url>         : unicast url (default unicast)
         -m <url>         : multicast url (default multicast)
         -M <addr>        : multicast group:port (default is random_address:20000)
         -c               : don't repeat config (default repeat config before IDR frame)
         -t <timeout>     : RTCP expiration timeout in seconds (default 65)
         -T               : send Transport Stream instead of elementary Stream
         -S[<duration>]   : enable HLS & MPEG-DASH with segment duration  in seconds (default 5)
         V4L2 options
         -r               : V4L2 capture using read interface (default use memory mapped buffers)
         -w               : V4L2 capture using write interface (default use memory mapped buffers)
         -s               : V4L2 capture using live555 mainloop (default use a reader thread)
         -f               : V4L2 capture using current capture format (-W,-H,-F are ignored)
         -f<format>       : V4L2 capture using format (-W,-H,-F are used)
         -W <width>       : V4L2 capture width (default 0)
         -H <height>      : V4L2 capture height (default 0)
         -F <fps>         : V4L2 capture framerate (default 25)
         -G <w>x<h>[x<f>] : V4L2 capture format (default 0x0x25)
         Devices :
         [V4L2 device][,ALSA device] : V4L2 capture device or/and ALSA capture device (default /dev/video0,/dev/video0)
```
** I notice that it don't have ALSA options as in README.md
```
ALSA options :
	 -A freq    : ALSA capture frequency and channel (default 44100)
	 -C channels: ALSA capture channels (default 2)
	 -a fmt     : ALSA capture audio format (default S16_LE)
```

then I run command

$ ./v4l2rtspserver -W 1280 -H 720 -F 25 -v  /dev/video0
```
2018-05-26 05:51:13,189 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
        level:INFO
2018-05-26 05:51:13,191 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:612
        Create V4L2 Source.../dev/video0

[NOTICE] src/V4l2Device.cpp:117
        driver:raspicam capabilities:5000001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video0:H264 size:1280x720
[NOTICE] src/V4l2Device.cpp:229
        fps:1/25
[NOTICE] src/V4l2Device.cpp:230
        nbBuffer:4
[NOTICE] src/V4l2MmapDevice.cpp:71
2018-05-26 05:51:13,455 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:635
        Create Source .../dev/video0
2018-05-26 05:51:13,455 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:165
        Play this stream using the URL "rtsp://192.168.1.10:8554/unicast"
2018-05-26 05:51:13,456 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
2018-05-26 05:51:13,457 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527313873 fps:1 bandwidth:0kbps
2018-05-26 05:51:13,457 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:38
        SPS size:14 bufSize:9
2018-05-26 05:51:13,458 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:39
        PPS size:5 bufSize:0
2018-05-26 05:51:13,493 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:40
        IDR size:6873 bufSize:0
2018-05-26 05:51:14,018 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527313874 fps:14 bandwidth:3103kbps
```
On client open VLC: rtsp://192.168.43.61:8554/unicast and video works perfectly, after that I run another command to add ALSA option

$ ./v4l2rtspserver -W 1280 -H 720 -F 25 -v  /dev/video0,hw:1,0
```
2018-05-26 05:56:39,120 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
        level:INFO
2018-05-26 05:56:39,122 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:612
        Create V4L2 Source.../dev/video0

[NOTICE] src/V4l2Device.cpp:117
        driver:raspicam capabilities:5000001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video0:H264 size:1280x720
[NOTICE] src/V4l2Device.cpp:229
        fps:1/25
[NOTICE] src/V4l2Device.cpp:230
        nbBuffer:4
[NOTICE] src/V4l2MmapDevice.cpp:71
2018-05-26 05:56:39,375 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:635
        Create Source .../dev/video0
2018-05-26 05:56:39,375 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:165
        Play this stream using the URL "rtsp://192.168.1.10:8554/unicast"
2018-05-26 05:56:39,375 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
2018-05-26 05:56:39,376 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527314199 fps:1 bandwidth:0kbps
2018-05-26 05:56:39,376 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:38
        SPS size:14 bufSize:9
2018-05-26 05:56:39,376 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:39
        PPS size:5 bufSize:0
2018-05-26 05:56:39,425 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:40
        IDR size:6794 bufSize:0
2018-05-26 05:56:40,036 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527314200 fps:16 bandwidth:4000kbps
2018-05-26 05:56:41,038 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527314201 fps:25 bandwidth:8678kbps
2018-05-26 05:56:41,826 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:38
        SPS size:14 bufSize:9
2018-05-26 05:56:41,827 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:39
        PPS size:5 bufSize:0
2018-05-26 05:56:41,844 [INFO  ] - /home/pi/v4l2rtspserver/src/H264_V4l2DeviceSource.cpp:40
        IDR size:96148 bufSize:0
2018-05-26 05:56:42,035 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        intv_sec:1527314202 fps:26 bandwidth:8245kbps
```
on client with VLC, It's have only video but no sound (in VLC press Ctrl+j will show codec information). So I try

$ ./v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -C 1 -a S16_LE -A 48000
$ ./v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -C 1 
$ ./v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -a S16_LE 
$ ./v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -a S16_BE
$ ./v4l2rtspserver -W 1920 -H 1080 /dev/video0,hw:1,0 -a S16_LE -A 44100
..... 
.....
.....
switch all alsa option 
.....
.....
....  
and result is same as v4l2rtspserver -h.

I also put alsa option before /dev/video0,hw:1,0 but result was same.

Can you help me how to use microphone.
Thanks

** Sorry for my poor English.

mpromonet, 2018-05-26:
Hi sukhum,

In the log there is no log for ALSA, did you build with ALSA support.
This is choose by cmake depending if it find ALSA support.

Best Regards,
Michel.

sukhum, 2018-05-26:
Hi Michel, 

How to build with ALSA support . What I did was
$ cmake .
$ make
$ sudo make install 

Best Regards,
Sukhum

mpromonet, 2018-05-26:
Hi Sukhum,
You should try : 

      sudo apt install libasound2-dev

I think supporting alaw or mulaw should be quite easy if the device support capture in this format, it just need to set the RTP payload to audio/PCMU or audio/PCMA

Best Regards,
Michel.

sukhum, 2018-05-26:
Thanks Michel.

Best Regards,
Sukhum

closed 2018-05-28 by mpromonet

jamesee, 2019-01-29:
Hi Michel

I am facing the same issue as Sukhum, there is no sound for v4l2rtspserver.

I verified that the v4lsrtspserver is properly compiled with ALSA options using the -h option.

I executed the following and it seems to be proper:

v4l2rtspserver -H 720 -W 1280 -F 25 /dev/video0,default &
[1] 3971
$ 2019-01-30 05:03:15,248 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:594
	Create V4L2 Source.../dev/video0
[NOTICE] src/V4l2Device.cpp:117
	driver:uvcvideo capabilities:84200001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
	/dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:123
	/dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
	/dev/video0:MJPG size:1280x720
[ERROR] src/V4l2Device.cpp:195
	Cannot set pixelformat to:H264 format is:MJPG
[NOTICE] src/V4l2Device.cpp:229
	fps:1/30
[NOTICE] src/V4l2Device.cpp:230
	nbBuffer:0
[NOTICE] src/V4l2MmapDevice.cpp:71
2019-01-30 05:03:15,339 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:617
	Create Source .../dev/video0
2019-01-30 05:03:15,339 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
2019-01-30 05:03:15,497 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:647
	Create ALSA Source...default
2019-01-30 05:03:15,497 [NOTICE] - /home/pi/v4l2rtspserver/src/ALSACapture.cpp:49
	Open ALSA device: "default"
2019-01-30 05:03:15,504 [NOTICE] - /home/pi/v4l2rtspserver/src/ALSACapture.cpp:117
	set sample format device: default to:2 ok
2019-01-30 05:03:15,511 [NOTICE] - /home/pi/v4l2rtspserver/src/ALSACapture.cpp:104
	ALSA device: "default" buffer_size:1048576 period_size:1024 rate:44100
2019-01-30 05:03:15,513 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.1.112:8554/unicast"
2019-01-30 05:03:15,513 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread

I played the rtsp stream using omxplayer:

$ omxplayer -o hdmi rtsp://192.168.1.112:8554/unicast
Video codec omx-mjpeg width 1280 height 720 profile -99 fps 29.916666
Audio codec pcm_s16be channels 2 samplerate 44100 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@29.92 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

Seems ok, but no sound :(

Please help. Where did I go wrong?

Rdgs,
James