mpromonet/v4l2rtspserver #102: make: *** [all] Error 2

Siggemada, 2018-07-10:
orangepiplus2e:~/v4l2rtspserver$ sudo make
Scanning dependencies of target v4l2rtspserver
[  9%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 18%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o
/home/hans/v4l2rtspserver/src/DeviceSource.cpp:17:20: fatal error: logger.h: No such file or directory
 #include "logger.h"
                    ^
compilation terminated.
CMakeFiles/v4l2rtspserver.dir/build.make:86: recipe for target 'CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o' failed
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

What am I doing wrong?

mpromonet, 2018-07-11:
Hi Siggemada,
The file is in submodule, probably something fails when you run `cmake`, you can also run `make VERBOSE=1` to check the include path.
Best Regards,
Michel.

Siggemada, 2018-07-11:
Hi, thanks,

I do not have 
/v4l2rtspserver/v4l2wrapper/inc

How can I add it?

mpromonet, 2018-07-11:
Hi Siggemada,
This is created by the command `git submodule init && git submodule update` in CMakeLists.txt.
If this is missing, you should remove all the cmake stuff to understand the error executing cmake.
Best Regards,
Michel.

Siggemada, 2018-07-11:
Hi and thank you again!

This is the output of cmake .

-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0") 
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- CMAKE_FIND_ROOT_PATH=/home/aaa/v4l2rtspserver/armv7l
-- CMAKE_STAGING_PREFIX=/home/aaa/v4l2rtspserver/armv7l
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Log4Cpp available
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- SystemD available
-- SystemD directory '/lib/systemd/system'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aaa/v4l2rtspserver

I have cloned libv4l2cpp and run make and make install on that. Looking at CMakeLists, it looks like I should not have done that?

Thanks again!

Siggemada, 2018-07-11:
Ok, so I have this in: /usr/include/libv4l2cpp/logger.h

Removing that, I get:
orangepiplus2e:~/v4l2rtspserver$ cmake .
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0") 
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- CMAKE_FIND_ROOT_PATH=/home/hans/v4l2rtspserver/armv7l
-- CMAKE_STAGING_PREFIX=/home/hans/v4l2rtspserver/armv7l
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Log4Cpp available
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- SystemD available
-- SystemD directory '/lib/systemd/system'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aaa/v4l2rtspserver

 Still no /v4l2rtspserver/v4l2wrapper/inc

Siggemada, 2018-07-11:
Success!

Ok, so I added /v4l2rtspserver/v4l2wrapper/inc manually;

orangepiplus2e:~/v4l2rtspserver$ make
[  9%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o
[ 18%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 27%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/HTTPServer.cpp.o
[ 36%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/MJPEGVideoSource.cpp.o
[ 45%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o
[ 54%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/SegmentServerMediaSubsession.cpp.o
[ 63%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o
[ 72%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
[ 81%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o
[ 90%] Linking CXX executable v4l2rtspserver
[100%] Built target v4l2rtspserver

closed 2018-07-11 by Siggemada