mpromonet/v4l2rtspserver #103: V4L2DeviceSource::doStopGettingFrames

jpsabo, 2018-07-22:
On my home network, I have 5 Raspberry Pi Zero W's running the lite version of 'Stretch' and v4l2rtspserver that wirelessly stream H264 video to a MotionEyeOS server which is connected by ethernet. Overall, this approach works well and the Zero's run for days of uptime and streaming without issue. However, occasionally, the MotionEye server loses connection to a Zero for a time (displaying a gray screen with the note that the video connection was lost), but restarts the connection after some time. In the Zero log I see the following (repeated many times in the log) at the time the video to the MotionEye server is lost: 

Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]: [NOTICE] /home/pi/live/v4l2rtspserver/src/DeviceSource.cpp:131
Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]:         V4L2DeviceSource::doStopGettingFrames
Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]: [NOTICE] /home/pi/live/v4l2rtspserver/src/DeviceSource.cpp:131
Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]:         V4L2DeviceSource::doStopGettingFrames
Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]: [NOTICE] /home/pi/live/v4l2rtspserver/src/DeviceSource.cpp:131
Jul 22 17:17:15 MEye-Livingroom v4l2rtspserver[452]:         V4L2DeviceSource::doStopGettingFrames

Is there a configuration option on the Zero (within v4l2rtspserver?) that should be set to resolve the loss of connection to the MotionEye server? 

Thanks in advance for help with this question.

aerik, 2018-12-06:
I'm building the same setup.  did you ever figure anything out?

jpsabo, 2018-12-06:
No. Still the same issue as I described above. Although the root cause might be Wi-Fi connectivity to my home network. Cameras close to the Wi-Fi router rarely lose connection, while those that are further away tend to disconnect more often.

closed 2020-02-14 by mpromonet