mpromonet/v4l2rtspserver #104: Segmentation fault

Siggemada, 2018-07-26:
On ARMBIAN 5.52.180717 nightly Debian GNU/Linux 9 (stretch) 4.17.7-sunxi64:

Setup:
v4l2-ctl -d /dev/video0 -vwidth=2304,height=1536,pixelformat=YUYV -p 2
sudo modprobe v4l2loopback video_nr=10,11,12
v4l2compress_h264 -W2304 -H1536 -F2 /dev/video0 /dev/video10
./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10

@orangepipc2:~/v4l2rtspserver$ ./v4l2rtspserver -W2304 -H1536 -F2 -vv /dev/video10
2018-07-26 17:24:37,961 [INFO  ] - /home/aaa/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
        level:DEBUG
2018-07-26 17:24:37,962 [NOTICE] - /home/aaa/v4l2rtspserver/src/main.cpp:622
        Create V4L2 Source.../dev/video10
2018-07-26 17:24:37,962 [NOTICE] - src/V4l2Device.cpp:117
        driver:v4l2 loopback capabilities:85208003 mandatory:4000001
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video10 support output
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video10 support capture
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video10 support read/write
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video10 support streaming
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:147
        /dev/video10:H264 size:2304x1536
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:229
        fps:1/2
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2Device.cpp:230
        nbBuffer:8
2018-07-26 17:24:37,963 [NOTICE] - src/V4l2MmapDevice.cpp:71
        Device /dev/video10 nb buffer:8
2018-07-26 17:24:37,963 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:0 size:14155776
2018-07-26 17:24:37,964 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:1 size:14155776
2018-07-26 17:24:37,966 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:2 size:14155776
2018-07-26 17:24:37,967 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:3 size:14155776
2018-07-26 17:24:37,969 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:4 size:14155776
2018-07-26 17:24:37,970 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:5 size:14155776
2018-07-26 17:24:37,971 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:6 size:14155776
2018-07-26 17:24:37,973 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:7 size:14155776
2018-07-26 17:24:37,974 [NOTICE] - /home/aaa/v4l2rtspserver/src/main.cpp:645
        Create Source .../dev/video10
2018-07-26 17:24:37,975 [NOTICE] - /home/aaa/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
Segmentation fault

Siggemada, 2018-07-26:
Hi, tried to do a backtrace with gdb in case it is useful. 

Type "apropos word" to search for commands related to "word"...
Reading symbols from v4l2rtspserver...done.
[New LWP 1903]
[New LWP 1902]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Core was generated by `./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000aaaae92ac544 in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()
[Current thread is 1 (Thread 0xffff8731f1e0 (LWP 1903))]

Thread 2 (Thread 0xffff8df22000 (LWP 1902)):
#0  0x0000ffff8e4f15cc in ?? () from /lib/ld-linux-aarch64.so.1
#1  0x0000ffff8e4f2060 in ?? () from /lib/ld-linux-aarch64.so.1
#2  0x0000ffff8e4f5ed8 in ?? () from /lib/ld-linux-aarch64.so.1
#3  0x0000ffff8e4fc6c0 in ?? () from /lib/ld-linux-aarch64.so.1
#4  0x0000ffff8e3c653c in RTSPServer::rtspURL(ServerMediaSession const*, int) const ()
   from /usr/lib/aarch64-linux-gnu/libliveMedia.so.57
#5  0x0000aaaae92a7210 in addSession(RTSPServer*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::list<ServerMediaSubsession*, std::allocator<ServerMediaSubsession*> > const&) ()
#6  0x0000aaaae92a93a4 in main ()

Thread 1 (Thread 0xffff8731f1e0 (LWP 1903)):
#0  0x0000aaaae92ac544 in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()
#1  0x0000aaaae929d194 in V4L2DeviceSource::getNextFrame() ()
#2  0x0000aaaae929c80c in V4L2DeviceSource::thread() ()
#3  0x0000aaaae929de18 in V4L2DeviceSource::threadStub(void*) ()
#4  0x0000ffff8e4c40a0 in start_thread () from /lib/aarch64-linux-gnu/libpthread.so.0
#5  0x0000ffff8e010eac in ?? () from /lib/aarch64-linux-gnu/libc.so.6

Thanks, grate program works rock solid up to 1080p.

mpromonet, 2018-09-27:
Hi Siggemada,

From the stack I don't understand the problem, it is like the VideoSource is broken.
Maybe some allocation fails ?

Did you make some test using the read interface adding `-r` option ?
Using `strace`, maybe you could trap what fails ?

Best Regards,
Michel.

Siggemada, 2018-10-01:
Hi,

I did try the -r option but got the same result.

It is a Logitech c930e and I am trying to read 2304 by 1536 pixels. In raw mode.

Unfortunately, the cam and hub got fried by lightning. So at the moment, I cant test.

Iĺl get back here when I have the equipment up and running again.

Thanks

closed 2019-04-10 by mpromonet