mpromonet/v4l2rtspserver #108: How to change from AVC (H.264) to HEVC (H.265)?

wicadmin, 2018-09-09:
How to change from AVC (H.264) to HEVC (H.265)?

wicadmin, 2018-09-09:
Michel,
Thanks for the quick response. I am getting this error including that:

```
Sep  9 12:33:28 raspberrypi systemd[1]: v4l2rtspserver.service: Service hold-off time over, scheduling restart.
Sep  9 12:33:28 raspberrypi systemd[1]: Stopped V4L2 RTSP server.
Sep  9 12:33:28 raspberrypi systemd[1]: Started V4L2 RTSP server.
Sep  9 12:33:29 raspberrypi startrtspserver.sh[3268]: 2018-09-09 12:33:29,089 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:623
Sep  9 12:33:29 raspberrypi startrtspserver.sh[3268]: #011Create V4L2 Source.../dev/video0
Sep  9 12:33:29 raspberrypi startrtspserver.sh[3268]: VIDIOC_STREAMOFF: Bad file descriptor
Sep  9 12:33:29 raspberrypi startrtspserver.sh[3268]: VIDIOC_REQBUFS: Bad file descriptor
```

wicadmin, 2018-09-09:
It doesn't appear the v1.3 it is supported on my device :-(

```
v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YU12'
	Name        : Planar YUV 4:2:0

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'RGB3'
	Name        : 24-bit RGB 8-8-8

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'JPEG' (compressed)
	Name        : JFIF JPEG

	Index       : 4
	Type        : Video Capture
	Pixel Format: 'H264' (compressed)
	Name        : H.264

	Index       : 5
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 6
	Type        : Video Capture
	Pixel Format: 'YVYU'
	Name        : YVYU 4:2:2

	Index       : 7
	Type        : Video Capture
	Pixel Format: 'VYUY'
	Name        : VYUY 4:2:2

	Index       : 8
	Type        : Video Capture
	Pixel Format: 'UYVY'
	Name        : UYVY 4:2:2

	Index       : 9
	Type        : Video Capture
	Pixel Format: 'NV12'
	Name        : Y/CbCr 4:2:0

	Index       : 10
	Type        : Video Capture
	Pixel Format: 'BGR3'
	Name        : 24-bit BGR 8-8-8

	Index       : 11
	Type        : Video Capture
	Pixel Format: 'YV12'
	Name        : Planar YVU 4:2:0

	Index       : 12
	Type        : Video Capture
	Pixel Format: 'NV21'
	Name        : Y/CrCb 4:2:0

	Index       : 13
	Type        : Video Capture
	Pixel Format: 'BGR4'
	Name        : 32-bit BGRA/X 8-8-8-8
```

mpromonet, 2018-09-09:
Hi wicadmin,
Your camera doesnot support HEVC, then it is better to use H264 that is hardware compressed than compress in HEVC using software.
Best Regards,
Michel.

closed 2018-09-09 by wicadmin