mpromonet/v4l2rtspserver #110: Onvif S audio compatibility

JacekDob, 2018-09-13:
Hi,

can you advice how to make it compatible with Onvif S profile for webcam, which should has on of following audio codecs?
G.711 [ITU-T G.711]
G.726 [ITU-T G.726]
AAC [ISO 14496-3]
?

mpromonet, 2018-09-27:
Hi JacekDob,
The audio supported MULAW & ALAW belongs to G711 format, then if you have a ALSA device that support one of them you should be able to provide them.
Basically the approach of this project is to capture and expose the stream using an RTSP server.
Then if you like to support others format, you should need a decompressor that provide an ALSA audio stream.
Best Regards,
Michel.

JacekDob, 2018-10-02:
Hi Michel,

Thank you for your answer.
I know how to compress audio ALSA stream using vlc or ffmpeg.
How to provide such a stream to v4l2rtspserver?
Or maybe you know how to make virtual ALSA with other format?

Best regards,

Jacek

closed 2019-04-10 by mpromonet