mpromonet/v4l2rtspserver #112: Corrupted HSL stream

ClaudiuCeia, 2018-09-29:
Hey,

v4l2rtpserver is awesome, exactly what I was looking for. However, although the rtsp stream is flawless (tested with VLC), I can't seem to get the HSL stream to run properly, the image is always corrupted (tested with Chrome and Firefox native extensions/addons, as well as the hsl.js player). I've tried different resolutions, frame-rates, etc, but not luck so far.

I ran the stream both with verbose and very verbose output but don't see anything that stands out. The camera I use is a Logitech C920, if that is useful in any way. How could I debug this? Any suggestions? 

I can attach a preview file if you want. The problem seems similar to #81, but `-c` didn't help me.

mpromonet, 2018-10-17:
HI Claudiu,
Just to be sure, did you add the `-S` option to enable HLS ?
Best Regards,
Michel.

closed 2018-12-16 by mpromonet

roger-, 2020-04-27:
I'm having the same issue with -S and HLS.