mpromonet/v4l2rtspserver #113: Got the wrong video stream

mo10, 2018-10-08:
Camera: Binocular camera with single device number('MJPG' compressed)
Hardware: Raspberry Pi 3B (armv7l)
Software version: release v0.0.8 7ead761

I tried using v4l2rtspserver on Raspberry Pi 3B,but I got a full green video stream (except for a small image in the upper left corner)
[image]
It is normally like this:
[image]

**Command**:`v4l2rtspServer -f MJPG -s -vv -W 640 -H 240 /dev/video0`
**Log file**:[stream.log](https://github.com/mpromonet/v4l2rtspserver/files/2454928/stream.log)

```
# v4l2-ctl --list-formats

ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG
# lsusb

Bus 001 Device 007: ID 05a3:9750 ARC International
```

mpromonet, 2018-10-08:
Hi mo10,
Just a quick remark, when you specify an argument to -f, you should not add a space (because -f could have an argument or not), then you should use `-fMJPEG` instead of `-f MJPEG`.
Anyway this just open 2 devices (one MJPEG that doesnot exists and the /dev/video0).
From the log there is something strange : 

[NOTICE] /root/v4l2rtspServer/src/MJPEGVideoSource.cpp:57
	Quantization table idx:0 precision:0 size:64 total size:64
[NOTICE] /root/v4l2rtspServer/src/MJPEGVideoSource.cpp:57
	Quantization table idx:1 precision:16 size:64 total size:128

Maybe the precision decoding is wrong... need to check.

Best Regards,
Michel.

mo10, 2018-10-09:
Thank you for your reply mpromonet,
I removed the space,but still the same problem.Maybe it is a problem with reading frames?I can get the correct stream using [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer)
`v4l2rtspServer -fMJPG -W640 -H240 -s -vv /dev/video0`
log file:[stream2.log](https://github.com/mpromonet/v4l2rtspserver/files/2458805/stream2.log)

Thanks!

reasonsolo, 2018-10-09:
Same problem here.

I got ffmpeg error message via VLC like this
```
Created new TCP socket 24 for connection
[mjpeg @ 0x7f3424058300] error dc
[mjpeg @ 0x7f3424058300] error y=0 x=41
[mjpeg @ 0x7f3424058300] error dc
[mjpeg @ 0x7f3424058300] error y=1 x=0
[mjpeg @ 0x7f3424058300] error dc
[mjpeg @ 0x7f3424058300] error y=1 x=0
[mjpeg @ 0x7f3424058300] error dc
[mjpeg @ 0x7f3424058300] error y=1 x=0
[mjpeg @ 0x7f3424058300] error dc
[mjpeg @ 0x7f3424058300] error y=1 x=0
[mjpeg @ 0x7f3424058300] error dc
```

v4l2rtspserver log
```
./v4l2rtspserver -fMJPG -s
log level:500

[NOTICE] /home/zlz/android/v4l2rtspserver/src/main.cpp:623
	Create V4L2 Source.../dev/video0
[NOTICE] src/V4l2Device.cpp:117
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
	/dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:123
	/dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
	/dev/video0:MJPG size:848x480
[NOTICE] src/V4l2Device.cpp:229
	fps:1/30
[NOTICE] src/V4l2Device.cpp:230
	nbBuffer:0
[NOTICE] src/V4l2MmapDevice.cpp:71
	Device /dev/video0 nb buffer:10
[NOTICE] /home/zlz/android/v4l2rtspserver/src/main.cpp:646
	Create Source .../dev/video0
[NOTICE] /home/zlz/android/v4l2rtspserver/src/main.cpp:676
	Create ALSA Source.../dev/video0
[NOTICE] /home/zlz/android/v4l2rtspserver/src/ALSACapture.cpp:49
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM /dev/video0
	Open ALSA device: "/dev/video0"
[ERROR] /home/zlz/android/v4l2rtspserver/src/ALSACapture.cpp:56
	cannot open audio device: /dev/video0 error:No such file or directory
[NOTICE] /home/zlz/android/v4l2rtspserver/src/ALSACapture.cpp:104
	ALSA device: "/dev/video0" buffer_size:0 period_size:0 rate:44100
[NOTICE] /home/zlz/android/v4l2rtspserver/src/main.cpp:165
RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
	Play this stream using the URL "rtsp://10.1.100.136:8554/unicast"
[NOTICE] /home/zlz/android/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
	Quantization table idx:0 precision:0 size:64 total size:64
[NOTICE] /home/zlz/android/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
	Quantization table idx:1 precision:16 size:64 total size:128
[NOTICE] /home/zlz/android/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/zlz/android/v4l2rtspserver/src/DeviceSource.cpp:131
```
The framed data seems broken. I tried to run v4l2rtspserver on both my x86 laptop and armv8 android device and got same error. Any suggestion?

Thanks.

mpromonet, 2018-10-09:
Hi mo10,
mjpeg-streamer provide multipart JPEG over http, this is a different format from RTP video/JPEG describe in [rfc2435](https://tools.ietf.org/html/rfc2435).
A possibility to provide multipart JPEG could be to implement it using the live555 tcp server, but this is quite simple, then till now I only extent the HLS and add a MPEG-DASH support. I don't think I will implement a http multipart, if like to use this format, better to use project like mjpeg-streamer.
The implementation for JPEG stream is an extension of live555 that need some debug as you noticed.
Best Regards,
Michel.

mo10, 2018-10-10:
Hi mpromonet,
I am looking for a high resolution, high frame rate and low latency streaming solution.RTSP seems to be the best choice.But I observed that RTSP has a very high latency under VLC and FFPlay,Not even as good as mjpg-streamer.Is there a problem with the performance of Live555 or ffmpeg?
Thanks!

mpromonet, 2018-10-21:
Hi mo10,

This is always a trade-off multipart-JPEG is over TCP and RTP could use UDP (it could also use TCP). 
In case of congestion depending whether you prefer latency (TCP) or lossing image (UDP).

vlc use a large buffer for reordering frames, if you are on a LAN you could decrease --network-caching, but I don't think ffplay have such buffering.
You can reduce the buffering using option `-Q`. By default the FIFO contains 10 frames, you can try to reduce it.

I made a fix parsing the quantification buffer, maybe this will give a better result.

Best Regards,

Michel.

mpromonet, 2019-10-28:
@mo10 

Recntly a fix was done when JPEG contains DRI frame (for #140).
Did you still have the same problem after the commit https://github.com/mpromonet/v4l2rtspserver/commit/05b52703ecc7b3a9ab4328d16364ebcbb0ded079 ?

Best Regards,
Michel

mo10, 2019-11-17:
@mpromonet 

I found rpi out of the corner of the box. Use the latest version https://github.com/mpromonet/v4l2rtspserver/commit/036d4230d39373727c10eea8b50146f530a7dd41.
The problem doesn't seem to be eliminated, I still get the damaged image. What options should I include?
Log:[output.log](https://github.com/mpromonet/v4l2rtspserver/files/3855522/output.log)

` v4l2rtspserver  -fMJPG -W640 -H240 -s -vv /dev/video0`

[image]

mpromonet, 2019-11-17:
Hi @mo10,

The fix of DRI frame is working, we saw in the log : 
`	DRI restartInterval:20`

Then the JPEG image of your camera persing has another problem.
The image is about 72k, this is probably more than the corner box.
Maybe reading the log of VLC can give an indication of what is missing ?

Best Regards,
Michel.

mo10, 2019-11-19:
Some error from VLC: 
```
[mjpeg @ 06521380] error y=0 x=20
[mjpeg @ 06521380] error dc
[mjpeg @ 06521380] error y=0 x=20
[mjpeg @ 06521380] error dc
[mjpeg @ 06521380] error y=0 x=20
[mjpeg @ 06521380] error dc
[mjpeg @ 06521380] error y=0 x=20
[mjpeg @ 06521380] error dc
......
```

mpromonet, 2019-11-19:
This is ffmpeg logs, you may got more informations using ffplay enabling debug with `-loglevel debug`

mo10, 2019-11-19:
`ffplay.exe rtsp://192.168.50.77:8554/unicast -loglevel
debug`
[output.log](https://github.com/mpromonet/v4l2rtspserver/files/3864150/output.log)

mpromonet, 2020-04-12:
Hi,

The commit https://github.com/mpromonet/v4l2rtspserver/commit/c4ae12e70801a8deb9f199f09d2f49463965031e may help you ?

Best Regards,
Michel.

mpromonet, 2020-09-19:
This isssue is old and some improvment was done in the JPEG streamings, so I close it.
In case problems still occurs with some streams open a new one with information of the JPEG datas.

closed 2020-09-19 by mpromonet