mpromonet/v4l2rtspserver #114: v4l2rtspserver and raspidisp

gounthar, 2018-11-07:
Hi,

I have two Raspberry Pi 3B+ running Stretch on hand. On one of them, I have installed v4l2rtspserver and a CSI camera, and I have a rtsp stream that works wonder.
I would love to do almost the same for the second one, except it has no camera. I am trying to use raspidisp to create a rtsp stream from the HDMI output.
Is it possible?

```bash
pi@raspberrypi:~ $ sudo service uv4l_raspidisp status
● uv4l_raspidisp.service - UV4L Raspicam
   Loaded: loaded (/etc/systemd/system/uv4l_raspidisp.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-07 17:48:01 CET; 13min ago
 Main PID: 328 (uv4l)
   CGroup: /system.slice/uv4l_raspidisp.service
           └─328 /usr/bin/uv4l -f -k --sched-fifo --mem-lock --config-file=/etc/uv4l/uv4l-raspidisp.conf --driver raspidisp --driver-config-file=/etc/uv4l/uv4l-raspidisp.conf --server-option=-

nov. 07 17:48:01 raspberrypi uv4l[328]: <notice> [core] Loading driver 'raspidisp' from external plug-in's...
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [driver] RaspiDisp Video4Linux2 Driver built Oct  7 2018
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [core] Device detected!
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [core] Trying to load the the Streaming Server plug-in...
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [server] HTTP/HTTPS Streaming & WebRTC Signalling Server v1.1.118 built on Oct  7 2018
nov. 07 17:48:02 raspberrypi uv4l[328]: <warning> [server] SSL is not enabled for the Streaming Server. Using unsecure HTTP.
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [core] Streaming Server loaded!
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [core] Registering device node /dev/video1
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [driver] input display #0 res.: 1920x1080 30fps, crop. window: 0 0 1920 1080, output res.: 1920x1080
nov. 07 17:48:02 raspberrypi uv4l[328]: <notice> [server] Web Streaming Server listening on port 9080
```

I guess I installed too many things, as I don't need the web streaming server... Anyway.

But I have this output:

```bash
./v4l2rtspserver -I 0.0.0.0 -P 2222 -v /dev/video1,hw:0,0
log level:600

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:623
        Create V4L2 Source.../dev/video1
[NOTICE] src/V4l2Device.cpp:117
        driver:raspidisp capabilities:5000001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video1 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video1 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video1 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video1:YU12 size:1920x1080
[ERROR] src/V4l2Device.cpp:195
        Cannot set pixelformat to:H264 format is:YU12
[ERROR] src/V4l2Device.cpp:195
        Cannot set pixelformat to:MJPG format is:YU12
[ERROR] src/V4l2Device.cpp:67
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
        Cannot init device:/dev/video1
```

So I tried this

```bash
v4l2-ctl -d /dev/video1 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YU12'
        Name        : YUV420
```

I have read [https://github.com/mpromonet/v4l2rtspserver](there) that I could use v4l2compress in order to transform to h264:

```bash
v4l2compress_h264 /dev/video0 /dev/video10 &
```
But the compilation of this project doesn't give me v4l2compress_h264.
Thanks.

mpromonet, 2018-11-07:
Hi Bruno,

v4l2compress_h264 is part of project https://github.com/mpromonet/v4l2tools
I will update the readme.

Best Regards,

Michel.

closed 2018-11-07 by mpromonet

gounthar, 2018-11-07:
Thanks a lot, Michel.
I had already compiled/installed v4l2tools, but this binary is not available.
I will open another issue.

mpromonet, 2018-11-07:
This is not link to this project, but v4l2tools, you can get x86_64 executables in https://circleci.com/gh/mpromonet/v4l2tools/30#artifacts/containers/0