mpromonet/v4l2rtspserver #115: VIDIOC_DQBUF: Resource temporarily unavailable

j1elo, 2018-11-16:
## Environment

```
# lsb_release -d
Description:	Linux Mint 18.3 Sylvia
# (This is same as Ubuntu 16.04 "Xenial")

# docker --version
Docker version 18.09.0, build 4d60db4

# gst-launch-1.0 --version
gst-launch-1.0 version 1.8.3
GStreamer 1.8.3
```

## Reproduction steps:

**HOST TERMINAL 1**

```
# In case GStreamer is not yet installed
sudo apt-get update
sudo apt-get install \
  gstreamer1.0-{tools,libav} \
  gstreamer1.0-plugins-{base,good,bad,ugly}

# Ensure the module is not already loaded, and reload
modprobe --remove v4l2loopback
modprobe v4l2loopback video_nr=5,6,7

# Feed the loopback device 5 with some valid stream
export GST_DEBUG=3,v4l2:4,v4l2sink:4
gst-launch-1.0 \
  filesrc location=VIDEO_H264.mp4 ! decodebin \
  ! queue ! videoscale ! videorate ! videoconvert \
  ! "video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, framerate=(fraction)15/1" \
  ! tee ! v4l2sink device=/dev/video5
```

**HOST TERMINAL 2**

```
docker run \
    --device=/dev/video5 --device=/dev/video6 --device=/dev/video7 \
    -p 8554:8554 -it --entrypoint=/bin/bash mpromonet/v4l2rtspserver
```

**CONTAINER**

```
# [1]: FAILS
# /v4l2rtspserver/v4l2tools/v4l2convert_yuv -o YUYV /dev/video5 /dev/video6 &
# > VIDIOC_DQBUF: Resource temporarily unavailable

# [2]: WORKS 
/v4l2rtspserver/v4l2tools/v4l2compress_h264 -r /dev/video5 /dev/video7 &
[...]
2018-11-16 14:58:25,448 [NOTICE] - src/v4l2compress_h264.cpp:185
	Start Compressing /dev/video5 to /dev/video7

# [3]: FAILS
/v4l2rtspserver/v4l2rtspserver /dev/video7
# > VIDIOC_DQBUF: Resource temporarily unavailable
```

**HOST TERMINAL 3**

```
gst-launch-1.0 \
  rtspsrc location='rtsp://127.0.0.1:8554/unicast' \
    ! rtph264depay ! decodebin ! autovideosink
```

Couldn't make it work after several hours of trying different combinations of parameters and commands. I'm using `v4l2loopback` so I don't need an actual webcam in my machine.

By running the provided commands in the given order, these are the errors that appear:

In **[1]**, it's simply not possible to convert the input video. Doesn't matter if capture mode is changed between `(none)`, `-r`, or `-w`, the output is always the same:
`Resource temporarily unavailable`.

However, the output format of the *gst-launch* command seems to be already compatible with *v4l2compress_h264*, so actually it's possible to launch it (**[2]**) directly, without any previous colorspace conversion.

Lastly, launching the actual RTSP server in **[3]** also fails like in **[1]**. Doesn't matter if capture mode is changed between `(none)`, `-r`, or `-w`, the output is always the same:
`Resource temporarily unavailable`.

j1elo, 2018-11-16:
I've also tried not using the Docker images; just building the source code of both the server and the tools and running them directly on my machine; the results are the same.

OrHirshfeldFFR, 2019-08-29:
I got a similar problem with Intel Realsense d435 on Ubuntu and i ran the following commands:

sudo rmmod uvcvideo
sudo modprobe uvcvideo nodrop=1 timeout=5000

it worked like a charm

thanks to this thread
https://www.raspberrypi.org/forums/viewtopic.php?t=35689

closed 2020-04-12 by mpromonet