mpromonet/v4l2rtspserver #116: Compiling with androideabi toolchain

kydoaod, 2018-11-20:
I am trying to compile this project with androideabi toolchain that I have in my ubuntu since I need it to make it work with android with this command:

cmake -D CMAKE_C_COMPILER=arm-linux-androideabi-gcc -D CMAKE_CXX_COMPILER=arm-linux-androideabi-g++ -D V4L2WRAPPER_LIBRARY=./v4l2wrapper . && make

But I am getting errors when it is building in cmake object files:

[  9%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o
In file included from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/MediaSink.hh:25:0,
                 from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/RTPSink.hh:25,
                 from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/MultiFramedRTPSink.hh:26,
                 from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/AudioRTPSink.hh:25,
                 from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/MPEG1or2AudioRTPSink.hh:25,
                 from /home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/liveMedia.hh:27,
                 from /home/kydo/Downloads/v4l2rtspserver/inc/ServerMediaSubsession.h:20,
                 from /home/kydo/Downloads/v4l2rtspserver/inc/MulticastServerMediaSubsession.h:12,
                 from /home/kydo/Downloads/v4l2rtspserver/src/MulticastServerMediaSubsession.cpp:10:
/home/kydo/Downloads/v4l2rtspserver/x86_64/include/liveMedia/FramedSource.hh:25:23: fatal error: NetCommon.h: No such file or directory
 #include "NetCommon.h"
                       ^
compilation terminated.
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o] Error 1
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [all] Error 2

What could I possibly be missing? Please help me about this issue. Thank you very much

mpromonet, 2018-11-20:
Hi kydoaod,

`NetCommon.h` is a file in live555, then something badly compute the include path.

Did you try to create a toolchain file like:
```
SET(CMAKE_SYSTEM_NAME Android)
SET(CMAKE_SYSTEM_PROCESSOR arm)
SET(CMAKE_C_COMPILER arm-linux-androideabi-gcc)
SET(CMAKE_CXX_COMPILER arm-linux-androideabi-g++)

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```
and next run : 
`cmake -DCMAKE_TOOLCHAIN_FILE=android.toolchain`

Best Regards,
Michel.

kydoaod, 2018-11-21:
Hi Michel,

I have tried on what you have said but now I got this:
Scanning dependencies of target v4l2rtspserver
[  9%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 18%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o
/mnt/c/Users/kdvso/Downloads/v4l2rtspserver/src/DeviceSource.cpp:17:10: fatal error: 'logger.h' file not found
#include "logger.h"
         ^
1 error generated.
CMakeFiles/v4l2rtspserver.dir/build.make:86: recipe for target 'CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o' failed
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

By the way, here is my android.toolchain file:
SET(CMAKE_SYSTEM_NAME Android)
SET(CMAKE_ANDROID_ARCH_ABI armeabi)
SET(CMAKE_ANDROID_STANDALONE_TOOLCHAIN /usr/lib/android)
SET(CMAKE_C_COMPILER arm-linux-androideabi-gcc)
SET(CMAKE_CXX_COMPILER arm-linux-androideabi-g++)

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

Best Regards,
Kydo

mpromonet, 2018-11-21:
Hi Kydo,

This is a part of submodule v4l2wrapper that maps https://github.com/mpromonet/libv4l2cpp.

This is normaly done by cmake, maybe removing libv4l2wrapper.a and running cmake will call the git submodule update and rebuild the lib. If it is not working you should try with a fresh clone.

If you like, you can submit your toolchain as a pull request.

Best Regards,

Michel.

kydoaod, 2018-11-21:
Hi Michel,

Sorry I didn't understand what to do exactly with libv4l2cpp and where should those static files located to be deleted. So can you provide me the next steps for this? Thank you.

Best regards,
Kydo

kydoaod, 2018-12-03:
Hi Michel,

Going back to this, I have tried modifying and recompile the libv4l2wrapper since by default, the classes inside it uses standard linux headers which are not compatible with the android toolchain. Though, I was able to build it, I am going back to the problem regarding missing NetCommon.hh. So is there any way to build this using this live555 ported for android: https://github.com/leepood/live555-android

Once again, thanks for your help.

Best regards,
Kydo

closed 2020-02-14 by mpromonet