mpromonet/v4l2rtspserver #119: Problem during MAKE

morganPigeron, 2018-12-12:
Hi guys ! I have some trouble to build v4l2rtspserver on a raspberry pi Model B Revision 2.0 (Raspbian GNU/Linux 9.6 (stretch)).

I followed the "Setup on Pi" process described in the wiki and when I do the make command I got this error (with a lot of " undefined reference to " before ... ): 

``` bash
collect2: error: ld returned 1 exit status
CMakeFiles/v4l2rtspserver.dir/build.make:357: recipe for target 'v4l2rtspserver' failed
make[2]: *** [v4l2rtspserver] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

```
Does anyone got a clue with this problem ? 
If further informations is needed please ask :)
thanks

mpromonet, 2018-12-12:
Hi morganPigeron,
Surely the undefined reference give some hints to what was not built, maybe live555, maybe libv4l2wrapper ?
The wiki is an old page basically, the build can be done with "cmake . && make"
Best Regards,
Michel.

morganPigeron, 2018-12-12:
Thanks for your answer. I redid the process and got the same problem. Seems like the linking process cant be done : 

``` bash

pi@raspberrypi:~/v4l2rtspserver $ cmake . && make
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0")
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- CMAKE_FIND_ROOT_PATH=/home/pi/v4l2rtspserver/armv6l
-- Installing v4l2wrapper
running mkdir -p /home/pi/v4l2rtspserver/armv6l/include/libv4l2cpp && mkdir -p /home/pi/v4l2rtspserver/armv6l/lib 2>&1
running git submodule init && git submodule update && make -C v4l2wrapper CC=/usr/bin/cc PREFIX=/home/pi/v4l2rtspserver/armv6l all install 2>&1
Submodule 'hls.js' (https://github.com/video-dev/hls.js) registered for path 'hls.js'
Submodule 'v4l2wrapper' (https://github.com/mpromonet/v4l2wrapper) registered for path 'v4l2wrapper'
Cloning into '/home/pi/v4l2rtspserver/hls.js'...
Cloning into '/home/pi/v4l2rtspserver/v4l2wrapper'...
Submodule path 'hls.js': checked out 'e90a1b999071e88af0f734ef18b9e8520c839cec'
Submodule path 'v4l2wrapper': checked out 'a46253f20c7d466bc270aead1efe5149a2239329'
make: Entering directory '/home/pi/v4l2rtspserver/v4l2wrapper'
/usr/bin/cc -c -o src/V4l2MmapDevice.o src/V4l2MmapDevice.cpp -W -Wall -pthread -g -pipe  -I inc
/usr/bin/cc -c -o src/V4l2Capture.o src/V4l2Capture.cpp -W -Wall -pthread -g -pipe  -I inc
/usr/bin/cc -c -o src/V4l2Device.o src/V4l2Device.cpp -W -Wall -pthread -g -pipe  -I inc
/usr/bin/cc -c -o src/V4l2Output.o src/V4l2Output.cpp -W -Wall -pthread -g -pipe  -I inc
/usr/bin/cc -c -o src/logger.o src/logger.cpp -W -Wall -pthread -g -pipe  -I inc
ar rcs libv4l2wrapper.a src/V4l2MmapDevice.o src/V4l2Capture.o src/V4l2Device.o src/V4l2Output.o src/logger.o
install -D -m 0755 inc/*.h /home/pi/v4l2rtspserver/armv6l/include/libv4l2cpp/
install -D -m 0755 *.a /home/pi/v4l2rtspserver/armv6l/lib
make: Leaving directory '/home/pi/v4l2rtspserver/v4l2wrapper'
-- v4l2wrapper built in V4L2WRAPPER_LIBRARY=/home/pi/v4l2rtspserver/armv6l/lib/libv4l2wrapper.a
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Alsa available
-- Log4Cpp available
-- live555 available '/usr/local/include'
-- live555 available ''
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- SystemD available
-- SystemD directory '/lib/systemd/system'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/v4l2rtspserver
Scanning dependencies of target v4l2rtspserver
[  8%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 16%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/DeviceSource.cpp.o
[ 25%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 33%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/HTTPServer.cpp.o
[ 41%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/MJPEGVideoSource.cpp.o
[ 50%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/MemoryBufferSink.cpp.o
[... 399 lines cut ...]
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x10): undefined reference to `Medium::isRTCPInstance() const'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x14): undefined reference to `Medium::isRTSPClient() const'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x18): undefined reference to `RTSPServer::isRTSPServer() const'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x1c): undefined reference to `Medium::isMediaSession() const'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x20): undefined reference to `Medium::isServerMediaSession() const'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x2c): undefined reference to `GenericMediaServer::lookupServerMediaSession(char const*, unsigned char)'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x34): undefined reference to `RTSPServer::createNewClientSession(unsigned int)'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x38): undefined reference to `RTSPServer::allowedCommandNames()'
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x3c): undefined reference to `RTSPServer::weImplementREGISTER(char const*, char const*, char*&)'
[... cut ...]
CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o:(.rodata._ZTI10HTTPServer[_ZTI10HTTPServer]+0x8): undefined reference to `typeinfo for RTSPServer'
collect2: error: ld returned 1 exit status
CMakeFiles/v4l2rtspserver.dir/build.make:357: recipe for target 'v4l2rtspserver' failed
make[2]: *** [v4l2rtspserver] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

```

mpromonet, 2018-12-12:
Hi,

Then it is live555 that is missing, when live555 is available in system but not in cmake root it fails.
With the commit https://github.com/mpromonet/v4l2rtspserver/commit/30c5f88403dba553ffb538b9921264698983b20a it should look to include and librairies consistently and solve your case.

Best Regards,
Michel.

morganPigeron, 2018-12-13:
Thanks :) Its solved .

Best Regards,
Morgan

closed 2018-12-13 by mpromonet