mpromonet/v4l2rtspserver #121: docker run error

dony71, 2018-12-28:
"docker run -p 8554:8554 -it mpromonet/v4l2rtspserver" on raspberry pi

resulting error

standard_init_linux.go:190: exec user process caused "exec format error"

mpromonet, 2018-12-29:
Hi,
The docker image is x86 64bits, it could not run on arm architecture.
You could run the arm release.
Best regards
Michel

juanpescador, 2019-01-06:
Is there an ARM docker release? I checked the tags at https://hub.docker.com/r/mpromonet/v4l2rtspserver but didn't see anything ARM-related.

mkowalczuk, 2019-01-16:
I managed to build RPi docker image by changing `Dockerfile` in the following way:
 1. Replace `FROM` with `balenalib/armv7hf-debian-node` (and optionally `LABEL` to your email)
 1. Move `apt-get` line **before** `git clone`
 1. Add `git` and `wget` to `apt-get` line

mpromonet, 2019-01-25:
Hi,
I commit https://github.com/mpromonet/v4l2rtspserver/commit/c1d23dcfabc1a0354eae3814e4bc7906b9390dd4 that make avialable a docker for raspberry.
Best Regards,
Michel.

closed 2019-01-25 by mpromonet