mpromonet/v4l2rtspserver #127: RTSP stream slows down

ekavet, 2019-02-01:
Hello, at the moment I am using RPi 3 a+.
When I start v4l2rtspserver to stream video to my dahua nvr, everything works fine for couple of hours. 
In the evening I left camera at work to stream video to nvr, when I came back in the morning stream was working but very slow like 1 frame every 2 min or so.  I restarted RPi and everything came back to normal stream delay was close to none 1080p 30 frames.

mpromonet, 2019-02-03:
Hi ekavet,
Did you look to CPU, memory consumpsion ? 
Did you test to connect using another RTSP client, like VLC, ffplay or whatever.
Best Regards,
Michel.

ekavet, 2019-02-04:
Will try to repeat everything. Now just added a bash script to reboot every 4h.

ekavet, 2019-02-05:
I think it is not v4l2rtspserver problem. 
At night in office works fine when I turn lights on it freeze.

nettings, 2020-05-03:
Not directly related, but I'm seeing latency increase when using mpv --live <stream>. After leaving it running overnight, it can be tens of seconds late. mplayer -nocache <stream> is always running at minimal latency, even after many hours. 
But I guess @ekavet you actually saw the framerate drop, not the latency go up, right?