mpromonet/v4l2rtspserver #128: Keep losing stream after 30 minutes

shahonseven, 2019-02-05:
Hi, I keep losing stream after 30 minutes or so and raspberry pi zero w become unresponsive.
[   41.167383] random: 7 urandom warning(s) missed due to ratelimiting
[  441.111534] bcm2835-v4l2: Failed enabling camera, ret -2
[  441.111552] bcm2835-v4l2: Failed to enable camera
[  441.111560] ------------[ cut here ]------------
[  441.111631] WARNING: CPU: 0 PID: 667 at drivers/media/v4l2-core/videobuf2-core.c:1347 vb2_start_streaming+0xd8/0x160 [videobuf2_core]
[  441.111637] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil snd_bcm2835(C) snd_pcm snd_timer cfg80211 snd rfkill uio_pdrv_genirq fixed uio bcm2835_v4l2(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ip_tables x_tables ipv6
[  441.111734] CPU: 0 PID: 667 Comm: vlc Tainted: G         C      4.14.94+ #1193
[  441.111739] Hardware name: BCM2835
[  441.111786] [<c0016420>] (unwind_backtrace) from [<c0013d40>] (show_stack+0x20/0x24)
[  441.111813] [<c0013d40>] (show_stack) from [<c06385c4>] (dump_stack+0x20/0x28)
[  441.111835] [<c06385c4>] (dump_stack) from [<c0021ecc>] (__warn+0xe4/0x10c)
[  441.111848] [<c0021ecc>] (__warn) from [<c0021fc0>] (warn_slowpath_null+0x30/0x38)
[  441.111884] [<c0021fc0>] (warn_slowpath_null) from [<bf0d9584>] (vb2_start_streaming+0xd8/0x160 [videobuf2_core])
[  441.111934] [<bf0d9584>] (vb2_start_streaming [videobuf2_core]) from [<bf0dab0c>] (vb2_core_streamon+0x110/0x168 [videobuf2_core])
[  441.111977] [<bf0dab0c>] (vb2_core_streamon [videobuf2_core]) from [<bf0dadc8>] (__vb2_init_fileio+0x264/0x2fc [videobuf2_core])
[  441.112020] [<bf0dadc8>] (__vb2_init_fileio [videobuf2_core]) from [<bf0db078>] (vb2_core_poll+0x218/0x22c [videobuf2_core])
[  441.112089] [<bf0db078>] (vb2_core_poll [videobuf2_core]) from [<bf0e5738>] (vb2_poll+0x54/0xa8 [videobuf2_v4l2])
[  441.112133] [<bf0e5738>] (vb2_poll [videobuf2_v4l2]) from [<bf0e5924>] (vb2_fop_poll+0x60/0xf4 [videobuf2_v4l2])
[  441.112426] [<bf0e5924>] (vb2_fop_poll [videobuf2_v4l2]) from [<bf09d1c4>] (v4l2_poll+0x6c/0xa0 [videodev])
[  441.112590] [<bf09d1c4>] (v4l2_poll [videodev]) from [<c016f510>] (do_sys_poll+0x218/0x4fc)
[  441.112608] [<c016f510>] (do_sys_poll) from [<c01701c0>] (SyS_poll+0x88/0x110)
[  441.112632] [<c01701c0>] (SyS_poll) from [<c000fd80>] (ret_fast_syscall+0x0/0x28)
[  441.112640] ---[ end trace 1407828c55c82a52 ]---
[  441.136900] bcm2835-v4l2: Failed enabling camera, ret -2
[  441.136913] bcm2835-v4l2: Failed to enable camera
[  441.136921] ------------[ cut here ]------------

mpromonet, 2019-02-08:
Hi shahonseven,
This seems in the v4l2 driver, maybe you should update it.
Did you try to use the read interface (adding -r option) ?
Best Regards,
Michel.

shahonseven, 2019-02-09:
I added the -r option
```
#!/bin/bash

sudo modprobe -v bcm2835-v4l2
v4l2rtspserver -r -F25 -H768 -W1280 -P 8554 /dev/video0
```

```
pi@raspberrypi:~ $ ./startrtspserver.sh 
2019-02-09 09:38:42,603 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:594
	Create V4L2 Source.../dev/video0

[NOTICE] src/V4l2Device.cpp:117
	driver:bm2835 mmal capabilities:85200005 mandatory:1000001
[NOTICE] src/V4l2Device.cpp:120
	/dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
	/dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
	/dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
	/dev/video0:H264 size:1280x720
[NOTICE] src/V4l2Device.cpp:208
	/dev/video0:H264 size:1280x768 bufferSize:983040
[NOTICE] src/V4l2Device.cpp:229
	fps:1/25
[NOTICE] src/V4l2Device.cpp:230
2019-02-09 09:38:42,621 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:617
	Create Source .../dev/video0
2019-02-09 09:38:42,624 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.0.173:8554/unicast"
2019-02-09 09:38:42,629 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
2019-02-09 09:39:19,714 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-02-09 09:39:19,718 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-02-09 09:41:22,160 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-02-09 09:41:22,165 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-02-09 09:41:55,741 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-02-09 09:41:55,747 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
packet_write_wait: Connection to 192.168.0.173 port 22: Broken pipe
```

shahonseven, 2019-02-09:
after installing uv4l it seems more stable, but please don't close this issue yet. I will report back soon.

closed 2019-02-10 by shahonseven