mpromonet/v4l2rtspserver #130: v4l2rtspserver randomly stop

shahonseven, 2019-02-12:
Here is my log:
```
[28350.179594] Unable to handle kernel NULL pointer dereference at virtual address 000005a0
[28350.183456] pgd = ccdf8000
[28350.185041] [000005a0] *pgd=0cdb5831, *pte=00000000, *ppte=00000000
[28350.186729] Internal error: Oops: 17 [#1] ARM
[28350.188363] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil snd_bcm2835(C) snd_pcm cfg80211 snd_timer rfkill snd fixed uio_pdrv_genirq uio cuse fuse bcm2835_v4l2(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ip_tables x_tables ipv6
[28350.198596] CPU: 0 PID: 553 Comm: v4l2rtspserver Tainted: G         C      4.14.97+ #1197
[28350.203728] Hardware name: BCM2835
[28350.205759] task: ccf929a0 task.stack: cbf60000
[28350.207885] PC is at __tcp_transmit_skb+0x60c/0xa78
[28350.209961] LR is at __local_bh_enable_ip+0x98/0x114
[28350.212008] pc : [<c05abaa4>]    lr : [<c002646c>]    psr: 20000013
[28350.214358] sp : cbf61cd0  ip : cbf61b88  fp : cbf61d34
[28350.216507] r10: 000005a8  r9 : 00000000  r8 : 00000000
[28350.218875] r7 : cbf5d000  r6 : cbf5d0b8  r5 : cbf68000  r4 : 00000000
[28350.221622] r3 : 00000001  r2 : 00000000  r1 : 00000001  r0 : 00000000
[28350.223797] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[28350.225922] Control: 00c5387d  Table: 0cdf8008  DAC: 00000055
[28350.228010] Process v4l2rtspserver (pid: 553, stack limit = 0xcbf60188)
[28350.230983] Stack: (0xcbf61cd0 to 0xcbf62000)
[28350.233753] 1cc0:                                     cbf61d0c cbf61ce0 c007d784 c04f3e58
[28350.239076] 1ce0: 9a095bca 00000000 00007560 00000000 00000002 00000000 00000000 69e68bfe
[28350.244546] 1d00: b1d23377 00000000 9a095bdc cbf5d000 000005a8 00000000 000005a8 00000001
[28350.248826] 1d20: cbf68000 000005a8 cbf61da4 cbf61d38 c05ac1a4 c05ab4a4 0d644a7a 00000000
[28350.253353] 1d40: 00000598 cbf61eb8 00000000 cbf5d000 cbf68000 00000598 cbf61dbc cbf61d68
[28350.257801] 1d60: cbf68508 00000004 00000001 cbf68440 61a44dcf 00000000 cbf61d9c 000005a8
[28350.263431] 1d80: cbf61ec0 cbf68140 cbf5d000 cbf68000 00000598 61a4759f cbf61dbc cbf61da8
[28350.268319] 1da0: c05acfd8 c05abf1c 014000c0 00000598 cbf61e54 cbf61dc0 c059d808 c05acfa0
[28350.275123] 1dc0: 000005a8 cbf61dd0 c004857c c004c878 cbd7c560 cbd7c560 00000000 00000000
[28350.282008] 1de0: 00000000 bea5d6a0 00000000 00000000 00000004 00000000 00000010 c0363f10
[28350.287554] 1e00: cbf680ec 00000040 000005a8 c052ad68 cbf61ec0 000005a8 00000000 6574a328
[28350.292988] 1e20: c002646c c00c0000 cbf68000 cbf68000 000005b0 cbf61ec0 00000121 c000ff24
[28350.298911] 1e40: cbf60000 00000000 cbf61e74 cbf61e58 c059de98 c059cfe0 c059de60 cbf68000
[28350.305105] 1e60: 00000000 cc5b7900 cbf61e94 cbf61e78 c05c988c c059de6c c01786f0 c0178644
[28350.311517] 1e80: cbf61ea4 00000000 cbf61ea4 cbf61e98 c0527390 c05c9854 cbf61f8c cbf61ea8
[28350.319378] 1ea0: c0528484 c0527378 cbf61ec8 00000001 fffffff7 00000001 b613f008 000005b0
[28350.326190] 1ec0: 00000000 00000000 00000001 00000598 00000018 cbf61eb8 00000001 cbf61ee8
[28350.331929] 1ee0: 00000000 00000000 00000040 c016fde8 00000000 ce8bfba0 cbf61f1c c016fdb0
[28350.337465] 1f00: cbd7c560 c094e590 00000000 ce8bfba0 cbf61f74 cbf61f20 c064e314 c0046c30
[28350.343856] 1f20: 00000001 00400000 002acd0b 00000010 bea5d720 bea5d618 00000010 bea5d720
[28350.349559] 1f40: bea5d6a0 bea5d620 cbf61f70 00000000 cbf61fa4 cbf61f60 c016fde8 c016e96c
[28350.357050] 1f60: cbf61f70 b6f5a000 00000000 00000000 00000000 000005b0 0000000d 0000000d
[28350.363922] 1f80: cbf61fa4 cbf61f90 c05284d4 c05283c8 00000000 00000000 00000000 cbf61fa8
[28350.369502] 1fa0: c000fd80 c05284b8 000005b0 0000000d 0000000d b613f008 000005b0 00000000
[28350.375140] 1fc0: 000005b0 0000000d 0000000d 00000121 000005b0 014963e8 b6f5a000 0148a4ac
[28350.381525] 1fe0: 00000000 bea5d480 00000000 b6efa984 80000010 0000000d 00000000 00000000
[28350.387155] [<c05abaa4>] (__tcp_transmit_skb) from [<c05ac1a4>] (tcp_write_xmit+0x294/0xfd4)
[28350.393569] Unable to handle kernel paging request at virtual address d3a48040
[28350.393574] pgd = ccdf8000
[28350.393578] [d3a48040] *pgd=00000000
[28350.393584] Internal error: Oops: 5 [#2] ARM
[28350.393586] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil snd_bcm2835(C) snd_pcm cfg80211 snd_timer rfkill snd fixed uio_pdrv_genirq uio cuse fuse bcm2835_v4l2(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ip_tables x_tables ipv6
[28350.393630] CPU: 0 PID: 553 Comm: v4l2rtspserver Tainted: G         C      4.14.97+ #1197
[28350.393632] Hardware name: BCM2835
[28350.393635] task: ccf929a0 task.stack: cbf60000
[28350.393637] PC is at cfb_imageblit+0x42c/0x980
[28350.393639] LR is at 0xd1000000
[28350.393642] pc : [<c03ab204>]    lr : [<d1000000>]    psr: 20000193
[28350.393645] sp : cbf61710  ip : 00000000  fp : cbf61764
[28350.393648] r10: 00000010  r9 : 00000001  r8 : 00000000
[28350.393651] r7 : 00000020  r6 : 00000020  r5 : ff000000  r4 : ffaaaaaa
[28350.393654] r3 : d3a48040  r2 : cf03ef60  r1 : 00000a40  r0 : cf274800
[28350.393657] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[28350.393660] Control: 00c5387d  Table: 0cdf8008  DAC: 00000055
[28350.393663] Process v4l2rtspserver (pid: 553, stack limit = 0xcbf60188)
[28350.393666] Stack: (0xcbf61710 to 0xcbf62000)
[28350.393670] 1700:                                     00000000 00002900 00000a40 d3a48040
[28350.393674] 1720: 00000008 00000040 0000000f 00000000 cf03ef60 d3a48040 c03abc0c 00000000
[28350.393678] 1740: cf03ef68 cf29098c 00000000 cf028400 00000001 000000ff cbf61774 cbf61768
[... 62 lines cut ...]
[28350.393924] 1f20: 00000001 00400000 002acd0b 00000010 bea5d720 bea5d618 00000010 bea5d720
[28350.393928] 1f40: bea5d6a0 bea5d620 cbf61f70 00000000 cbf61fa4 cbf61f60 c016fde8 c016e96c
[28350.393932] 1f60: cbf61f70 b6f5a000 00000000 00000000 00000000 000005b0 0000000d 0000000d
[28350.393936] 1f80: cbf61fa4 cbf61f90 c05284d4 c05283c8 00000000 00000000 00000000 cbf61fa8
[28350.393940] 1fa0: c000fd80 c05284b8 000005b0 0000000d 0000000d b613f008 000005b0 00000000
[28350.393944] 1fc0: 000005b0 0000000d 0000000d 00000121 000005b0 014963e8 b6f5a000 0148a4ac
[28350.393948] 1fe0: 00000000 bea5d480 00000000 b6efa984 80000010 0000000d 00000000 00000000
[28350.393952] [<c03ab204>] (cfb_imageblit) from [<c03abc0c>] (bcm2708_fb_imageblit+0x18/0x1c)
[28350.393955] [<c03abc0c>] (bcm2708_fb_imageblit) from [<c03a9864>] (bit_putcs+0x254/0x400)
[28350.393959] [<c03a9864>] (bit_putcs) from [<c03a3424>] (fbcon_putcs+0xf4/0x12c)
[28350.393962] [<c03a3424>] (fbcon_putcs) from [<c03a5800>] (fbcon_redraw+0x164/0x1c4)
[28350.393966] [<c03a5800>] (fbcon_redraw) from [<c03a8220>] (fbcon_scroll+0x27c/0xdd0)
[28350.393970] [<c03a8220>] (fbcon_scroll) from [<c03dddfc>] (con_scroll+0xf0/0xfc)
[28350.393973] [<c03dddfc>] (con_scroll) from [<c03dde9c>] (lf+0x94/0x9c)
[28350.393977] [<c03dde9c>] (lf) from [<c03dea28>] (vt_console_print+0x254/0x370)
[28350.393980] [<c03dea28>] (vt_console_print) from [<c005f95c>] (console_unlock+0x248/0x50c)
[28350.393984] [<c005f95c>] (console_unlock) from [<c005fd78>] (vprintk_emit+0x158/0x358)
[28350.393988] [<c005fd78>] (vprintk_emit) from [<c0060140>] (vprintk_default+0x4c/0x7c)
[28350.393992] [<c0060140>] (vprintk_default) from [<c0061424>] (vprintk_func+0x50/0xb8)
[28350.393995] [<c0061424>] (vprintk_func) from [<c0060dc0>] (printk+0x30/0x38)
[28350.393999] [<c0060dc0>] (printk) from [<c0013bfc>] (dump_backtrace_entry+0x38/0xa4)
[28350.394003] [<c0013bfc>] (dump_backtrace_entry) from [<c00164a4>] (unwind_backtrace+0x84/0x104)
[28350.394006] [<c00164a4>] (unwind_backtrace) from [<c0013f54>] (die+0x210/0x294)
[28350.394011] [<c0013f54>] (die) from [<c0018f90>] (__do_kernel_fault.
```

shahonseven, 2019-02-12:
it stops again..

```
[12724.040840] Unable to handle kernel NULL pointer dereference at virtual address 000005a0
[12724.045086] pgd = cc0a0000
[12724.046675] [000005a0] *pgd=0c109831, *pte=00000000, *ppte=00000000
[12724.048328] Internal error: Oops: 17 [#1] ARM
[12724.049952] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd fixed uio_pdrv_genirq uio cuse fuse bcm2835_v4l2(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ip_tables x_tables ipv6
[12724.059800] CPU: 0 PID: 428 Comm: v4l2rtspserver Tainted: G         C      4.14.97+ #1197
[12724.065085] Hardware name: BCM2835
[12724.067191] task: ce8def00 task.stack: cd140000
[12724.069258] PC is at __tcp_transmit_skb+0x60c/0xa78
[12724.071333] LR is at __local_bh_enable_ip+0x98/0x114
[12724.073385] pc : [<c05abaa4>]    lr : [<c002646c>]    psr: 20000013
[12724.075531] sp : cd141cd0  ip : cd141b88  fp : cd141d34
[12724.077853] r10: 000005a8  r9 : 00000000  r8 : 00000000
[12724.079996] r7 : cc01c180  r6 : cc01c238  r5 : cc0d6400  r4 : 00000000
[12724.082644] r3 : 00000001  r2 : 00000000  r1 : 00000001  r0 : 00000000
[12724.085185] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[12724.087334] Control: 00c5387d  Table: 0c0a0008  DAC: 00000055
[12724.089431] Process v4l2rtspserver (pid: 428, stack limit = 0xcd140188)
[12724.091596] Stack: (0xcd141cd0 to 0xcd142000)
[12724.094906] 1cc0:                                     c01481ac c00c5dd8 cfd920a0 cf001840
[12724.100189] 1ce0: 00000000 000007a0 00007560 c052e968 00000002 00000000 00000000 c03de553
[12724.105636] 1d00: b364fdb3 00000000 f6a58538 cc01c180 000005a8 00000000 000005a8 00000001
[12724.109811] 1d20: cc0d6400 000005a8 cd141da4 cd141d38 c05ac1a4 c05ab4a4 980eda9f c052e974
[12724.114242] 1d40: 000005a4 cd141eb8 00000000 cc01c180 cc0d6400 000005a4 cd141dbc cd141d68
[12724.118707] 1d60: cc0d6908 00000004 00000001 cc0d6840 979171c9 00000000 cd141d9c 000005a8
[12724.124270] 1d80: cd141ec0 cc0d6540 cc01c180 cc0d6400 000005a4 979199a5 cd141dbc cd141da8
[12724.129227] 1da0: c05acfd8 c05abf1c 014000c0 000005a4 cd141e54 cd141dc0 c059d808 c05acfa0
[12724.135585] 1dc0: 000005a8 cd141dd0 c004857c c004c878 cd154560 cd154560 00000000 00000000
[12724.141766] 1de0: 00000000 bed526a0 00000000 00000000 00000004 00000000 00000004 00000000
[12724.148510] 1e00: cc0d64ec 00000040 000005a8 c052ad68 cd141ec0 000005a8 00000000 4e9853b1
[12724.153988] 1e20: c002646c c00c0000 cc0d6400 cc0d6400 000005b0 cd141ec0 00000121 c000ff24
[12724.159774] 1e40: cd140000 00000000 cd141e74 cd141e58 c059de98 c059cfe0 c059de60 cc0d6400
[12724.166199] 1e60: 00000000 cc9b3780 cd141e94 cd141e78 c05c988c c059de6c c01786f0 c0178644
[12724.171680] 1e80: cd141ea4 00000000 cd141ea4 cd141e98 c0527390 c05c9854 cd141f8c cd141ea8
[12724.179279] 1ea0: c0528484 c0527378 cd141ec8 c04cf7a8 fffffff7 00000001 b623e008 000005b0
[12724.187159] 1ec0: 00000000 00000000 00000001 000005a4 0000000c cd141eb8 00000001 60000113
[12724.192808] 1ee0: 00000000 00000000 00000040 c016fde8 cd141f14 cd141f00 c002604c c016fdb0
[12724.198419] 1f00: c09db59c c09c96d4 cd141f5c cd141f18 c0009718 c002600c cf02b900 c006560c
[12724.204830] 1f20: 00000001 00400000 0012f51e 0000000c bed52720 bed52618 0000000c bed52720
[12724.210355] 1f40: bed526a0 bed52620 cd141f70 00000000 cd141fa4 cd141f60 c016fde8 c016e96c
[12724.217466] 1f60: cd141f70 cd141f70 00000000 00000000 00000000 000005b0 00000009 00000009
[12724.224677] 1f80: cd141fa4 cd141f90 c05284d4 c05283c8 00000000 00000000 00000000 cd141fa8
[12724.230367] 1fa0: c000fd80 c05284b8 000005b0 00000009 00000009 b623e008 000005b0 00000000
[12724.235928] 1fc0: 000005b0 00000009 00000009 00000121 000005b0 008e63e8 b6f06000 008da4ac
[12724.241931] 1fe0: 00000000 bed52480 00000000 b6ea6984 80000010 00000009 00000000 00000000
[12724.248044] [<c05abaa4>] (__tcp_transmit_skb) from [<c05ac1a4>] (tcp_write_xmit+0x294/0xfd4)
[12724.254609] [<c05ac1a4>] (tcp_write_xmit) from [<c05acfd8>] (tcp_push_one+0x44/0x50)
[12724.261872] [<c05acfd8>] (tcp_push_one) from [<c059d808>] (tcp_sendmsg_locked+0x834/0xe8c)
[12724.268174] [<c059d808>] (tcp_sendmsg_locked) from [<c059de98>] (tcp_sendmsg+0x38/0x4c)
[12724.273922] [<c059de98>] (tcp_sendmsg) from [<c05c988c>] (inet_sendmsg+0x44/0x78)
[12724.279509] [<c05c988c>] (inet_sendmsg) from [<c0527390>] (sock_sendmsg+0x24/0x34)
[12724.285917] [<c0527390>] (sock_sendmsg) from [<c0528484>] (SyS_sendto+0xc8/0xf0)
[12724.291448] [<c0528484>] (SyS_sendto) from [<c05284d4>] (SyS_send+0x28/0x30)
[12724.295705] [<c05284d4>] (SyS_send) from [<c000fd80>] (ret_fast_syscall+0x0/0x28)
[12724.302742] Code: e596300c e3530000 0afffee0 e7f001f2 (e14a20d8) 
[12724.306344] ---[ end trace d7ad9dff0e44c63d ]---
```

mpromonet, 2019-02-12:
Hi shahonseven,

Nice logs :)
Maybe you should run a debugger to get more information about your problem.

Best Regards,
Michel.

lahdekorpi, 2019-04-16:
I'm having this exact issue with bcm2835-v4l2. Any suggestions on debugging? Just strace it and wait for a crash?

lahdekorpi, 2019-04-16:
```c
mmap2(NULL, 2093056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb4d01000
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [163840], [4]) = 0
setsockopt(6, SOL_SOCKET, SO_SNDBUF, [31783900], 4) = 0
getsockname(5, {sa_family=AF_INET, sin_port=htons(8554), sin_addr=inet_addr("192.168.0.240")}, [16]) = 0
fcntl64(7, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(7, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
getsockopt(7, SOL_SOCKET, SO_SNDBUF, [163840], [4]) = 0
getsockopt(7, SOL_SOCKET, SO_SNDBUF, [163840], [4]) = 0
mmap2(NULL, 2093056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb4b02000
open("/proc/sys/vm/overcommit_memory", O_RDONLY|O_CLOEXEC) = 8
read(8, "0", 1)                         = 1
close(8)                                = 0
madvise(0xb4f54000, 61440, MADV_DONTNEED) = 0
futex(0xb4f00010, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x26cef4, FUTEX_WAKE_PRIVATE, 1)  = 1
madvise(0xb4f4e000, 24576, MADV_DONTNEED) = 0
madvise(0xb4f36000, 98304, MADV_DONTNEED) = 0
madvise(0xb4f27000, 61440, MADV_DONTNEED) = 0
madvise(0xb4f21000, 24576, MADV_DONTNEED) = 0
munmap(0xb4add000, 151552)              = 0
munmap(0xb4adb000, 159744)              = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x5b4} ---
+++ killed by SIGSEGV +++
```

🤷‍♂️

closed 2019-10-25 by mpromonet