mpromonet/v4l2rtspserver #132: [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:599

NathanLKing, 2019-02-26:
Howdy everyone.

Thanks for the great system. I've been using it lately, yet I've now come across this issue.
`2019-02-26 14:09:13,932 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:599
        Create V4L2 Source.../dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
`

I've tried akutruff's solution, but nothing changed.
https://github.com/mpromonet/v4l2rtspserver/issues/122

I've tried a complete refresh off raspbian lite and all installs. Nothing new.

Any help would be greatly appreciated.
Thank you.

Nathan

mhaddy, 2019-02-26:
@NathanLKing I'm experiencing the same issue, also with a fresh build and RPI Zero W.

NathanLKing, 2019-02-26:
Howdy @mhaddy 

Thanks for sharing and the fix. I'll try that out now.

NathanLKing, 2019-02-26:
Unfortunately, this didn't change anything @mhaddy

mpromonet, 2019-02-26:
Hi NathanLKing,
I have no rasperry zero, but recently there is several issue reporting mmap problem with this.
Did you try to capture frame using v4l2-ctl, with some command like : 

    v4l2-ctl --device /dev/video0 --stream-mmap --stream-to=frame.raw --stream-count=1

Another thing that you can try is to add the '-r' option to use the read interface instead of the mmap interface.

Best Regards,
Michel.

NathanLKing, 2019-02-26:
Hi @mpromonet thanks for responding and your suggestions.

I tried running `v4l2-ctl --device /dev/video0 --stream-mmap --stream-to=frame.raw --stream-count=1`
And received this back "Failed to open /dev/video0: No such file or directory"

I tried modifying startrtspserver.sh to the -r  this is the contents of the file:
`#!/bin/bash
sudo modprobe -v bcm2835-v4l2
v4l2rtspserver -r -F15 -H 720 -W1280 -P 8555 /dev/video0
`

I noticed with this fresh install that the /lib/systemd/system/v4l2rtspserver.service file had been automatically created and yet the instructions haven't changed. Plus, the file contents were different to the instructions. An issue?

Any assistance would be very much appreciated. 

Thanks.

NathanLKing, 2019-02-27:
UPDATE

I've had some embarrassment, some failure and some successes.

Firstly, the amusing part. I checked (for the fifteenth time) the ribbon connections were ok on the Pi and the Camera. All ok. BUT, I did notice that the connector that goes from the camera lens to connect to the camera module was not properly seated. (silly mistake made when removing the plastic covering from the sticky when mounting the camera to the board. 
DUH!!

So, booted up and I was experiencing the following errors:
`2019-02-27 16:58:13,419 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:599
        Create V4L2 Source.../dev/video0

[NOTICE] src/V4l2Device.cpp:117
        driver:bm2835 mmal capabilities:85200005 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video0:H264 size:1024x768
[ERROR] src/V4l2Device.cpp:190
        Cannot set format for device:/dev/video0 Device or resource busy
[ERROR] src/V4l2Device.cpp:190
        Cannot set format for device:/dev/video0 Device or resource busy
[ERROR] src/V4l2Device.cpp:67
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
`
This occurs when I use the (newly) automatically installed info in the file "/lib/systemd/system/v4l2rtspserver.service" which is: 
`[Unit]
Description=V4L2 RTSP server
After=network.target

[Service]
Type=simple
Restart=always
RestartSec=5
ExecStart=/usr/local/bin/v4l2rtspserver -S1
WorkingDirectory=/usr/local/share/v4l2rtspserver

[Install]
WantedBy=multi-user.target
`
If I revert back to the current instruction contents, the way it was working before all of this started, it would contain the quite different Service info:
`[Service]
Type=simple
Environment=LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so
ExecStart=/usr/local/bin/v4l2rtspserver -F15 -H 720 -W1280 -P 8555 /dev/video0
User=pi
Restart=always
RestartSec=1
StartLimitIntervalSec=0
`
So, after reboot, I then only get this message and the stream functions:
`2019-02-27 17:02:35,485 [ERROR ] - /home/pi/v4l2rtspserver/src/main.cpp:573
        Failed to create RTSP server: bind() error (port number: 8555): Address already in use
`
If I remove "-P 8555" from the startrtspserver.sh file, I get this load of errors, but the camera still works:
`2019-02-27 17:06:38,587 [NOTICE] - /home/pi/v4l2rtspserver/src/main.cpp:599
        Create V4L2 Source.../dev/video0

[NOTICE] src/V4l2Device.cpp:117
        driver:bm2835 mmal capabilities:85200005 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video0:H264 size:1280x720
[NOTICE] src/V4l2Device.cpp:229
        fps:1/15
[NOTICE] src/V4l2Device.cpp:230
VIDIOC_REQBUFS: Device or resource busy
VIDIOC_STREAMOFF: Device or resource busy
VIDIOC_REQBUFS: Device or resource busy
`
Hope this update clears up  something for someone, Would appreciate any help.

Thanks again.

mpromonet, 2019-02-27:
Hi,
bcm2835-v4l2 and uv4l are 2 differents implmentaion of a V4L2 driver.
You should use one but not both, they access to the same resource.
Adding libuv4lext.so in the LD_LIBRARY path using bcm2835-v4l2 doesnot have sense.
If v4l2-ctl command have access problem, v4l2rtspserver will have the same access problem.
Best Regards,
Michel.

NathanLKing, 2019-02-27:
Hi Michel,

Thanks for your response.
I understand what you've said. I don't know how to fix it though. 
Does the [SERVICE] info in the automatically created "/lib/systemd/system/v4l2rtspserver.service" require some modification?

Nathan

mpromonet, 2020-09-19:
I close this isssue that is often related to problem in the driver, maybe in libv4l2cpp.

closed 2020-09-19 by mpromonet