mpromonet/v4l2rtspserver #133: CMake can not determine linker language for target: live555

birdynm, 2019-03-08:
Hi,

firtst I wanted to thank you for this tool! It really helped me to solve a problem I had for a long time!
I wanted to reinstall v4l2rtspserver on a second rpi but this time I can't make it through the cmake process. A couple of weeks ago it worked just fine.

I tried it as well with pre-installing the Packages liblog4cpp5-dev + libv4l-dev as well without pre-installing them.

Thanks!

root@raspberrypi:/home/pi/v4l2rtspserver# cmake .
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0")
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Alsa available
running git submodule update --init 2>&1
Submodule 'hls.js' (https://github.com/video-dev/hls.js) registered for path 'hls.js'
Submodule 'v4l2wrapper' (https://github.com/mpromonet/v4l2wrapper) registered for path 'v4l2wrapper'
Cloning into '/home/pi/v4l2rtspserver/hls.js'...
Cloning into '/home/pi/v4l2rtspserver/v4l2wrapper'...
Submodule path 'hls.js': checked out 'e90a1b999071e88af0f734ef18b9e8520c839cec'
Submodule path 'v4l2wrapper': checked out 'c7e735f10927d91bd65b525f381886d5f4bad9c4'
You have called ADD_LIBRARY for library live555 without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- SystemD available
-- SystemD directory '/lib/systemd/system'
-- Configuring done
CMake Error: Cannot determine link language for target "live555".
CMake Error: CMake can not determine linker language for target: live555
-- Generating done
-- Build files have been written to: /home/pi/v4l2rtspserver

sammy2142, 2019-03-09:
The http://www.live555.com website is currently down so the source files are not downloaded
Regards

mpromonet, 2019-03-09:
You can get live555 from https://download.videolan.org/pub/contrib/live555/

birdynm, 2019-03-10:
Hi, Thank you very much! Such a basic mistake. Got it running :)
In case someone else stumbles over it: I downloaded the live555 archive and extracted it. Then I copied the "live" folder into the "v4l2rtspserver" folder an proceeded with the normal installation process...

closed 2019-03-10 by birdynm