mpromonet/v4l2rtspserver #134: Raspberry PI Compilation failure

sq6emm, 2019-03-10:
Hi I am trying to compile your code, however I fail each time, please take a look:

`
pi@raspberrypi:~ $ git clone https://github.com/mpromonet/v4l2rtspserver.git
Cloning into 'v4l2rtspserver'...
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 1816 (delta 26), reused 27 (delta 10), pack-reused 1763
Receiving objects: 100% (1816/1816), 584.04 KiB | 496.00 KiB/s, done.
Resolving deltas: 100% (1182/1182), done.
pi@raspberrypi:~ $ cd v4l2rtspserver/
pi@raspberrypi:~/v4l2rtspserver $ cmake .
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0") 
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
running git submodule update --init 2>&1
Submodule 'hls.js' (https://github.com/video-dev/hls.js) registered for path 'hls.js'
Submodule 'v4l2wrapper' (https://github.com/mpromonet/v4l2wrapper) registered for path 'v4l2wrapper'
Cloning into '/home/pi/v4l2rtspserver/hls.js'...
Cloning into '/home/pi/v4l2rtspserver/v4l2wrapper'...
Submodule path 'hls.js': checked out 'e90a1b999071e88af0f734ef18b9e8520c839cec'
Submodule path 'v4l2wrapper': checked out 'c7e735f10927d91bd65b525f381886d5f4bad9c4'
You have called ADD_LIBRARY for library live555 without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- SystemD available
-- SystemD directory '/lib/systemd/system'
-- Configuring done
CMake Error: Cannot determine link language for target "live555".
CMake Error: CMake can not determine linker language for target: live555
-- Generating done
-- Build files have been written to: /home/pi/v4l2rtspserver
pi@raspberrypi:~ $

`

mpromonet, 2019-03-10:
Hi,

This is exactly the same problem than https://github.com/mpromonet/v4l2rtspserver/issues/133, live555 is down
You can give overide the live555 url with : 

     cmake LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2019.03.06.tar.gz .

Best Regards,
Michel.

closed 2019-03-10 by mpromonet

sq6emm, 2019-03-10:
> ```
>  cmake LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2019.03.06.tar.gz .
> ```

Thanks Michel, it did not work on my side. I had to manually download and unpack file.
OTOH thanks for solution.