mpromonet/v4l2rtspserver #135: Ability to remove contextroot

calibx, 2019-03-11:
With the -u option it is possible to change the contextroot but it doesn't seems possible to remove it to improve compatibility with chinese nvr.

mpromonet, 2019-03-11:
Hi,
Not sure to understand. Did you try -u"" ?
Best Regards,
Michel.

calibx, 2019-03-12:
Hi Michel,

Yes I have tried but I'd got this error :
# /system/sdcard/bin/v4l2rtspserver-master  -E MP3:16000:44100 -r 2  -P 8554 -u""
/system/sdcard/bin/v4l2rtspserver-master: option requires an argument -- u
/

Regards,
Sébastien.

mpromonet, 2019-03-13:
Hi Sebastien,
I don't know which version you are using, -E doesnot exist, -r doesnot have argument.
Using -u "" or -u '' with version 0.1.6 allow to get the stream using rtsp://...:8554/ 
Best Regards,
Michel.

calibx, 2019-04-04:
Hi,
I'm using the version included in the Dafang-Hack's firmware. (https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks)
Is there a way to display the version ?

mpromonet, 2019-04-10:
Hi calibx,
The version could be displayed with -V option, however it needed to be build from a git repository. 
I had a quick look to https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks it seems that it was built from zip and then have no git information.
I think the best thing is to ask  in https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues
Best Regards,
Michel.

closed 2019-04-10 by mpromonet