mpromonet/v4l2rtspserver #136: 'Inappropriate ioctl for device' with raspberry camera

Mific78, 2019-03-29:
Hello,

I'm in trouble with the v4l2rtspserver. I had installed it on my Raspberry Pi and everything was OK but when I have reboot and start again the rtspserver, I have the following error :
`log level:500`
`[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:599`
`VIDIOC_REQBUFS: Inappropriate ioctl for device`
`VIDIOC_STREAMOFF: Inappropriate ioctl for device`
`VIDIOC_REQBUFS: Inappropriate ioctl for device`
`Create V4L2 Source.../dev/video0`

The result of the command `sudo v4l2-ctl --all` is :
`VIDIOC_QUERYCAP: failed: Inappropriate ioctl for device`
`/dev/video0: not a v4l2 node`

The result of the command `ls -al /dev/video*` is :
`-rwxrw-rw-+ 1 root root 0 mars  29 11:23 /dev/video0`

Thanks in advance for your help !

mpromonet, 2019-03-29:
Hi,
The problem seems the v4l2 device and maybe some strange stuff that create a file /dev/video0.
/dev/video0 is not a regular file, it is a char device, the permission should start with `-c...`
Best Regards,
Michel.

uglurass, 2019-03-30:
does trying different resolution help ?

mpromonet, 2019-03-31:
@uglurass probably not, till /dev/video0 is a regular file and not a device

triplepoint, 2019-04-02:
Well, for what it's worth I got this to work from this condition by doing the following:
- sudo rm /dev/video0
- sudo modprobe bcm2835-v4l2
This creates the appropriate /dev/video0 device.

Then you can start the `v4l2rtspserver` service as defined in the docs.

Hope that helps.

uglurass, 2019-04-05:
can anyone provide an img of a standard setup ?

mpromonet, 2019-04-05:
Hi @uglurass,
You can use the docker image with :

   docker run --device=/dev/video0 -p 8554:8554 -it mpromonet/v4l2rtspserver

It should run on armv6, armv7 and amd64.
Best Regards,
Michel.

closed 2019-04-10 by mpromonet