mpromonet/v4l2rtspserver #140: the video is green  except the top left

zhanwei, 2019-05-31:
follow the documents.
using vlc to play the rtsp stream 
[image]

mpromonet, 2019-05-31:
Hi,
What did you tried to understand your problem ?
Best Regards,
Michel.

mpromonet, 2019-06-06:
Hi Pablo ....
Did you use the same format ? the same os ? the same log ?
Best Regards,
Michel.

pabclsn, 2019-06-06:
I think my problem is the webcam it's YUYV ... 
I should try the loopback thing !

mpromonet, 2019-06-06:
YUYV is really new, it is in progress.
live555 add this format recently, and it is not sure that VLC support it nor sink is correctly filled.

pabclsn, 2019-06-06:
My webcam is a LOGI C270, I'm doing this for Homebridge projet. Sometime I'm getting this error Bad header => dropping frame, sometime this one RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

pabclsn, 2019-06-06:
I see my finger in the little rectangle btw ;)
[image]

pabclsn, 2019-06-08:
Hi I ordered a new cam to test, this one [https://www.amazon.fr/ELP-Surveillance-Infrarouge-lextérieur-lIndustrie/dp/B07FXDWB1K/ref=sr_1_4?keywords=elp+camera&qid=1560002639&s=electronics&sr=1-4](url).

[image]

this is before I connect to rstp with VLC.
[image]

The in connect.
[image]

And there is the result in vlc.

[image]

Any idea ?

mpromonet, 2019-06-10:
Hi,
Maybe the network is lossing most of the frame.
Did you try to run VLC using RTP over TCP ? (something like vlc --rtsp-tcp rtsp://path/to/stream)
Best Regards,
Michel.

pabclsn, 2019-06-11:
I did try , same result ...

mpromonet, 2019-06-11:
Hi,
Then it is not a network issue.
Did you tried to reduce the capture size ? I am also not sure that `-F0` is a good idea ...
Best Regards,
Michel.

yotamr, 2019-06-20:
Hello! :)
I am not sure, but I believe I am experiencing something similar.
The scenario is this. I stream JPG's from an HTTP source into the loopback device. No transcoding is done in this phase, so only JPG's are being written to the loopback device.

v4l2rtspserver is used to perform the JPG streaming into MJPEG. When using VLC as a client, I also see a small part of the frame correctly, while the rest is garbled.

I've sniffed the RTP traffic. The size of the sent frame is correct, and all packets are received. I believe that only the first RTP fragment is displayed correctly, while the rest is not. 
To reproduce this, you can perform the following:
 1. Download mjpeg-server # Requires go
 2. Run it: go run main.go -a 0.0.0.0:9000 -- go run mpjpeg/main.go -- gopher.jpg
 3. Load a loopback device (modprobe v4l2loopback)
 4. Stream the JPG's into the loopback device:
 5. ffmpeg  -re -i http://localhost:9000/?action=stream -r 10 -qmax 1 -qmin 1 -c copy -q:v 0 -f v4l2 /dev/video0
 6. Start the v4l2rtspserver: ./v4l2rtspserver -P 8554 -u h264 /dev/video0 -vvvv

When you stream from the RTSP server, you will be able to see the same issue.

hoonkai, 2019-06-26:
Hi all,

I have the same issue and can't figure out what the problem is.

```
$ v4l2-ctl -V
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'MJPG'
	Field             : None
	Bytes per Line    : 0
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             :
```

[image]

mpromonet, 2019-07-06:
Hi @yotamr,

Thanks for describe a scenario to reproduce the problem.
However I cannot get run the ffmpeg command, I get :

> [v4l2 @ 0x55f4ae3b59c0] V4L2 output device supports only a single raw video stream
> Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
> 

I will try to write the frame directly to the v4l2 device.

Best Regards,
Michel.

yotamr, 2019-08-05:
Hi @mpromonet. I've just returned from a long holiday, so I apologize for missing this.
Have you had any luck? In any event, I forgot to mention that in order to perform this kind of loopback v4l2 streaming, a small patch needs to be to ffmpeg. Here is the batch, based on ffmpeg 3.4.6:

`diff -ur ffmpeg-3.4.6/libavdevice/v4l2enc.c ffmpeg-3.4.6-patched/libavdevice/v4l2enc.c
--- ffmpeg-3.4.6/libavdevice/v4l2enc.c  2018-07-18 07:33:10.000000000 -0700
+++ ffmpeg-3.4.6-patched/libavdevice/v4l2enc.c  2019-03-17 08:12:25.306547200 -0700
@@ -47,8 +47,7 @@
     }

     if (s1->nb_streams != 1 ||
-        s1->streams[0]->codecpar->codec_type != AVMEDIA_TYPE_VIDEO ||
-        s1->streams[0]->codecpar->codec_id   != AV_CODEC_ID_RAWVIDEO) {
+        s1->streams[0]->codecpar->codec_type != AVMEDIA_TYPE_VIDEO) {
         av_log(s1, AV_LOG_ERROR,
                "V4L2 output device supports only a single raw video stream\n");
         return AVERROR(EINVAL);
@@ -56,7 +55,7 @@

     par = s1->streams[0]->codecpar;

-    v4l2_pixfmt = ff_fmt_ff2v4l(par->format, AV_CODEC_ID_RAWVIDEO);
+    v4l2_pixfmt = ff_fmt_ff2v4l(AV_PIX_FMT_NONE, s1->streams[0]->codecpar->codec_id);
     if (!v4l2_pixfmt) { // XXX: try to force them one by one?
         av_log(s1, AV_LOG_ERROR, "Unknown V4L2 pixel format equivalent for %s\n",
                av_get_pix_fmt_name(par->format));
`

yotamr, 2019-09-22:
I've isolated this issue to images that have restart markers. The current
v4l2rtspserver code does not handle the existence of the DRI JPG marker in
input images, which results in incorrect RTP headers. I've implemented a
fix, which seems to work. I will open a pull request soon.

Thanks,
 Yotam

sndnvaps, 2019-10-19:
i have the same problem

sndnvaps, 2019-10-19:
get the same problem

[image]

closed 2019-10-25 by mpromonet