mpromonet/v4l2rtspserver #142: Video Broken up

simonwhill, 2019-06-27:
Hi Sorry to post two issues, i have just posted about dropped frames, but this is a different issue.

So if i start the server with this command line

 sudo v4l2rtspserver   -v -P 7554 -p 7554 -W 1280 -H 720  /dev/video0

I get the following broken up video
[image]

What am i doing incorrectly, its running on a raspberry Pi 3 with a Logitech C920 as the camera source, being played back by VLC on Windows 10 Pro.

Thanks

Simon

mpromonet, 2019-06-27:
Hi Simon,

This usually means your network is loosing I-frame. As I-frame are bigger P-frame, when the network capacity is not enough, the I-frame are lost.

In order to avoid UDP dropping, you can use RTP over tcp adding to vlc command `--rtsp-tcp` option.

Best Regards,

Michel.

simonwhill, 2019-06-28:
Im not sure its network related, the pi is on a 1gb link to the local switch which the receving pc is also connected too. so its all local, ive atached a pic of the use of the switch you provided and it still seems to be the same.
[image]
Thanks 
Simon

simonwhill, 2019-06-28:
also i noticed that if the scean changes the image cleans up, not completly, but it will revert to the gray image after a few seconds 

Simon
[image]

mpromonet, 2019-06-28:
Hi Simon,

> also i noticed that if the scene changes the image cleans up, not completly, but it will revert to the gray image after a few seconds

This is typical of loosing I-frame.

Did you try to set stream over TCP ?
[image]

Best Regards,

Michel

simonwhill, 2019-07-01:
Hi Michel

So making VLC use TCP did not help, however adding the -c switch to the command line did, so sudo v4l2rtspserver -v -c -P 7554 -p 7554 -W 1280 -H 720 /dev/video0 solves the video breakup.

I'm not sure why not sending the config solves the issue with VLC but it is working now.

Thanks for your help and advice

Simon

mpromonet, 2019-07-01:
Hi Simon,
Interesting, this means the mechanism of repeting SPS/PPS breaks following I-Frame. Never saw this problem, probably the capture frame splitting to get SPS/PPS have some errors.
Regards,
Michel.

careyer, 2019-07-24:
Hi @mpromonet, Hi @simonwhill ,

please excuse me! I don't want to hijack your discussion but I am experiencing a similar problem. I try to stream the video from a Insta360 and have similar video-breakups like simonwhill is seeing.
I am using a different videoplayer here but using VLC the result is the same. VLC is even a bit more picky and won't display anything if it gets too worse.

Let me give you two examples:
[image]

[image]

I think the problem is that the H264 video stream somehow is either missing enough keygframe/IDR frames or stream config information. (I already tried the "-c" parameter but that does not help either). 
Do you guys have any suggestions what I can try to solve that issue?

Thanks!

roger-, 2020-04-26:
BTW I had the same issue in #81 and it still happens.

tachang, 2020-05-31:
I've definitely seen this as well. It seems like the h264 video file that is created is missing some keyframes or something.

sshaikh, 2020-06-29:
Perhaps the easiest way would be to introduce a parameter that can set the iframe/keyframe/GOP frequency?

mpromonet, 2020-06-29:
Hi @sshaikh,

You can do this running v4l2-ctl.

Best Regards,
Michel.

sshaikh, 2020-06-29:
Thanks for the tip - I'll look to seeing if that helps with my set up.

It would be interesting to see if this fixes the issue for those in this issue too.

closed 2020-09-19 by mpromonet

vimagick, 2025-04-26:
`-c` works like charm!!! ✌