mpromonet/v4l2rtspserver #144: Segmentation fault in DeviceCaptureAccess<V4l2Capture>::getBufferSize()

Siggemada, 2019-07-10:
Ok, I am giving this a try again. Got my equipment back up and running. However, still having problem with Segmentation fault.

orangepiplus2e:~/v4l2rtspserver$ gdb v4l2rtspserver ./core -ex 'thread apply all where' -ex quit
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from v4l2rtspserver...done.
[New LWP 8053]
[New LWP 8052]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x004b1c9e in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()
[Current thread is 1 (Thread 0xaffa4450 (LWP 8053))]

Thread 2 (Thread 0xb6fe5260 (LWP 8052)):
#0  0xb6fb3410 in write () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xb6f83926 in log4cpp::FileAppender::_append(log4cpp::LoggingEvent const&) () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#2  0xb6f8a0e2 in log4cpp::Category::callAppenders(log4cpp::LoggingEvent const&) () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#3  0xb6f89f6e in log4cpp::Category::_logUnconditionally2(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#4  0xb6f8aeae in log4cpp::CategoryStream::flush() () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#5  0xb6f8af84 in log4cpp::CategoryStream::~CategoryStream() () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#6  0x004ad9bc in addSession(RTSPServer*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::list<ServerMediaSubsession*, std::allocator<ServerMediaSubsession*> > const&) ()
#7  0x004af6d8 in main ()

Thread 1 (Thread 0xaffa4450 (LWP 8053)):
#0  0x004b1c9e in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()
#1  0x004a5c6e in V4L2DeviceSource::getNextFrame() ()
#2  0x004a55aa in V4L2DeviceSource::thread() ()
#3  0x004a6542 in V4L2DeviceSource::threadStub(void*) ()
#4  0xb6fac5d8 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#5  0xb6c5b6fa in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Siggemada, 2019-07-10:
orangepiplus2e:~/v4l2rtspserver$ strace ./v4l2rtspserver -r -W2304 -H1536 -F2 /dev/video10
execve("./v4l2rtspserver", ["./v4l2rtspserver", "-r", "-W2304", "-H1536", "-F2", "/dev/video10"], [/* 30 vars */]) = 0
brk(NULL)                               = 0x18c4000
uname({sysname="Linux", nodename="orangepiplus2e", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6feb000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=40755, ...}) = 0
mmap2(NULL, 40755, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6fe1000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\341F\0\0004\0\0\0"..., 512) = 512
lseek(3, 105412, SEEK_SET)              = 105412
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1560) = 1560
lseek(3, 70056, SEEK_SET)               = 70056
read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
fstat64(3, {st_mode=S_IFREG|0755, st_size=106972, ...}) = 0
mmap2(NULL, 143952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6fa2000
mprotect(0xb6fb3000, 61440, PROT_NONE)  = 0
mmap2(0xb6fc2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6fc2000
mmap2(0xb6fc4000, 4688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6fc4000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\7\1\0004\0\0\0"..., 512) = 512
lseek(3, 157544, SEEK_SET)              = 157544
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 157184, SEEK_SET)              = 157184
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=158664, ...}) = 0
mmap2(NULL, 223216, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f6b000
mprotect(0xb6f90000, 65536, PROT_NONE)  = 0
mmap2(0xb6fa0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0xb6fa0000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libgroupsock.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\330L\0\0004\0\0\0"..., 512) = 512
lseek(3, 38000, SEEK_SET)               = 38000
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 37640, SEEK_SET)               = 37640
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=39120, ...}) = 0
mmap2(NULL, 154456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f45000
mprotect(0xb6f4e000, 61440, PROT_NONE)  = 0
mmap2(0xb6f5d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xb6f5d000
mmap2(0xb6f5f000, 47960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f5f000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libliveMedia.so.57", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200>\5\0004\0\0\0"..., 512) = 512
lseek(3, 706872, SEEK_SET)              = 706872
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 706512, SEEK_SET)              = 706512
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=707992, ...}) = 0
mmap2(NULL, 777200, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e87000
mprotect(0xb6f27000, 65536, PROT_NONE)  = 0
mmap2(0xb6f37000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xa0000) = 0xb6f37000
mmap2(0xb6f44000, 3056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f44000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libUsageEnvironment.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0D\21\0\0004\0\0\0"..., 512) = 512
lseek(3, 8652, SEEK_SET)                = 8652
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
lseek(3, 8304, SEEK_SET)                = 8304
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=9732, ...}) = 0
[... 154 lines cut ...]
	/dev/video10 support streaming
) = 90
ioctl(4, VIDIOC_G_FMT, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, fmt.pix={width=2304, height=1536, pixelformat=v4l2_fourcc('H', '2', '6', '4'), field=V4L2_FIELD_NONE, bytesperline=0, sizeimage=14155776, colorspace=V4L2_COLORSPACE_SRGB}}) = 0
[... cut ...]
mprotect(0xb639f000, 4096, PROT_NONE)   = 0
clone(child_stack=0xb6b9ef88, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xb6b9f4b8, tls=0xb6b9f910, child_tidptr=0xb6b9f4b8) = 8233
2019-07-10 17:04:46,569 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
uname( <ptrace(SYSCALL):No such process>
<... uname resumed> <unfinished ...>)   = ?
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault

Siggemada, 2019-07-10:
Without -r option:

orangepiplus2e:~/v4l2rtspserver$ strace ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
execve("./v4l2rtspserver", ["./v4l2rtspserver", "-W2304", "-H1536", "-F2", "/dev/video10"], [/* 30 vars */]) = 0
brk(NULL)                               = 0x1117000
uname({sysname="Linux", nodename="orangepiplus2e", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f32000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=40755, ...}) = 0
mmap2(NULL, 40755, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f28000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\341F\0\0004\0\0\0"..., 512) = 512
lseek(3, 105412, SEEK_SET)              = 105412
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1560) = 1560
lseek(3, 70056, SEEK_SET)               = 70056
read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
fstat64(3, {st_mode=S_IFREG|0755, st_size=106972, ...}) = 0
mmap2(NULL, 143952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ee9000
mprotect(0xb6efa000, 61440, PROT_NONE)  = 0
mmap2(0xb6f09000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6f09000
mmap2(0xb6f0b000, 4688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f0b000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\7\1\0004\0\0\0"..., 512) = 512
lseek(3, 157544, SEEK_SET)              = 157544
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 157184, SEEK_SET)              = 157184
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=158664, ...}) = 0
mmap2(NULL, 223216, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6eb2000
mprotect(0xb6ed7000, 65536, PROT_NONE)  = 0
mmap2(0xb6ee7000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0xb6ee7000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libgroupsock.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\330L\0\0004\0\0\0"..., 512) = 512
lseek(3, 38000, SEEK_SET)               = 38000
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 37640, SEEK_SET)               = 37640
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=39120, ...}) = 0
mmap2(NULL, 154456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e8c000
mprotect(0xb6e95000, 61440, PROT_NONE)  = 0
mmap2(0xb6ea4000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xb6ea4000
mmap2(0xb6ea6000, 47960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ea6000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libliveMedia.so.57", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200>\5\0004\0\0\0"..., 512) = 512
lseek(3, 706872, SEEK_SET)              = 706872
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 706512, SEEK_SET)              = 706512
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=707992, ...}) = 0
mmap2(NULL, 777200, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6dce000
mprotect(0xb6e6e000, 65536, PROT_NONE)  = 0
mmap2(0xb6e7e000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xa0000) = 0xb6e7e000
mmap2(0xb6e8b000, 3056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6e8b000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libUsageEnvironment.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0D\21\0\0004\0\0\0"..., 512) = 512
lseek(3, 8652, SEEK_SET)                = 8652
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
lseek(3, 8304, SEEK_SET)                = 8304
[... 184 lines cut ...]
mmap2(NULL, 14155776, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0x4380000) = 0xb19e7000
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=6, memory=V4L2_MEMORY_MMAP, m.offset=0x5100000, length=14155776, bytesused=14155776, flags=V4L2_BUF_FLAG_QUEUED, ...}) = 0
mmap2(NULL, 14155776, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0x5100000) = 0xb0c67000
[... cut ...]
mmap2(NULL, 8392704, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0xaf6e6000
mprotect(0xaf6e6000, 4096, PROT_NONE)   = 0
clone(child_stack=0xafee5f88, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xafee64b8, tls=0xafee6910, child_tidptr=0xafee64b8) = 8267
uname(2019-07-10 17:07:46,587 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
{sysname="Linux", nodename="orangepiplus2e", ...}) = 0
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault

mpromonet, 2019-07-10:
Hi Siggemada,

You may add `-f` to strace in order to follow threads.
I guess some incompatible libthread with your plateform, you may try using `-s` that disable the thread.

Best Regards,

Michel

Siggemada, 2019-07-10:
Thanks,

here is strace with f option:

orangepiplus2e:~/v4l2rtspserver$ strace -f ./v4l2rtspserver -w -W2304 -H1536 -F2 /dev/video10
execve("./v4l2rtspserver", ["./v4l2rtspserver", "-w", "-W2304", "-H1536", "-F2", "/dev/video10"], [/* 30 vars */]) = 0
brk(NULL)                               = 0x8b8000
uname({sysname="Linux", nodename="orangepiplus2e", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f48000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=40755, ...}) = 0
mmap2(NULL, 40755, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f3e000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\341F\0\0004\0\0\0"..., 512) = 512
lseek(3, 105412, SEEK_SET)              = 105412
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1560) = 1560
lseek(3, 70056, SEEK_SET)               = 70056
read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
fstat64(3, {st_mode=S_IFREG|0755, st_size=106972, ...}) = 0
mmap2(NULL, 143952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6eff000
mprotect(0xb6f10000, 61440, PROT_NONE)  = 0
mmap2(0xb6f1f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6f1f000
mmap2(0xb6f21000, 4688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f21000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\7\1\0004\0\0\0"..., 512) = 512
lseek(3, 157544, SEEK_SET)              = 157544
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 157184, SEEK_SET)              = 157184
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=158664, ...}) = 0
mmap2(NULL, 223216, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ec8000
mprotect(0xb6eed000, 65536, PROT_NONE)  = 0
mmap2(0xb6efd000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0xb6efd000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libgroupsock.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\330L\0\0004\0\0\0"..., 512) = 512
lseek(3, 38000, SEEK_SET)               = 38000
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 37640, SEEK_SET)               = 37640
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=39120, ...}) = 0
mmap2(NULL, 154456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ea2000
mprotect(0xb6eab000, 61440, PROT_NONE)  = 0
mmap2(0xb6eba000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xb6eba000
mmap2(0xb6ebc000, 47960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ebc000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libliveMedia.so.57", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200>\5\0004\0\0\0"..., 512) = 512
lseek(3, 706872, SEEK_SET)              = 706872
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 706512, SEEK_SET)              = 706512
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=707992, ...}) = 0
mmap2(NULL, 777200, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6de4000
mprotect(0xb6e84000, 65536, PROT_NONE)  = 0
mmap2(0xb6e94000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xa0000) = 0xb6e94000
mmap2(0xb6ea1000, 3056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ea1000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libUsageEnvironment.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0D\21\0\0004\0\0\0"..., 512) = 512
lseek(3, 8652, SEEK_SET)                = 8652
[... 196 lines cut ...]
ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=5, memory=V4L2_MEMORY_MMAP, m.offset=0, length=0, bytesused=0, flags=0, ...}) = 0
ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=6, memory=V4L2_MEMORY_MMAP, m.offset=0, length=0, bytesused=0, flags=0, ...}) = 0
ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=7, memory=V4L2_MEMORY_MMAP, m.offset=0, length=0, bytesused=0, flags=0, ...}) = 0
ioctl(4, VIDIOC_STREAMON, [V4L2_BUF_TYPE_VIDEO_CAPTURE]) = 0
[... cut ...]
[pid 10495] _newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 1 (in [4], left {tv_sec=0, tv_usec=999986})
[pid 10494] <... clone resumed> child_stack=0xafefbf88, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xafefc4b8, tls=0xafefc910, child_tidptr=0xafefc4b8) = 10495
[pid 10495] --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xaf17bcb8} ---
[pid 10495] +++ killed by SIGSEGV (core dumped) +++
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault

Siggemada, 2019-07-10:
.. and strace -f using -s option (also tried adding -r without luck):

orangepiplus2e:~/v4l2rtspserver$ strace -f ./v4l2rtspserver -s -w -W2304 -H1536 -F2 /dev/video10
execve("./v4l2rtspserver", ["./v4l2rtspserver", "-s", "-w", "-W2304", "-H1536", "-F2", "/dev/video10"], [/* 30 vars */]) = 0
brk(NULL)                               = 0xb1d000
uname({sysname="Linux", nodename="orangepiplus2e", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f8c000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=40755, ...}) = 0
mmap2(NULL, 40755, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f82000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\341F\0\0004\0\0\0"..., 512) = 512
lseek(3, 105412, SEEK_SET)              = 105412
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1560) = 1560
lseek(3, 70056, SEEK_SET)               = 70056
read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
fstat64(3, {st_mode=S_IFREG|0755, st_size=106972, ...}) = 0
mmap2(NULL, 143952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f43000
mprotect(0xb6f54000, 61440, PROT_NONE)  = 0
mmap2(0xb6f63000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6f63000
mmap2(0xb6f65000, 4688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f65000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\7\1\0004\0\0\0"..., 512) = 512
lseek(3, 157544, SEEK_SET)              = 157544
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 157184, SEEK_SET)              = 157184
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=158664, ...}) = 0
mmap2(NULL, 223216, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f0c000
mprotect(0xb6f31000, 65536, PROT_NONE)  = 0
mmap2(0xb6f41000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0xb6f41000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libgroupsock.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\330L\0\0004\0\0\0"..., 512) = 512
lseek(3, 38000, SEEK_SET)               = 38000
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 37640, SEEK_SET)               = 37640
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=39120, ...}) = 0
mmap2(NULL, 154456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ee6000
mprotect(0xb6eef000, 61440, PROT_NONE)  = 0
mmap2(0xb6efe000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xb6efe000
mmap2(0xb6f00000, 47960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f00000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libliveMedia.so.57", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200>\5\0004\0\0\0"..., 512) = 512
lseek(3, 706872, SEEK_SET)              = 706872
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 706512, SEEK_SET)              = 706512
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=707992, ...}) = 0
mmap2(NULL, 777200, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e28000
mprotect(0xb6ec8000, 65536, PROT_NONE)  = 0
mmap2(0xb6ed8000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xa0000) = 0xb6ed8000
mmap2(0xb6ee5000, 3056, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ee5000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libUsageEnvironment.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0D\21\0\0004\0\0\0"..., 512) = 512
lseek(3, 8652, SEEK_SET)                = 8652
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
lseek(3, 8304, SEEK_SET)                = 8304
[... 185 lines cut ...]
ioctl(4, VIDIOC_QUERYBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=6, memory=V4L2_MEMORY_MMAP, m.offset=0x5100000, length=14155776, bytesused=14155776, flags=V4L2_BUF_FLAG_QUEUED, ...}) = 0
mmap2(NULL, 14155776, PROT_READ|PROT_WRITE, MAP_SHARED, 4, 0x5100000) = 0xb0cc1000
[... cut ...]
write(1, "2019-07-10 21:44:54,848 [NOTICE]"..., 1462019-07-10 21:44:54,848 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:165
	Play this stream using the URL "rtsp://192.168.1.201:8554/unicast"
) = 146
rt_sigaction(SIGINT, {sa_handler=0x44c4dd, sa_mask=[INT], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6b87fd1}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
_newselect(5, [3 4], [], [], {tv_sec=0, tv_usec=0}) = 1 (in [4], left {tv_sec=0, tv_usec=0})
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xbdcecde0} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault

Siggemada, 2019-07-10:
... and gdb trace if helpful:

s@orangepiplus2e:~/v4l2rtspserver$ gdb v4l2rtspserver ./core -ex 'thread apply all where' -ex quit
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from v4l2rtspserver...done.
[New LWP 10517]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `./v4l2rtspserver -s -w -W2304 -H1536 -F2 /dev/video10'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00450c9e in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()

Thread 1 (Thread 0xb6f81260 (LWP 10517)):
#0  0x00450c9e in DeviceCaptureAccess<V4l2Capture>::getBufferSize() ()
#1  0x00444c6e in V4L2DeviceSource::getNextFrame() ()
#2  0x00444bb6 in V4L2DeviceSource::incomingPacketHandler() ()
#3  0x00445572 in V4L2DeviceSource::incomingPacketHandlerStub(void*, int) ()
#4  0xb6e02a04 in BasicTaskScheduler::SingleStep(unsigned int) () from /usr/lib/arm-linux-gnueabihf/libBasicUsageEnvironment.so.1
#5  0xb6e020a2 in BasicTaskScheduler0::doEventLoop(char volatile*) () from /usr/lib/arm-linux-gnueabihf/libBasicUsageEnvironment.so.1
#6  0x0044e76e in main ()

Siggemada, 2019-07-10:
And, just to let you now. This works fine:

v4l2-ctl -d /dev/video0 -vwidth=1920,height=1080,pixelformat=YUYV -p 5
sudo modprobe v4l2loopback video_nr=10
v4l2compress_h264 -W1920 -H1080 -F5 /dev/video0 /dev/video10

./v4l2rtspserver -W1920 -H1080 -F5 /dev/video10

So, I  think it has something to do with the resolution?

mpromonet, 2019-07-10:
Siggemada ,

Then using 1 thread it's the same.
The resolution is not big enough to overflow an int, may be a buffer overflow, becasue the getBufferSize just return the m_buffersize member.
Maybe building in debug and running under valgrind will bring more information.

Best Regards,

Michel.

Siggemada, 2019-07-11:
Could you walk me thru doing that?

mpromonet, 2019-07-12:
Hi Siggemada,

To build in debug, you could use `cmake -DCMAKE_BUILD_TYPE=Debug  . && make`.
Next to run with valgrind, just run `valgrind ./v4l2rtspserver ....`

Best Regards,
Michel.

Siggemada, 2019-07-12:
Ok, thanks. I am no programmer. Happy if I can help iron out this problem.

Here goes:

orangepiplus2e:~/v4l2rtspserver$ valgrind ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
==32726== Memcheck, a memory error detector
==32726== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==32726== Using Valgrind-3.12.0.SVN and LibVEX; rerun with -h for copyright info
==32726== Command: ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
==32726== 

valgrind: m_transtab.c:2459 (vgPlain_init_tt_tc): Assertion 'sizeof(TTEntryC) <= 88' failed.
Segmentation fault

mpromonet, 2019-07-27:
Hi,

valgrind crashed... good.
Did you made something like compiling with a `libc` and run it with another one ?

Best Regards,
Michel.

Siggemada, 2019-08-01:
Hi, no I just followed the build instructions.

Is this the version you ask for:
ldd --version
ldd (Debian GLIBC 2.28-10) 2.28

I am making/building on the same machine (32 bit). I did post some problems with a 64 BIT machine but never got that one to build.

mpromonet, 2019-08-01:
Hi Siggemada,
If valgrind cannot help, maybe some option can helps.
You could run in debug mode and print the value using `info locals`, `print this`, `print *this`
Best Regards,
Michel.

Siggemada, 2019-08-07:
After a git pull, it now behaves diffrently:

~/v4l2rtspserver$ valgrind ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
==2899== Memcheck, a memory error detector
==2899== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2899== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==2899== Command: ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
==2899== 
2019-08-07 22:42:57,655 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:605
	Create V4L2 Source.../dev/video10
2019-08-07 22:42:58,101 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:116
	driver:v4l2 loopback capabilities:85208003 mandatory:4000001
2019-08-07 22:42:58,107 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:118
	/dev/video10 support output
2019-08-07 22:42:58,113 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:119
	/dev/video10 support capture
2019-08-07 22:42:58,119 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:121
	/dev/video10 support read/write
2019-08-07 22:42:58,125 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:122
	/dev/video10 support streaming
2019-08-07 22:42:58,143 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:146
	/dev/video10:H264 size:2304x1536
2019-08-07 22:42:58,164 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
	fps:1/2
2019-08-07 22:42:58,169 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:229
	nbBuffer:8
2019-08-07 22:42:58,182 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:71
	Device /dev/video10 nb buffer:8
2019-08-07 22:42:58,319 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:628
	Create Source .../dev/video10
2019-08-07 22:42:58,639 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
2019-08-07 22:42:58,809 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.1.201:8554/unicast"
==2899== Warning: client switching stacks?  SP change: 0xc383cc8 --> 0xb603cc8
==2899==          to suppress, use: --max-stackframe=14155776 or greater
==2899== Warning: client switching stacks?  SP change: 0xc383cc8 --> 0xb603cc8
==2899==          to suppress, use: --max-stackframe=14155776 or greater
==2899== Warning: client switching stacks?  SP change: 0xc383cc8 --> 0xb603cc8
==2899==          to suppress, use: --max-stackframe=14155776 or greater
==2899==          further instances of this message will not be shown.

Siggemada, 2019-08-07:
And it works! Great! Thanks!!

Output:
==2899== Use of uninitialised value of size 4
==2899==    at 0x4A99754: _itoa_word (_itoa.c:179)
==2899==    by 0x4A9CB23: vfprintf (vfprintf.c:1637)
==2899==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==2899==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==2899==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==2899==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==2899==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==2899==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==2899==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==2899==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==2899==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==2899==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==2899== 
==2899== Conditional jump or move depends on uninitialised value(s)
==2899==    at 0x4A9975A: _itoa_word (_itoa.c:179)
==2899==    by 0x4A9CB23: vfprintf (vfprintf.c:1637)
==2899==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==2899==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==2899==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==2899==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==2899==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==2899==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==2899==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==2899==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==2899==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==2899==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==2899== 
==2899== Conditional jump or move depends on uninitialised value(s)
==2899==    at 0x4A9C136: vfprintf (vfprintf.c:1637)
==2899==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==2899==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==2899==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==2899==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==2899==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==2899==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==2899==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==2899==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==2899==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==2899==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==2899==    by 0x16F3EB: BasicTaskScheduler0::doEventLoop(char volatile*) (BasicTaskScheduler0.cpp:80)
==2899== 
==2899== Conditional jump or move depends on uninitialised value(s)
==2899==    at 0x4A9C1DC: vfprintf (vfprintf.c:1637)
==2899==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==2899==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==2899==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==2899==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==2899==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==2899==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==2899==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==2899==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==2899==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==2899==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==2899==    by 0x16F3EB: BasicTaskScheduler0::doEventLoop(char volatile*) (BasicTaskScheduler0.cpp:80)
==2899== 
==2899== Conditional jump or move depends on uninitialised value(s)
==2899==    at 0x4A9C238: vfprintf (vfprintf.c:1637)
==2899==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==2899==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==2899==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
[... cut ...]
==2899==    by 0x18399F: RTCPInstance::onExpire1() (RTCP.cpp:1244)
==2899==    by 0x1830BB: RTCPInstance::onExpire(RTCPInstance*) (RTCP.cpp:975)
==2899==    by 0x16F983: AlarmHandler::handleTimeout() (BasicTaskScheduler0.cpp:34)
==2899==    by 0x170465: DelayQueue::handleAlarm() (DelayQueue.cpp:187)
==2899==    by 0x16EDBF: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:212)
==2899==    by 0x16F3EB: BasicTaskScheduler0::doEventLoop(char volatile*) (BasicTaskScheduler0.cpp:80)
==2899==    by 0x16921B: main (main.cpp:745)
==2899==

Siggemada, 2019-08-07:
2019-08-07 22:48:55,164 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
2019-08-07 22:48:55,169 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
^CSIGINT
2019-08-07 22:49:04,467 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:746
	Exiting....
==2899== Warning: invalid file descriptor -1 in syscall close()
==2899== 
==2899== HEAP SUMMARY:
==2899==     in use at exit: 1,068,201 bytes in 49 blocks
==2899==   total heap usage: 95,775 allocs, 95,726 frees, 199,437,061 bytes allocated
==2899== 
==2899== LEAK SUMMARY:
==2899==    definitely lost: 125 bytes in 1 blocks
==2899==    indirectly lost: 0 bytes in 0 blocks
==2899==      possibly lost: 144 bytes in 1 blocks
==2899==    still reachable: 1,067,932 bytes in 47 blocks
==2899==         suppressed: 0 bytes in 0 blocks
==2899== Rerun with --leak-check=full to see details of leaked memory
==2899== 
==2899== For counts of detected and suppressed errors, rerun with: -v
==2899== Use --track-origins=yes to see where uninitialised values come from
==2899== ERROR SUMMARY: 109 errors from 11 contexts (suppressed: 0 from 0)

Siggemada, 2019-08-07:
... um ok but not without debug?

cmake . && make
 ./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10
2019-08-07 22:54:46,728 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:605
	Create V4L2 Source.../dev/video10
2019-08-07 22:54:46,728 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:116
	driver:v4l2 loopback capabilities:85208003 mandatory:4000001
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:118
	/dev/video10 support output
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:119
	/dev/video10 support capture
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:121
	/dev/video10 support read/write
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:122
	/dev/video10 support streaming
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:146
	/dev/video10:H264 size:2304x1536
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
	fps:1/2
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:229
	nbBuffer:8
2019-08-07 22:54:46,729 [NOTICE] - /home/xxxx/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:71
	Device /dev/video10 nb buffer:8
2019-08-07 22:54:46,818 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:628
	Create Source .../dev/video10
2019-08-07 22:54:46,819 [NOTICE] - /home/xxxx/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.1.201:8554/unicast"
2019-08-07 22:54:46,819 [NOTICE] - /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
Segmentation fault

Siggemada, 2019-08-07:
No, not debug releted? It works running with valgrind?

xxxx8 --> 0xb603cc8
==4870==          to suppress, use: --max-stackframe=14155776 or greater
==4870== Warning: client switching stacks?  SP change: 0xc383cc8 --> 0xb603cc8
==4870==          to suppress, use: --max-stackframe=14155776 or greater
==4870== Warning: client switching stacks?  SP change: 0xc383cc8 --> 0xb603cc8
==4870==          to suppress, use: --max-stackframe=14155776 or greater
==4870==          further instances of this message will not be shown.
==4870== Use of uninitialised value of size 4
==4870==    at 0x4A99754: _itoa_word (_itoa.c:179)
==4870==    by 0x4A9CB23: vfprintf (vfprintf.c:1637)
==4870==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==4870==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==4870==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==4870==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==4870==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==4870==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==4870==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==4870==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==4870==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==4870==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==4870== 
==4870== Conditional jump or move depends on uninitialised value(s)
==4870==    at 0x4A9975A: _itoa_word (_itoa.c:179)
==4870==    by 0x4A9CB23: vfprintf (vfprintf.c:1637)
==4870==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==4870==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==4870==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==4870==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==4870==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==4870==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==4870==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==4870==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==4870==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==4870==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==4870== 
==4870== Conditional jump or move depends on uninitialised value(s)
==4870==    at 0x4A9C136: vfprintf (vfprintf.c:1637)
==4870==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==4870==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==4870==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==4870==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==4870==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==4870==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==4870==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==4870==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==4870==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==4870==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
==4870==    by 0x16F3EB: BasicTaskScheduler0::doEventLoop(char volatile*) (BasicTaskScheduler0.cpp:80)
==4870== 
==4870== Conditional jump or move depends on uninitialised value(s)
==4870==    at 0x4A9C1DC: vfprintf (vfprintf.c:1637)
==4870==    by 0x4AB1291: vsprintf (iovsprintf.c:42)
==4870==    by 0x4AA0CE1: sprintf (sprintf.c:32)
==4870==    by 0x180D5D: OnDemandServerMediaSubsession::setSDPLinesFromRTPSink(RTPSink*, FramedSource*, unsigned int) (OnDemandServerMediaSubsession.cpp:455)
==4870==    by 0x18015D: OnDemandServerMediaSubsession::sdpLines() (OnDemandServerMediaSubsession.cpp:76)
==4870==    by 0x18D9FB: ServerMediaSession::generateSDPDescription() (ServerMediaSession.cpp:240)
==4870==    by 0x187C83: RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE(char const*, char const*, char const*) (RTSPServer.cpp:345)
==4870==    by 0x188B0F: RTSPServer::RTSPClientConnection::handleRequestBytes(int) (RTSPServer.cpp:757)
==4870==    by 0x17632B: GenericMediaServer::ClientConnection::incomingRequestHandler() (GenericMediaServer.cpp:246)
==4870==    by 0x1762D3: GenericMediaServer::ClientConnection::incomingRequestHandler(void*, int) (GenericMediaServer.cpp:239)
==4870==    by 0x16EC47: BasicTaskScheduler::SingleStep(unsigned int) (BasicTaskScheduler.cpp:171)
[... cut ...]
==4870==    indirectly lost: 0 bytes in 0 blocks
==4870==      possibly lost: 144 bytes in 1 blocks
==4870==    still reachable: 1,071,179 bytes in 47 blocks
==4870==         suppressed: 0 bytes in 0 blocks
==4870== Rerun with --leak-check=full to see details of leaked memory
==4870== 
==4870== For counts of detected and suppressed errors, rerun with: -v
==4870== Use --track-origins=yes to see where uninitialised values come from
==4870== ERROR SUMMARY: 37 errors from 11 contexts (suppressed: 0 from 0)

Siggemada, 2019-08-07:
gdb trace:

gdb v4l2rtspserver ./core -ex 'thread apply all where' -ex quit
GNU gdb (Debian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from v4l2rtspserver...done.
[New LWP 7300]
[New LWP 7299]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `./v4l2rtspserver -W2304 -H1536 -F2 /dev/video10'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  DeviceCaptureAccess<V4l2Capture>::getBufferSize (this=<error reading variable: Cannot access memory at address 0xaf244cbc>) at /home/xxxx/v4l2rtspserver/inc/DeviceInterface.h:43
43			virtual unsigned long getBufferSize()                { return m_device->getBufferSize(); }
[Current thread is 1 (Thread 0xaffc5450 (LWP 7300))]

Thread 2 (Thread 0xb6f05490 (LWP 7299)):
#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0xb6ed5152 in __lll_lock_wait (futex=futex@entry=0x144051c, private=0) at lowlevellock.c:46
#2  0xb6ecf98c in __GI___pthread_mutex_lock (mutex=0x144051c) at pthread_mutex_lock.c:80
#3  0xb6ea9f56 in log4cpp::Category::callAppenders(log4cpp::LoggingEvent const&) () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#4  0xb6ea9e0e in log4cpp::Category::_logUnconditionally2(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#5  0xb6eaad6e in log4cpp::CategoryStream::flush() () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#6  0xb6eaadf0 in log4cpp::CategoryStream::~CategoryStream() () from /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5
#7  0x004673dc in addSession (rtspServer=0x1440f20, sessionName="unicast", subSession=std::__cxx11::list = {...}) at /home/xxxx/v4l2rtspserver/src/main.cpp:150
#8  0x00469186 in main (argc=5, argv=0xbedf6584) at /home/xxxx/v4l2rtspserver/src/main.cpp:738

Thread 1 (Thread 0xaffc5450 (LWP 7300)):
#0  DeviceCaptureAccess<V4l2Capture>::getBufferSize (this=<error reading variable: Cannot access memory at address 0xaf244cbc>) at /home/xxxx/v4l2rtspserver/inc/DeviceInterface.h:43
#1  0x0045f1ce in V4L2DeviceSource::getNextFrame (this=0x1441f78) at /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:199
#2  0x0045eafe in V4L2DeviceSource::thread (this=0x1441f78) at /home/xxxx/v4l2rtspserver/src/DeviceSource.cpp:105
#3  0x0045fb12 in V4L2DeviceSource::threadStub (clientData=0x1441f78) at /home/xxxx/v4l2rtspserver/inc/DeviceSource.h:77
#4  0xb6ecdbbe in start_thread (arg=0x87789b0) at pthread_create.c:486
#5  0xb6c8062c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

closed 2020-09-19 by mpromonet