mpromonet/v4l2rtspserver #146: Intermediate v4l2loopback device - not working

careyer, 2019-07-24:
Hi @mpromonet ,

I want to use a intermediate video device like you describe it in the workflow because my camera does not support h264 encoding.

Here is a simple test-setup that I try to run on a Raspberry Pi 3:
```
#Create v4l2loopback devices /dev/video5 and /dev/video6
sudo modprobe v4l2loopback video_nr5,6 

#Write raw testvideo to /dev/video5
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video5 &

#Now try to compress rawvideo in /de/video5 with h264 and make it available at /dev/video6
v4l2compress_omx /dev/video5 /dev/video6
```

Now this gives me this error:
```
log level:500

[NOTICE] src/V4l2Device.cpp:117
        driver:v4l2 loopback capabilities:85208003 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:119
        /dev/video5 support output
[NOTICE] src/V4l2Device.cpp:120
        /dev/video5 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video5 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video5 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video5:YUYV size:320x240
[ERROR] src/V4l2Device.cpp:190
        Cannot set format for device:/dev/video5 Invalid argument
[ERROR] src/V4l2Device.cpp:67
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
        Cannot init device:/dev/video5
[WARN] src/v4l2compress_omx.cpp:109
        Cannot create V4L2 capture interface for device:/dev/video5
```
Do you have any ideas what I can try next?
The above should be the simplest test that one can think of, right?

mpromonet, 2019-07-24:
Hi careyer,

This seems related to project v4l2tools.
v4l2compress_omx  is expecting V4L2_PIX_FMT_YUV420 (YU12), not YUYV.
You should be able to convert with `v4l2convert_yuv`

Best Regards,
Michel.

mpromonet, 2019-10-25:
Hi careyer,
I made a modification, now v4l2compress_omx convert YUV to YU12 if needed internally.
Best Regards,
Michel.

closed 2019-10-25 by mpromonet

careyer, 2019-10-27:
Many thanks @mpromonet !!