mpromonet/v4l2rtspserver #149: crash with raspi camera after a couple of days

robertmuth, 2019-08-30:
The  server crashes after a day or so.
There is no stack trace - all I see on the console is:

MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 1140 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:1216 bufferSize:201187
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 1216 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:2032 bufferSize:202003
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 2032 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:1540 bufferSize:201511
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 1540 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:1747 bufferSize:201718
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 1747 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:1630 bufferSize:201601
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 1630 bytes!
send file:index.html
send file:index.html
send file:index.html
send file:index.html
send file:snap
terminate called after throwing an instance of 'std::ios_base::failure[abi:cxx11]'
  what():  basic_filebuf::underflow error reading the file: iostream error
Aborted

I am happy to run some more experiments

BumpyClock, 2019-08-30:
Can confirm, I see the same issue.

mpromonet, 2019-08-31:
Hi @robertmuth, @BumpyClock 

In order to get more information, you probably need to enable core file or run using a debugger.
If it is a memory leak or corruption, valgrind can also helps.

Best Regards,

Michel.

closed 2020-09-19 by mpromonet