mpromonet/v4l2rtspserver #150: problems with 1080p streaming

robertmuth, 2019-09-02:
This maybe a raspi hw issue but it would be nice if v4l2rtspserver could help diagnose the problem

720p video looks great 
./v4l2rtspserver -S1 -p8080  -F15 -H 720 -W 1440  
when viewed with 
vlc rtsp://192.168.1.148:8554/unicast

1080p video looks terrible
( lots of noise and the image seem to be cropped to a small area)
./v4l2rtspserver -S1 -p8080  -F15 -H 1080 -W 1920  

I did increase GPU memory to 256M
(gpu_mem=256)

And added options to the /bcm2835_v4l2 modules
cat /sys/module/bcm2835_v4l2/parameters/* 
0
2464
3280
-1,-1

mpromonet, 2019-09-02:
Hi @robertmuth,

It could be just a network overflow, you could try to connect in tcp mode using:
`vlc --rtsp-tcp rtsp://192.168.1.148:8554/unicast`

Best Regards,

Michel.

TechMasterJoe, 2019-09-03:
The imx219 aka v2 camera will crop in 1080p mode..
1644x1232 is 4x4 binning but full sensor.

I use 1080p like a 2x digital zoom...

But also have problems with 1080p even with my nas

closed 2020-09-19 by mpromonet