mpromonet/v4l2rtspserver #151: can not change resolution

rayzie, 2019-09-06:
I have a question.

I want to change the RTSP transmission resolution.

My camera supports v4l2 and streaming to 1920x1080 works fine.

`v4l2-ctl -d /dev/video0 --list-formats-ext`

When using the above command,

`ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'H264' (compressed)
        Name        : H.264
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (30.000 fps)`

I ran RTSP streaming using the following command:
`sudo ./h264_v4l2_rtspserver/v4l2rtspserver -W 1280 -H 720 -P /dev/video0`

However, the 1280x720 resolution does not change.

The 1920x1080 and 3840x2160 is normally resized.

Can you solve this?

rayzie, 2019-09-06:
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:602
        Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:117
        driver:uvcvideo capabilities:84200001 mandatory:4000001
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:147
        /dev/video0:H264 size:1280x720
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:229
        fps:1/30
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:230
        nbBuffer:0
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:71
        Device /dev/video0 nb buffer:10
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:625
        Create Source .../dev/video0
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:93
        begin thread
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:150
        Play this stream using the URL "rtsp://172.20.2.95:8554/"
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131

rayzie, 2019-09-06:
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:602
        Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:117
        driver:uvcvideo capabilities:84200001 mandatory:4000001
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:147
        /dev/video0:H264 size:1280x720
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:229
        fps:1/30
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2Device.cpp:230
        nbBuffer:0
[NOTICE] /home/pi/h264_v4l2_rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:71
        Device /dev/video0 nb buffer:10
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:625
        Create Source .../dev/video0
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:93
        begin thread
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/main.cpp:150
        Play this stream using the URL "rtsp://myip:8554/"
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/h264_v4l2_rtspserver/src/DeviceSource.cpp:131

mpromonet, 2019-10-25:
Hi rayzie,
Maybe there is some error in the output log ? or maybe increasing the log level ?
Best Regards,
Michel.

closed 2020-04-12 by mpromonet