mpromonet/v4l2rtspserver #153: Can't start v4l2rtspserver because of: mmap: No such device.

noliveleger, 2019-09-14:
Hi there, 

I've followed the steps for [RPi install](https://github.com/mpromonet/v4l2rtspserver/wiki/Setup-on-Pi "") but I can't get make it work with an USB UVC compatible Camera. 

I receive this error: 
``` 
pi@raspberry:~$ v4l2rtspserver -F 30 -H 720 -W 1280 -P 8555 /dev/video0
log level:500

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:607
        Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:116
        driver:uvc capabilities:5000001 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:119
        /dev/video0 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:121
        /dev/video0 support read/write
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:122
        /dev/video0 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:146
        /dev/video0:MJPG size:1280x720
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:185
        Cannot set format:H264 for device:/dev/video0 Invalid argument
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:203
        /dev/video0:MJPG size:1280x720 bufferSize:1843789
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:224
        fps:333333/10000000
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
        nbBuffer:4
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device /dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
mmap: No such device
        Device /dev/video0 nb buffer:10
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
munmap: Invalid argument
pi@raspberry:~$
```
Any idea why ? 

I have a working configuration which another RPi and the same camera model, but different OS. I guess, it may be related to some libraries which I don't have and different versions? But which ones?
The non-working is with `Raspbian stretch` but the working one uses `Raspbian buster`. 
I can't swap devices because I don't have access physically at the non working one. Everything is done by ssh. 

Any help would be appreciated.
Thank you.

mpromonet, 2019-09-15:
Hi noliveleger,
Probably you are right, it may be a problem of preloaded libraries, there was some issues about that.
Using `ldd ./v4l2rtspserver` you can get the list and compare between the 2 RPi.
Otherwise you can run with `-r` in order to use read interface instead of mmap.
Best Regards,
Michel.

noliveleger, 2019-09-15:
@mpromonet , that was it. Thank you.
I had to export `LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so`, even if it's already in the `pi` user env (on both RPi).

```
pi@raspberry:~ $ export LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so
pi@raspberry:~ $ v4l2rtspserver -F 30 -H 720 -W 1280 -P 8555 /dev/video0
```
Maybe there's a difference how the env is loaded between `Stretch` and `Buster`. 
BTW, thank you very much for your work. Thanks to it, Motion recordings fps jumped from 2 to ~25. (using the rstp stream instead of mjpeg).

closed 2019-09-15 by noliveleger