mpromonet/v4l2rtspserver #156: v4l2compress_omx + v4l2rtspserver + VLC player does not work

jasaw, 2019-09-21:
I have just cloned from the latest master from both v4l2rtspserver and v4l2tools repositories, compiled them and tried the example, but can't seem to get the examples to work. From the console output, it looks like v4l2compress_omx and v4l2rtspserver are running OK, but I don't see any video on my VLC player. The only clue I see is this error message: `RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!`
How do I get this to work? Any help would be appreciated.

Here's the console output.
>sudo ./v4l2compress_omx /dev/video0 /dev/video90
log level:500
[NOTICE] src/V4l2Device.cpp:117
        driver:bm2835 mmal capabilities:85200005 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:120
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video0:JPEG size:1024x768
[NOTICE] src/V4l2Device.cpp:208
        /dev/video0:YU12 size:640x480 bufferSize:460800
[NOTICE] src/V4l2Device.cpp:229
        fps:1/10
[NOTICE] src/V4l2Device.cpp:230
        nbBuffer:1
[NOTICE] src/V4l2MmapDevice.cpp:71
        Device /dev/video0 nb buffer:10
[NOTICE] src/V4l2Device.cpp:117
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
[NOTICE] src/V4l2Device.cpp:119
        /dev/video90 support output
[NOTICE] src/V4l2Device.cpp:120
        /dev/video90 support capture
[NOTICE] src/V4l2Device.cpp:122
        /dev/video90 support read/write
[NOTICE] src/V4l2Device.cpp:123
        /dev/video90 support streaming
[NOTICE] src/V4l2Device.cpp:147
        /dev/video90:BGR4 size:0x0
[NOTICE] src/V4l2Device.cpp:208
        /dev/video90:H264 size:640x480 bufferSize:1228800
[NOTICE] src/V4l2MmapDevice.cpp:71
        Device /dev/video90 nb buffer:8
[NOTICE] src/v4l2compress_omx.cpp:122
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 160x64 160x64 @1966080 20
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 640x480 640x480 @1966080 20
OMX_SetParameter(OMX_IndexParamVideoPortFormat) for video_encode:201...
OMX_SetParameter(OMX_IndexParamBrcmVideoAVCInlineHeaderEnable) for video_encode:201...
Current Bitrate=10000000
encode to idle...
enabling port buffers for 200...
enabling port buffers for 201...
encode to executing...
        Start Capturing from /dev/video0
[NOTICE] src/v4l2compress_omx.cpp:138
        Start Compressing /dev/video0 to /dev/video90

>./v4l2rtspserver -v -F 5 /dev/video90
log level:600
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:607
        Create V4L2 Source.../dev/video90
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:116
        driver:v4l2 loopback capabilities:85208003 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:118
        /dev/video90 support output
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:119
        /dev/video90 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:121
        /dev/video90 support read/write
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:122
        /dev/video90 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:146
        /dev/video90:H264 size:640x480
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:203
        /dev/video90:H264 size:640x480 bufferSize:1228800
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:224
        fps:1/5
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
        nbBuffer:8
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device /dev/video90
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
        Device /dev/video90 nb buffer:8
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:0 size:1228800 offset:0
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:1 size:1228800 offset:1228800
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:2 size:1228800 offset:2457600
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:3 size:1228800 offset:3686400
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:4 size:1228800 offset:4915200
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:5 size:1228800 offset:6144000
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:6 size:1228800 offset:7372800
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video90 buffer idx:7 size:1228800 offset:8601600
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:630
        Create Source .../dev/video90
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:150
        Play this stream using the URL "rtsp://10.10.59.235:8554/unicast"
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
        intv_sec:1569045352 fps:1 bandwidth:9600kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
        outtv_sec:1569045357 fps:1 bandwidth:0kbps
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames

mpromonet, 2019-09-21:
Hi,
It seems /dev/video0 is the bm2835 module that support H264 capture.
In this case you can access to the H264 stream direclty using : 

     v4l2rtspserver /dev/video0

Best Regards,
Michel.

jasaw, 2019-09-21:
@mpromonet Thank you for responding so quickly. My goal is to actually do something very similar to v4l2compress_omx, but instead of using /dev/video0 as the source, I'm feeding it from somewhere else. If I can get v4l2compress_omx to work, then I can get my modified version to work.

If you can give me any pointers on where or what to look for, I would be very appreciative.

jasaw, 2019-09-21:
I've figured it out on my modified version. I just need to re-send SPS and PPS timing in the NAL unit every now and then.

closed 2019-09-21 by jasaw