mpromonet/v4l2rtspserver #157: it's better to add docker-compose.yaml for docker

sndnvaps, 2019-10-19:
add docker-compose.yaml for docker make it easy to use.

mpromonet, 2019-10-19:
maybe

sndnvaps, 2019-10-19:
this is my docker-compose-armhf.yaml
```yaml
version: "3.5"
services:
  v4l2rtspserver:
    restart: unless-stopped
    image: mpromonet/v4l2rtspserver
    container_name: rpi-v4l2rtspserver-armhf
    ports:
      - "8554:8554"
    devices:
      - "/dev/video0:/dev/video0"
    command:
      - '-F25 -H640 -W480' 
```

closed 2019-10-26 by sndnvaps

konstantinj, 2020-02-08:
@sndnvaps Did you do something else to get it working with docker-compose?
When using the command like you it doesn't work at all. I need to write it in the `["", ""]` form. That way the container starts successfully but I still can't access the stream. When doing the same settings via normal docker it works. I've also tried stdin_open and tty true. No success either.

sndnvaps, 2020-02-08:
@konstantinj i don't use it .. 
i drop it and use ffmpeg to parse use webcam stream...