mpromonet/v4l2rtspserver #159: v4l2rtspserver stops responding after a while (random time)

noliveleger, 2019-10-28:
Hi, 
I'm using `v4l2rtspserver` to create a RSTP stream for Motion (TCP).
It gives me way more fps than using `uv4l` MJPEG stream in Motion. 
Hardware: 
The camera is a MJPEG USB 1080P camera installed 
Raspberry PI 3B+ (Strech)
Wired on Gigabit network.

I've created a service like this:
```
[Unit]
Description=Streamer daemon
After=network.target
StartLimitIntervalSec=30

[Service]
WorkingDirectory=/home/pi
Environment="PATH=/home/pi/.local/bin:/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:"
Environment="LD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so"
Type=simple
Restart=on-failure
RestartSec=1
User=pi
ExecStart=/usr/local/bin/v4l2rtspserver -F 30 -H 720 -W 1280 -P 8555 /dev/video0

[Install]
WantedBy=multi-user.target
```

After a while (usually several hours), Motion is not able to read the feed from `v4l2rtspserver`. 
Trying to access the feed with VLC from my laptop doesn't work either. 
If I stop the service and restart it, sometimes it works, sometimes it does not. Sometimes when it works, it works only for few minutes and I don't have other choices than rebooting.
I don't have so much info in the logs except that:
```
Oct 28 09:17:15 doorbell v4l2rtspserver[452]: log level:500
Oct 28 09:17:23 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:07:56 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:08:36 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:08:57 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:09:18 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:09:45 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:10:05 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:10:26 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:10:47 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:11:05 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
Oct 28 10:11:24 doorbell v4l2rtspserver[452]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
```
If I stop the service and I try to access the `MJPEG` stream of `uv4l` server, I can see the video. 
There are several options with `v4l2rtspserver` but I don't understand the advantage of each one against others

```
-r       : V4L2 capture using read interface (default use memory mapped buffers)
-w       : V4L2 capture using write interface (default use memory mapped buffers)
-B       : V4L2 capture using blocking mode (default use non-blocking mode)
-s       : V4L2 capture using live555 mainloop (default use a separated reading thread)
```

I guess, it uses `-s` by default?

Any idea what I could to get a stable feed?
Thank you in advance.

mpromonet, 2019-10-28:
Hi noliveleger,
Did you tried with the official driver bcm2835-v4l2 ?
As uv4l is closed source, it is difficult to debug it.
Best Regards,
Michel.

noliveleger, 2019-10-28:
@mpromonet I thought `bcm2835-v4l2` was only for Camera Serial Interface. Not for USB cameras.

mpromonet, 2019-10-28:
for usb cameras you can use uvcvideo

noliveleger, 2019-10-29:
I'm gonna try with `uvcvideo`.
Do you think I should start `v4l2rtspserver` with one of the options I mentioned above (`-r`, `-w` ...)

noliveleger, 2019-10-30:
I was not able to make the camera work with `uvcvideo`. 
No I didn't try `strace` or `gdb`but I will.
I'm gonna dig a little bit further when I can spend time on this and I'll give some feedback here.
Feel free to close this issue. I can still leave the comments in this issue even if it's closed.
Thank you for the help.

noliveleger, 2019-11-04:
@mpromonet, it's not related to `v4l2rtspserver`. 
I finally made it work with `uvcvideo`. 

I got this error in `/var/log/messages`
```
Nov  4 16:59:23 doorbell kernel: [284741.124596] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124668] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124704] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124726] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124748] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124770] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124787] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124807] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124826] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124843] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124863] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124886] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124905] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124924] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124944] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124957] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.124971] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.154853] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.154875] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:23 doorbell kernel: [284741.154890] NYET/NAK/ACK/other in non-error case, 0x00000002
Nov  4 16:59:24 doorbell kernel: [284741.264789] usb 1-1: USB disconnect, device number 2
Nov  4 16:59:24 doorbell kernel: [284741.264796] usb 1-1.1: USB disconnect, device number 3
Nov  4 16:59:24 doorbell kernel: [284741.264802] usb 1-1.1.1: USB disconnect, device number 5
Nov  4 16:59:24 doorbell kernel: [284741.265062] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265073] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265089] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265096] lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265281] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000098. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265288] lan78xx 1-1.1.1:1.0 eth0: Failed to write register index 0x00000098. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.265312] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -19
Nov  4 16:59:24 doorbell kernel: [284741.266023] device eth0 left promiscuous mode
Nov  4 16:59:24 doorbell kernel: [284741.319698] usb 1-1.3: USB disconnect, device number 4
Nov  4 16:59:24 doorbell kernel: [284741.664833] Indeed it is in host mode hprt0 = 00001501
Nov  4 16:59:24 doorbell kernel: [284741.874667] usb 1-1: new high-speed USB device number 6 using dwc_otg
Nov  4 16:59:24 doorbell kernel: [284741.874766] Indeed it is in host mode hprt0 = 00001101
Nov  4 16:59:24 doorbell kernel: [284742.114867] usb 1-1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
Nov  4 16:59:24 doorbell kernel: [284742.114876] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Nov  4 16:59:24 doorbell kernel: [284742.115362] hub 1-1:1.0: USB hub found
Nov  4 16:59:24 doorbell kernel: [284742.115417] hub 1-1:1.0: 4 ports detected
Nov  4 16:59:25 doorbell kernel: [284742.444688] usb 1-1.1: new high-speed USB device number 7 using dwc_otg
Nov  4 16:59:25 doorbell kernel: [284742.574882] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
Nov  4 16:59:25 doorbell kernel: [284742.574892] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Nov  4 16:59:25 doorbell kernel: [284742.575376] hub 1-1.1:1.0: USB hub found
Nov  4 16:59:25 doorbell kernel: [284742.575435] hub 1-1.1:1.0: 3 ports detected
Nov  4 16:59:25 doorbell kernel: [284742.674662] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
Nov  4 16:59:25 doorbell kernel: [284742.851954] usb 1-1.3: New USB device found, idVendor=05a3, idProduct=9422, bcdDevice= 1.00
Nov  4 16:59:25 doorbell kernel: [284742.851964] usb 1-1.3: New USB device strings: Mfr=2, Product=1, SerialNumber=3
Nov  4 16:59:25 doorbell kernel: [284742.851969] usb 1-1.3: Product: USB 2.0 Camera
Nov  4 16:59:25 doorbell kernel: [284742.851973] usb 1-1.3: Manufacturer: Sonix Technology Co., Ltd.
Nov  4 16:59:25 doorbell kernel: [284742.851978] usb 1-1.3: SerialNumber: SN0001
Nov  4 16:59:25 doorbell kernel: [284742.854155] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (05a3:9422)
Nov  4 16:59:25 doorbell kernel: [284742.894436] uvcvideo: Unable to create debugfs 1-8 directory.
[... cut ...]
Nov  4 16:59:26 doorbell kernel: [284743.758074] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
Nov  4 16:59:26 doorbell kernel: [284743.758709] libphy: lan78xx-mdiobus: probed
Nov  4 16:59:26 doorbell kernel: [284743.766291] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64
```
It seems that the USB ports are disconnected. 
I've to figure out why. 
Maybe the PoE board doesn't give enough current. Maybe the cam saturates USB bandwidth.

closed 2019-11-04 by noliveleger

Raphael909, 2020-09-08:
Hello, I was wondering if this issue have been taken care off?. I am still seeing this problem with raspberry Pi zero w and Logitech c270 usb webcam.
As always , thank you for the effort and fine work

mpromonet, 2020-09-08:
Hi Raphael909,

This may be linked to V4L2 driver, and uv4l is closed source.
In order to separate the origin of the hang, you may run capture with v4l2-ctl or some tools like this.
When it hangs, you can run strace to see what is going on.

Best Regards,
Michel.