mpromonet/v4l2rtspserver #160: Can't play HLS stream

frankerooney, 2019-11-28:
Hi. Compiled from latest source, and the program starts fine. A few warnings in there, but no showstoppers.

Running the program with switches -v -S /dev/video0,default says that the following streams are accessible:

	Play this stream using the URL "rtsp://192.168.2.82:8554/ts"
	HLS       http://192.168.2.82:8554/ts.m3u8
	MPEG-DASH http://192.168.2.82:8554/ts.mpd
	Play this stream using the URL "rtsp://192.168.2.82:8554/unicast"

The only play "type" that seems to work though is rtsp://IP:8554/unicast with VLC though. I'm most interesting in trying the http streams, but accessing the ts.m3u8 and ts.mpd suffixes produce a 405 error. 
Going to the root and looking in the index.html, it references the url /getStreamList?streamList, and if I use curl to access that, it comes back with an empty list :

curl -L http://192.168.2.82:8554/getStreamList?streamList
var streamList=[
]

This is also supported by the console in Chrome, where it tries to connect to http://192.168.2.82:8554/undefined.m3u8 because the first element [0] in the returned list is essentially empty. This produces a 405 error and the browser doesn't get any further,

Am I missing something or doing something stupid?

Kind Regards
Andy

mpromonet, 2019-11-29:
Hi frankerooney,
This seems correct.
However HLS works only with H264/H265, are you using this capture format ?
Best Regards,
Michel.

frankerooney, 2019-11-29:
Thank you Michel,
  That makes sense for the HLS stream definitely - it works, without sound for some reason, if I use the loopback h264 compression module. Not sure about the dash stream, I may have the wrong program to play that; should the rtsp://ip:8554/ts stream also work? The unicast one does, but this doesn't seem to.
In the end I guess that without hardware h264, such as on the RPI, it's all just a bit much for even reasonable hardware to stream adequately. I was hoping to get webcam feed + sound, but although mjpeg works, it's difficult to find another low cpu use way of transmitting both over http.
Thanks!
Andy

closed 2020-04-12 by mpromonet