mpromonet/v4l2rtspserver #161: “Queue full size drop frame size:10” problem

bleeee, 2019-11-30:
I run a BreeeZe's [rpos](https://github.com/BreeeZe/rpos) on pi4B and stream video surveillance to Synology Surveillance Station.
Additionally I use following python3 script to detect human faces and give camera servo motors commands to move.

But the script would always stuck at cv2.VideoCapture after a few moment(several minutes to half an hour) later.

Then I digged a little bit and run v4l2rtspserver with "-vv" option and found something in the log that at timestamp 1575083373-1575083374, the python3 script stuck and v4l2 started logging "Queue full size drop frame size:10"  to the time I stopped v4l2 manually.

What caused the problem and how could I solve it please?
[nohupv4l2.zip](https://github.com/mpromonet/v4l2rtspserver/files/3906772/nohupv4l2.zip)
[facenohup.zip](https://github.com/mpromonet/v4l2rtspserver/files/3906773/facenohup.zip)

```
import numpy as np
import cv2
import time
import requests

faceCascade = cv2.CascadeClassifier('Cascades/haarcascade_frontalface_default.xml')
control_cam=requests.Session()
posturl="http://ble:nimeide777@192.168.50.105:8001/cmd"
postright="TurnRight"
postleft="TurnLeft"
postup="Up"
postdown="Down"
poststop="stop"

while True:
    print("start:{}".format(time.time()))
    cap = cv2.VideoCapture("rtsp://192.168.50.105:8554/h264")
    ret, img = cap.read()
    cap.release()
    if ret:
        print("on:{}".format(time.time()))
        gray = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
        faces = faceCascade.detectMultiScale(
            gray,
            scaleFactor=1.2,
            minNeighbors=5,     
            minSize=(10, 10)
        )
        maxw=0
        maxx=0
        maxy=0
        maxh=0
        exist_flag=0
        for (x,y,w,h) in faces:
            if w>maxw:
                maxw=w
                maxx=x
                maxy=y
                maxh=h
            exist_flag=1
        if exist_flag==1:
            center_x=maxx+maxw/2
            center_y=maxy+maxh/2
            print(center_x)
            if center_x< 300:
                print("turnleft")
                resp=control_cam.post(posturl,data=postleft)
                time.sleep(0.5)
                resp=control_cam.post(posturl,data=poststop)
            elif center_x> 340:
                print("turnright")
                resp=control_cam.post(posturl,data=postright)
                time.sleep(0.5)
                resp=control_cam.post(posturl,data=poststop)
            print(center_y)
            if center_y< 168:
                print("goup")
                resp=control_cam.post(posturl,data=postup)
                time.sleep(0.5)
                resp=control_cam.post(posturl,data=poststop)
            elif center_y> 192:
                print("godown")     
                resp=control_cam.post(posturl,data=postdown)
                time.sleep(0.5)
                resp=control_cam.post(posturl,data=poststop) 
```

mpromonet, 2019-11-30:
Hi bleeee,
Looping on `cv2.VideoCapture("rtsp://192.168.50.105:8554/h264")` may be a little aggessive.
Maybe keeping the RTSP connection will increase the living time of the script.
Did you look to openned socket (for instance in TIME_WAIT state) ? 
Best Regards,
Michel.

bleeee, 2019-11-30:
> Hi bleeee,
> Looping on `cv2.VideoCapture("rtsp://192.168.50.105:8554/h264")` may be a little aggessive.
> Maybe keeping the RTSP connection will increase the living time of the script.
> Did you look to openned socket (for instance in TIME_WAIT state) ?
> Best Regards,
> Michel.

Checked openned socket, there were like 10 of them.
After the script was stopped, all the relevant socket were also gone, but still, v4l2 wouldn't accept new request.
I am thinking the problem is somehow with queueframe, maybe I should enlarge it with "-Q" option?

bleeee, 2019-11-30:
Currently checking out what could cause queueframe full.

bleeee, 2019-11-30:
The same stuck happens when thing goes not aggressively by the way

mpromonet, 2019-11-30:
Hi,
The queue is full when there is no client in order to remove old frame, or if frame could not be sent enough quickly.
Looking in the log you provide, we see that `deliverFrame` is no more called, so increasing the queue will probably change nothing.  
The dispaching thread seems blocked (or no more running), gdb or strace could help to understand where. 
Best Regards,
Michel.

bleeee, 2019-11-30:
> Hi,
> The queue is full when there is no client in order to remove old frame, or if frame could not be sent enough quickly.
> Looking in the log you provide, we see that `deliverFrame` is no more [...]

Thanks, I'll check it out.

bleeee, 2019-11-30:
> Hi,
> The queue is full when there is no client in order to remove old frame, or if frame could not be sent enough quickly.
> Looking in the log you provide, we see that `deliverFrame` is no more [...]

```
accept(3, {sa_family=AF_INET, sin_port=htons(5290), sin_addr=inet_addr("192.168.50.183")}, [16]) = 1124
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6ad5120}, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6ad5120}, 8) = 0
fcntl64(1124, F_GETFL)                  = 0x2 (flags O_RDWR)
fcntl64(1124, F_SETFL, O_RDWR|O_NONBLOCK) = 0
getsockopt(1124, SOL_SOCKET, SO_SNDBUF, [102400], [4]) = 0
getsockopt(1124, SOL_SOCKET, SO_SNDBUF, [102400], [4]) = 0
```
That happened when vlc wanted a new rtsp stream from v4l2 with the situation mentioned above.

Any idea?

bleeee, 2019-12-07:
New in C++, I don't get where deliverFrame is called. Can you help me out?

bleeee, 2019-12-10:
Never mind, it's not v4l2, it's the options.
After tuned it a little bit, the stability is much better now.
I'm closing it.

closed 2019-12-10 by bleeee

Tuckie, 2020-05-28:
@bleeee could you let me know how you tuned this?

bleeee, 2020-06-11:
> @bleeee could you let me know how you tuned this?
Eventually I failed.

mpromonet, 2020-06-14:
Hi @bleeee 

I made some experiment with opencv, I don't have problem running :
```
import cv2
cap = cv2.VideoCapture("rtsp://192.168.0.16:8554/unicast")
while cap.isOpened():
    ret, img = cap.read()
    if ret:
        cv2.imshow('frame',img)
        cv2.waitKey(10)
```
But adding some detection processing, images are read slower than send, and probably the network buffer overflow.
In such a situtation it seems the live mainloop is no more running (probably locked by the mutex) and then disconnects other clients.... this could probably be improved.
If your problem is that one, it may be a way to not process all frames to speed up the frame consumption.

Best Regards,
Michel.