mpromonet/v4l2rtspserver #162: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

Mingala, 2019-11-30:
With raspbian 4.19 (20190926) on a pi zero W and the following setup (service) using the bcm2835-v4l2 module and (camera V1), i have this error displayed by v4l2rtspserver after "[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131".
Note that the rtsp stream is still accessible. I did not get this behaviour with previous release of raspbian/v4l2 (4.14).

Service :
# qi2.info-v4l2rtspserver.service
[Unit]
Description=qi2 V4l2 Rtsp server using https://github.com/mpromonet/v4l2rtspserver
[Service]
Type=simple
ExecStartPre=/usr/bin/v4l2-ctl --set-fmt-video=width=1296,height=972,pixelformat=H264 --set-parm=30 --overlay=0 --set-ctrl=vertical_flip=1,horizontal_flip=1
ExecStart=/usr/local/bin/v4l2rtspserver -f -F 0 -u live /dev/video0
[Install]
WantedBy=default.target

Best Regards,
Laurent.

cdgriffith, 2020-01-23:
I am having the same issue and **cannot** access the RTSP stream of a H264 camera (/dev/video2).  

Using [newest commit](https://github.com/mpromonet/v4l2rtspserver/commit/73358a52937d8d733e11279dc6300cac06e1749f) and raspbian version buster 10 kernel  4.19.93+

<details>
<summary>v4l2 device info</summary>
<p>

```
v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

Webcam 1080P HDR: Webcam 1080P  (usb-20980000.usb-1.2):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
```

</p>
</details>
<details><summary>v4l2 format info</summary>
<p>

```
v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'H264' (H.264, compressed)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x544
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
```

</p>
</details>
<details><summary>v4l2rtspserver log</summary>
<p>

```
/usr/local/bin/v4l2rtspserver /dev/video2
log level:500

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:607
        Create V4L2 Source.../dev/video2
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
        driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
        /dev/video2 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:130
        /dev/video2 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
        /dev/video2:H264 size:640x480 bufferSize:614400
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:206
        /dev/video2:H264 size:640x480 bufferSize:614400
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
        /dev/video2:H264 size:640x480 bufferSize:614400
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:227
        fps:1/30
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
        nbBuffer:0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device /dev/video2
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
        Device /dev/video2 nb buffer:10
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:630
        Create Source .../dev/video2
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:150
        Play this stream using the URL "rtsp://192.168.1.169:8554/unicast"
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
        begin thread
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
```

</p>
</details>

mpromonet, 2020-01-23:
Hi,
The log just show that the client disconnect from the RTSP server.
A possible case is that capture doesnot give SPS/PPS or give it but v4l2rtspserver doesnot split the frame correctly.
Looking to log without more verbosity should helps to understand if SPS/PPS is processed.
Best Regards,
Michel.

cdgriffith, 2020-01-25:
Trying again with `-vv`

```
v4l2rtspserver -vv /dev/video2
log level:700

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:607
        Create V4L2 Source.../dev/video2
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
        driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
        /dev/video2 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:130
        /dev/video2 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
        /dev/video2:H264 size:640x360 bufferSize:460800
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:206
        /dev/video2:H264 size:640x360 bufferSize:460800
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
        /dev/video2:H264 size:640x360 bufferSize:460800
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:227
        fps:1/30
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
        nbBuffer:0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device /dev/video2
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
        Device /dev/video2 nb buffer:10
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:0 size:460800 offset:0
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:1 size:460800 offset:462848
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:2 size:460800 offset:925696
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:3 size:460800 offset:1388544
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:4 size:460800 offset:1851392
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:5 size:460800 offset:2314240
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:6 size:460800 offset:2777088
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:7 size:460800 offset:3239936
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:8 size:460800 offset:3702784
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:9 size:460800 offset:4165632
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:630
        Create Source .../dev/video2
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:150
        Play this stream using the URL "rtsp://192.168.1.169:8554/unicast"
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
        begin thread
[DEBUG] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:146
RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
```

After  VLC or BlueIris try to connect to it for a while will start getting the: 

```
        Queue is empty
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[DEBUG] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:146
        Queue is empty
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[DEBUG] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:146
        Queue is empty
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[DEBUG] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:146
        Queue is empty
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
        V4L2DeviceSource::doStopGettingFrames
[DEBUG] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:146
```

mpromonet, 2020-01-25:
Hi,
From the logs, it seems that reading frame is not working, there is no trace from `V4L2DeviceSource::getNextFrame`, neither captured frame nor error.    
You may try `-r` option to use read interface instead of mmap and/or `-B` to capture in blocking mode. `strace` could also helps you to find which system call is failing.
Best Regards,
Michel.

cdgriffith, 2020-01-25:
Thanks for the help and ideas! I have also now tried ffmpeg and it just hangs when it tries to read from that video device, so it seems to be something with the raspberry pi zero itself (as it works on Pi 3 and Pi 4 just fine.)

ka-lauer, 2020-02-02:
Hi,
i have the same issue:
- Stream gives just a green screen with a smal stripe cutout of the streamed picture on left upper corner.
- It says " RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero! "

Fresh installed buster on raspberry 3b+ and 4 4Gb (both exactly same issue), full-upgrade finished before installation of v4l2rtspserver.

Tested with stream to VLC on Windows, on Android, same result when testing in surveillence station on Synology and when connection via gstream/jetcam from python. On Linux/VLC no picture appears but same massage on rtspserver.

-vv shows the same message than @cdgriffith 

ka-lauer

mpromonet, 2020-02-02:
@ka-lauer 
I guess the log are a bit different, and you are using MJPG and not H264 as shown in the log of @cdgriffith, isn't it ?
Best Regards,
Michel.

ka-lauer, 2020-02-02:
Thanks a lot for the quick response!

Log is:

`pi@raspberrypi:~ $ v4l2rtspserver -v /dev/video1
log level:600

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:607
	Create V4L2 Source.../dev/video1
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
	/dev/video1 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:130
	/dev/video1 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
	/dev/video1:MJPG size:752x416 bufferSize:409600
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:193
	**Cannot set pixelformat to:H264 format is:YUYV**
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:206
	/dev/video1:MJPG size:752x416 bufferSize:341333
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
	/dev/video1:MJPG size:752x416 bufferSize:341333
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:227
	fps:1/25
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
	nbBuffer:0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video1
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video1 nb buffer:10
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:0 size:409600 offset:0
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:1 size:409600 offset:409600
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:2 size:409600 offset:819200
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:3 size:409600 offset:1228800
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:4 size:409600 offset:1638400
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:5 size:409600 offset:2048000
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:6 size:409600 offset:2457600
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:7 size:409600 offset:2867200
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:8 size:409600 offset:3276800
[INFO] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video1 buffer idx:9 size:409600 offset:3686400
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:630
	Create Source .../dev/video1
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.2.97:8554/unicast"
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1580678322 fps:1 bandwidth:278kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1580678323 fps:10 bandwidth:2783kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1580678324 fps:24 bandwidth:6684kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1580678325 fps:24 bandwidth:6693kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
`
when starting to receive the RTSP stream with VLC:

`
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
**RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!**
	intv_sec:1580678348 fps:24 bandwidth:6668kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	outtv_sec:1580678348 fps:1 bandwidth:277kbps
[INFO] /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:43
	width:752 height:416 type:0 precision:8
[INFO] /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:43
	width:752 height:416 type:0 precision:8
[INFO] /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:43
	width:752 height:416 type:0 precision:8
[INFO] /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:43
`
~~

`	width:752 height:416 type:0 precision:8
[NOTICE] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1580678355 fps:24 bandwidth:6647kbps
[INFO] /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
^C	intv_sec:1580678356 fps:24 bandwidth:6651kbpsSIGINT

[NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:748
	Exiting....pi@raspberrypi:~ $ 
`
I receive at leased a little piece of the video...
And your right, its not H264! Camera is Logitech 270.

ka-lauer

closed 2020-04-12 by mpromonet

kubedzero, 2021-01-22:
@iamdavidcz  where did you find and change that `estBitrate` value, and what steps were needed to apply that change to the compiled files?

bennybubble, 2021-02-09:
@iamdavidcz : 
Could you be more clear on this? I'm having above issue and it is driving me crazy :-) 
`v4l2rtspserver[903]: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!`
It stops the v4l2rtspserver and I have to reboot my RPI, restarting the service does not help.
I have two similar setups (both RPI) using V4l2rtspserver and the one with a seperate (UVC) webcam is causing the trouble, the other one has a RPI cam (H264) and works like a charm.

mpromonet, 2021-02-09:
Hi @iamdavidcz 

It is probable there is no link between the warning `RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!` and stop of v4l2rtspserver.
You may use a debugging tools like gdp or strace or valgrind to get more information of the reason that cause process to stop.

Best Regards,
Michel.