mpromonet/v4l2rtspserver #163: Pi bcm2835-codec (bcm2835-v4l2) use of memory to memory (M2M) device /dev/video10

Mingala, 2019-11-30:
In the new raspbian 4.19, the bcm2835-codec module loaded by default expose three v4l2 memory to memory (M2M) devices for the video decoder, encoder, and ISP resize respectively : /dev/video10, /dev/video11 and /dev/video12 (https://lb.raspberrypi.org/forums/viewtopic.php?f=43&t=62364&sid=307c744f6f3825158bd68b8cd08b6318&start=525).
Is it possible to use the first directly with v4l2rtspserver command ? If so could you please be kind enough to provide a sample.
Best Regards,
Laurent.

mpromonet, 2019-11-30:
Hi Laurent,

In past I look to find information about M2M and this new devices, but I find very few information, too few to implement interface with it.   
I guess it should be possible to feed v4l2rtspserver with the output of an H264 or JPEG encoder.
This probably need some extension of libv4l2cpp to access M2M device.

Best Regards,
Michel.

Mingala, 2019-12-01:
Hi Michel,
Thanks for the quick comment. I will keep an eye on these new devices and watch uf any use through libv4l2cpp appears.
Best Regards,
Laurent.

closed 2019-12-01 by Mingala