mpromonet/v4l2rtspserver #164: Issue compiling on Raspberry Pi Zero W Raspbian Buster latest

tingles2233, 2019-12-08:
Hello, 

Having the following error compiling on a Raspberry Pi Zero W - Running Raspbian Buster Minimal

[ 99%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
[ 99%] Building CXX object CMakeFiles/v4l2rtspserver.dir/src/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/usr/bin/ld: liblive555.a(TLSState.cpp.o): in function `TLSState::connect(int)':
TLSState.cpp:(.text+0x128): undefined reference to `SSL_connect'
/usr/bin/ld: TLSState.cpp:(.text+0x144): undefined reference to `SSL_get_error'
/usr/bin/ld: liblive555.a(TLSState.cpp.o): in function `TLSState::write(unsigned char const*, unsigned int)':
TLSState.cpp:(.text+0x254): undefined reference to `SSL_write'
/usr/bin/ld: liblive555.a(TLSState.cpp.o): in function `TLSState::read(unsigned char*, unsigned int)':
TLSState.cpp:(.text+0x294): undefined reference to `SSL_read'
/usr/bin/ld: liblive555.a(TLSState.cpp.o): in function `TLSState::reset()':
TLSState.cpp:(.text+0x2d4): undefined reference to `SSL_shutdown'
/usr/bin/ld: TLSState.cpp:(.text+0x2f4): undefined reference to `SSL_free'
/usr/bin/ld: TLSState.cpp:(.text+0x320): undefined reference to `SSL_CTX_free'
/usr/bin/ld: liblive555.a(TLSState.cpp.o): in function `TLSState::setup(int)':
TLSState.cpp:(.text+0x35c): undefined reference to `OPENSSL_init_ssl'
/usr/bin/ld: TLSState.cpp:(.text+0x360): undefined reference to `TLS_client_method'
/usr/bin/ld: TLSState.cpp:(.text+0x37c): undefined reference to `SSL_CTX_new'
/usr/bin/ld: TLSState.cpp:(.text+0x3a8): undefined reference to `SSL_new'
/usr/bin/ld: TLSState.cpp:(.text+0x3d0): undefined reference to `BIO_new_socket'
/usr/bin/ld: TLSState.cpp:(.text+0x3f0): undefined reference to `SSL_set_bio'
/usr/bin/ld: TLSState.cpp:(.text+0x400): undefined reference to `SSL_set_connect_state'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:236: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
root@raspberrypi:~/v4l2rtspserver#
root@raspberrypi:~/v4l2rtspserver# 

I have tried a bunch and openSSL is installed, even did a factory reset on the device and started over to ensure I didn`t mess up anything long the way. 

Willing to test / check anything .. let me know

mpromonet, 2019-12-08:
Hi,
This is coming from live555 2019.12.07 that introduce a dependency with openssl.
I will update to not add the openssl dependency.
Best Regards,
Michel.

closed 2019-12-08 by mpromonet

tingles2233, 2019-12-08:
Just wanted to say thanks! All is working now! 

Have a great day!

mickjc750, 2021-07-02:
I'm getting this exact error (even after the fix), I've tried a bunch too! :-(

mpromonet, 2021-07-02:
Hi @mickjc750 

Very nice issue description :)

Best Regards,
Michel.

mickjc750, 2021-07-02:
> Hi @mickjc750
> 
> Very nice issue description :)
> 
> Best Regards,
> Michel.

Possibly some misunderstanding. I'm not trying to raise a NEW issue. By 'this exact error' I mean the exact same error as @tingles2233 experienced (which was apparently fixed). That's why I commented on the same issue.

I also have a PiZeroW with minimal Pi OS Buster.
I have apt update+upgraded.
I installed live555, which required me to first install libssl-dev.
I downloaded and built libv4l2cpp, and put the resulting libv4l2cpp.a in /usr/bin is that where it goes?
I installed the dependencies liblivemedia-dev and liblog4cpp5-dev (packages via apt).
I get no errors until link time, then my error output matches that of tingles.

What could I be missing?

Regards,

mpromonet, 2021-07-02:
Hi,

The error is fixed if you don't install live555 as a package and let cmake download and compile it. Note the library name requiring ssl dependency `liblive555.a`, your message is probably different.
Why not just run `cmake . && make` ? It normaly download what needs.
In your situation you probably need to add `-lssl` to the link.

Best Regards,
Michel.

commit 73358a52 (2019-12-08): fix #164: build live555 without tls support