mpromonet/v4l2rtspserver #166: Convert stream to black and white / greyscale

parautenbach, 2019-12-16:
Is there a way to do this? I can't seem to find info on this. I'd like to do something similar to ffmpeg where one can set hue=s=0. I don't seen options to do this directly with v4l2rtspserver or v4l2-ctl. 

The reason I want to do this is to make an infrared camera feed look a little more pleasing.

mpromonet, 2020-01-03:
Hi Peter,
This project doesnot modify image, it streams what it capture using the V4L2 interface.
Depending on the V4L2 device this could be possible using the appropiate control.
Using the driver bcm2835_v4l2, the black & white could be set using : 
```
v4l2-ctl -c saturation=-100
```
or 
```
v4l2-ctl -c color_effects=1
```
An other way is to add a v4l2loopback device that makes the tranformation, but this will need some processing ressources...
Best Regards,
Michel.

closed 2020-02-01 by mpromonet

parautenbach, 2020-02-02:
Thanks for the explanation. 

The saturation setting is easy to understand, but how do you know what the available colour effects are and what they mean? Does this depend on the specific device?

mpromonet, 2020-02-02:
Hi parautenbach,
Yes this is depending on the device driver. you can get list of controls supported by the driver using : 

`v4l2-ctl --all -d /dev/video0`

Best Regards,
Michel.

parautenbach, 2020-02-03:
Thanks, again.

I'm getting this:

```
color_effects 0x0098091f (menu)   : min=0 max=15 default=0 value=0
```

Do I need to check the vendor's page for more details on what the different values for this control would mean?

ostecke, 2022-03-07:
`v4l2-ctl -c saturation=-100`

worked perfectly for me using a Raspberry Pi Camera Module 2 NoIR.