mpromonet/v4l2rtspserver #169: Failed "make" but completed cmake

Condorello, 2020-01-15:
Hi, i'm try to compile for Arm uclibc, cmake compiled fine with:

`SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_PROCESSOR armv6l)
SET(CPACK_DEBIAN_PACKAGE_ARCHITECTURE arm)
SET(CMAKE_C_COMPILER /home/andrea/x-tools/arm-unknown-linux-uclibcgnueabi/bin/arm-unknown-linux-uclibcgnueabi-cc)
SET(CMAKE_CXX_COMPILER /home/andrea/x-tools/arm-unknown-linux-uclibcgnueabi/bin/arm-unknown-linux-uclibcgnueabi-g++)

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
`

but make fail with:

`[ 37%] Building CXX object CMakeFiles/live555.dir/live/liveMedia/Locale.cpp.o
In file included from /home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:22:
/home/andrea/v4l2rtspserver/live/liveMedia/include/Locale.hh:67:3: error: 'locale_t' does not name a type; did you mean 'Locale'?
   locale_t fLocale, fPrevLocale;
   ^~~~~~~~
   Locale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp: In constructor 'Locale::Locale(const char*, LocaleCategory)':
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:30:32: error: 'LC_ALL_MASK' was not declared in this scope
     case All: { categoryMask = LC_ALL_MASK; break; }
                                ^~~~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:30:32: note: suggested alternative: 'LC_ALL'
     case All: { categoryMask = LC_ALL_MASK; break; }
                                ^~~~~~~~~~~
                                LC_ALL
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:31:36: error: 'LC_NUMERIC_MASK' was not declared in this scope
     case Numeric: { categoryMask = LC_NUMERIC_MASK; break; }
                                    ^~~~~~~~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:31:36: note: suggested alternative: 'LC_NUMERIC'
     case Numeric: { categoryMask = LC_NUMERIC_MASK; break; }
                                    ^~~~~~~~~~~~~~~
                                    LC_NUMERIC
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:33:3: error: 'fLocale' was not declared in this scope
   fLocale = newlocale(categoryMask, newLocale, NULL);
   ^~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:33:3: note: suggested alternative: 'Locale'
   fLocale = newlocale(categoryMask, newLocale, NULL);
   ^~~~~~~
   Locale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:33:13: error: 'newlocale' was not declared in this scope
   fLocale = newlocale(categoryMask, newLocale, NULL);
             ^~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:33:13: note: suggested alternative: 'newLocale'
   fLocale = newlocale(categoryMask, newLocale, NULL);
             ^~~~~~~~~
             newLocale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:34:3: error: 'fPrevLocale' was not declared in this scope
   fPrevLocale = uselocale(fLocale);
   ^~~~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:34:3: note: suggested alternative: 'newLocale'
   fPrevLocale = uselocale(fLocale);
   ^~~~~~~~~~~
   newLocale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:34:17: error: 'uselocale' was not declared in this scope
   fPrevLocale = uselocale(fLocale);
                 ^~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:34:17: note: suggested alternative: 'setlocale'
   fPrevLocale = uselocale(fLocale);
                 ^~~~~~~~~
                 setlocale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp: In destructor 'virtual Locale::~Locale()':
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:49:7: error: 'fLocale' was not declared in this scope
   if (fLocale != (locale_t)0) {
       ^~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:49:7: note: suggested alternative: 'Locale'
   if (fLocale != (locale_t)0) {
       ^~~~~~~
       Locale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:49:19: error: 'locale_t' was not declared in this scope
   if (fLocale != (locale_t)0) {
                   ^~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:49:19: note: suggested alternative: 'Locale'
   if (fLocale != (locale_t)0) {
                   ^~~~~~~~
                   Locale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:49:28: error: expected ')' before numeric constant
   if (fLocale != (locale_t)0) {
      ~                     ^
                            )
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:50:15: error: 'fPrevLocale' was not declared in this scope
     uselocale(fPrevLocale);
               ^~~~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:50:15: note: suggested alternative: 'Locale'
     uselocale(fPrevLocale);
               ^~~~~~~~~~~
               Locale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:50:5: error: 'uselocale' was not declared in this scope
     uselocale(fPrevLocale);
     ^~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:50:5: note: suggested alternative: 'setlocale'
     uselocale(fPrevLocale);
     ^~~~~~~~~
     setlocale
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:51:5: error: 'freelocale' was not declared in this scope
     freelocale(fLocale);
     ^~~~~~~~~~
/home/andrea/v4l2rtspserver/live/liveMedia/Locale.cpp:51:5: note: suggested alternative: 'setlocale'
     freelocale(fLocale);
     ^~~~~~~~~~
     setlocale
CMakeFiles/live555.dir/build.make:1790: recipe for target 'CMakeFiles/live555.dir/live/liveMedia/Locale.cpp.o' failed
make[2]: *** [CMakeFiles/live555.dir/live/liveMedia/Locale.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/live555.dir/all' failed
make[1]: *** [CMakeFiles/live555.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
`

Any help is really apreciated!

mpromonet, 2020-01-15:
Hi @Condorello 

It seems your cross-compiler doesnot have locale.
You may try some options like : 
```
cmake -DLIVE555CFLAGS="-DNEWLOCALE_NOT_USED=1  -DBSD=1 -DSOCKLEN_T=socklen_t -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -DALLOW_RTSP_SERVER_PORT_REUSE=1 -DNO_OPENSSL=1" . 
```
Best Regards,
Michel.

Condorello, 2020-01-16:
Hi @mpromonet
Thanks for your kind support, i tried but now i get:

`[  5%] Building CXX object CMakeFiles/live555.dir/live/groupsock/GroupsockHelper.cpp.o
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp: In function 'int readSocket(UsageEnvironment&, int, unsigned char*, unsigned int, sockaddr_in&)':
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:343:7: error: invalid conversion from 'int*' to 'socklen_t*' {aka 'unsigned int*'} [-fpermissive]
       &addressSize);
       ^~~~~~~~~~~~
In file included from /home/andrea/v4l2rtspserver/live/groupsock/include/NetCommon.h:95,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/NetAddress.hh:29,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/GroupsockHelper.hh:25,
                 from /home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:21:
/home/andrea/x-tools/arm-unknown-linux-uclibcgnueabi/arm-unknown-linux-uclibcgnueabi/sysroot/usr/include/sys/socket.h:176:27: note:   initializing argument 6 of 'ssize_t recvfrom(int, void*, size_t, int, sockaddr*, socklen_t*)'
     socklen_t *__restrict __addr_len);
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp: In function 'unsigned int getBufferSize(UsageEnvironment&, int, int)':
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:429:21: error: invalid conversion from 'int*' to 'socklen_t*' {aka 'unsigned int*'} [-fpermissive]
    (char*)&curSize, &sizeSize) < 0) {
                     ^~~~~~~~~
In file included from /home/andrea/v4l2rtspserver/live/groupsock/include/NetCommon.h:95,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/NetAddress.hh:29,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/GroupsockHelper.hh:25,
                 from /home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:21:
/home/andrea/x-tools/arm-unknown-linux-uclibcgnueabi/arm-unknown-linux-uclibcgnueabi/sysroot/usr/include/sys/socket.h:219:32: note:   initializing argument 5 of 'int getsockopt(int, int, int, void*, socklen_t*)'
          socklen_t *__restrict __optlen) __THROW;
          ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp: In function 'Boolean getSourcePort0(int, portNumBits&)':
/home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:616:52: error: invalid conversion from 'int*' to 'socklen_t*' {aka 'unsigned int*'} [-fpermissive]
   if (getsockname(socket, (struct sockaddr*)&test, &len) < 0) return False;
                                                    ^~~~
In file included from /home/andrea/v4l2rtspserver/live/groupsock/include/NetCommon.h:95,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/NetAddress.hh:29,
                 from /home/andrea/v4l2rtspserver/live/groupsock/include/GroupsockHelper.hh:25,
                 from /home/andrea/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:21:
/home/andrea/x-tools/arm-unknown-linux-uclibcgnueabi/arm-unknown-linux-uclibcgnueabi/sysroot/usr/include/sys/socket.h:128:26: note:   initializing argument 3 of 'int getsockname(int, sockaddr*, socklen_t*)'
    socklen_t *__restrict __len) __THROW;
    ~~~~~~~~~~~~~~~~~~~~~~^~~~~
CMakeFiles/live555.dir/build.make:110: recipe for target 'CMakeFiles/live555.dir/live/groupsock/GroupsockHelper.cpp.o' failed
make[2]: *** [CMakeFiles/live555.dir/live/groupsock/GroupsockHelper.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/live555.dir/all' failed
make[1]: *** [CMakeFiles/live555.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
`

Any advice about GroupsockHelper? Thanks again!
Andrea

mpromonet, 2020-01-18:
Hi Andrea,
You may start compiling live555 with your crosscompiler to tune the compiler option. 
There may be a a config that correspond to it.
When you will got the correct options you could run `cmake -DLIVE555CFLAGS="..."`.
Best Regards,
Michel.

Buddy-Matt, 2023-01-30:
I know I'm reviving a necro'd thread here - but I was having the exact above issue trying to compile using dockcross targeting some horrible cheap ipc camera.

`cmake -DLIVE555CFLAGS="-DNEWLOCALE_NOT_USED=1  -DBSD=1 -DSOCKLEN_T=socklen_t -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -DALLOW_RTSP_SERVER_PORT_REUSE=1 -DNO_OPENSSL=1" .`
doesn't work, because behind-the-scenes it translates to this:
`-DNEWLOCALE_NOT_USED="1  -DBSD=1 -DSOCKLEN_T=socklen_t -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -DALLOW_RTSP_SERVER_PORT_REUSE=1 -DNO_OPENSSL=1"`
The issue being all the other defs are incorporated into NEWLOCALE_NOT_USED inadventantly.

This, however, appears to work (at least, I've got a new error now about V4L2_PIX_FMT_H264, implying the DEFS are working as expected):
`cmake -DLIVE555CFLAGS="-DNEWLOCALE_NOT_USED=1; -DBSD=1; -DSOCKLEN_T=socklen_t; -D_FILE_OFFSET_BITS=64; -D_LARGEFILE_SOURCE=1; -DALLOW_RTSP_SERVER_PORT_REUSE=1; -DNO_OPENSSL=1;" .`

Buddy-Matt, 2023-02-24:
single/double quotes made no difference. The trick was to add the semicolons.

mpromonet, 2023-02-24:
Interesting, there is no problem using openwrt build system. Maybe i will look to dockcross i never heard about.
Best Regards
Michel