mpromonet/v4l2rtspserver #170: Green screen

alduxvm, 2020-01-20:
Hi, 

I followed the steps to install, I'm using an ARM processor and an integrated camera, so I start the server using:

`v4l2rtspserver /dev/video0`

Then in another computer in the network, I open the stream: `rtsp://xx.xx.xx.xx:8554/unicast`, but I always get a green screen with a thin bar on the top that shows my image, no matter how long I wait, its always green...

https://i.imgur.com/sY44lzb.png

I also compiled live555 but is still the same... any ideas?

mpromonet, 2020-01-20:
Hi,
If you are using JPEG capture, it may be similar to https://github.com/mpromonet/v4l2rtspserver/issues/140 

Maybe some problem parsing the JPEG frame ?

Best Regards,
Michel.

alduxvm, 2020-01-20:
If I open the camera using `vlc /dev/video0` it works fine... I'm not sure if its a JPEG problem...

alduxvm, 2020-01-20:
This is the output of `v4l2-ctl`:
```aiSage@linaro-alip:~/v4l2rtspserver$ v4l2-ctl -V
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 3840
	Size Image        : 4147200
	Colorspace        : sRGB
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Flags             :
```

And this is the log for the server:
```aiSage@linaro-alip:~/v4l2rtspserver$ v4l2rtspserver /dev/video0
log level:500
[NOTICE] /home/aiSage/v4l2rtspserver/src/main.cpp:607
	Create V4L2 Source.../dev/video0
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
	/dev/video0 support capture
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:130
	/dev/video0 support streaming
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
	/dev/video0:YUYV size:1920x1080 bufferSize:4147200
[ERROR] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:193
	Cannot set pixelformat to:H264 format is:YUYV
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:206
	/dev/video0:MJPG size:1920x1080 bufferSize:4147200
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
	/dev/video0:MJPG size:1920x1080 bufferSize:4147200
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:227
	fps:1/30
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:228
	nbBuffer:0
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
[NOTICE] /home/aiSage/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
[NOTICE] /home/aiSage/v4l2rtspserver/src/main.cpp:630
	Create Source .../dev/video0
[NOTICE] /home/aiSage/v4l2rtspserver/src/main.cpp:150
	Play this stream using the URL "rtsp://192.168.0.129:8554/unicast"
[NOTICE] /home/aiSage/v4l2rtspserver/src/DeviceSource.cpp:93
```

mpromonet, 2020-01-20:
Hi,

From the log we see :
`/dev/video0:MJPG size:1920x1080 bufferSize:4147200`

Then it is MJPG capture, you can try different one using  `-fJPEG`.
It may be helpful to run enabling logs, maybe some information about parsing could helps.

Best Regards,
Michel.

f00dl3, 2020-01-28:
I am having this same issue. I have a raspberry pi 3 using the Linux raspberrypi2 4.19.66-v7+ #1253

Green image is all that I can capture from a USB camera with the v4l2rtspstream. On another Raspberry Pi, I do have a Pi NoIR camera and the older h264_v4l2_rtspstream program (which looks like was abandoned.). That works. using H264. But when I tried to use the USB camera with H264, it says not supported. 

When I try doing ffmpeg rtsp://localhost:8554/unicast -update 1 test.jpg, I get tons of "mjepg_decode_dc: bad vlc: 0:0 (0x216179c) errors from the stream. It does eventually pull a frame, but the frame again is all green except a few pixels with an image in the upper right hand corner.

mpromonet, 2020-01-28:
Hi @f00dl3,
h264_v4l2_rtspserver was renamed v4l2rtspserver years ago, but this is the same project.   
Now v4l2rtspserver try different format, first H264, and if it fails, try MJPG. In past if H264 was not supported, there was not fallback.   
You still can to force format using `-fH264`.   
If the USB camera doesnot support H264, you should probably look to the driver (if the hardware support this format).   
One possible way is to use an intermediate v4l2 device or investigate what cause problems parsing JPEG and forwarding over RTP.
Best Regards,
Michel.

isolin, 2020-02-01:
@alduxvm I had the same problem with a Logitech C510 webcam. I solved the green video (caused by corrupt mjpeg data that could not form a valid stream) by placing a raw to h264 compressor in front of the rtsp server. 

TL;DR
```
modprobe v4l2loopback video_nr=10
v4l2compress_omx /dev/video0 /dev/video10 &
v4l2rtspserver /dev/video10 &
```

So I had to 
1. `sudo apt install v4l2loopback-dk git cmake autoconf libtool-bin`
2. `cd ~ && git clone https://github.com/mpromonet/v4l2tools.git` then in the folder call `make` and `make install`
3. `modprobe v4l2loopback video_nr=10` to create a virtual video device with number 10 (if occupied choose another one)
4. `v4l2compress_omx /dev/video0 /dev/video10 &` to compress your raw video to h264 using Raspberry's HW encoder
5. `v4l2rtspserver /dev/video10 &` start the server

@mpromonet Awesome tools! Thank you very much for creating and sharing them.

f00dl3, 2020-02-03:
Thank you! I was unable to get v4l2tools to compile because of the
libtool-bin dependency - but that did it! Also, the package is
v4l2loopback-dkms, at least on Buster. It's now working excellently.

mpromonet, 2020-02-14:
Hi,

I fix a problem in the DRI parsing in commit https://github.com/mpromonet/v4l2rtspserver/commit/c4ae12e70801a8deb9f199f09d2f49463965031e 
This allow to stream MJPG video from Logitech C270, maybe for others models.

Best Regards,
Michel.

closed 2020-02-19 by mpromonet